RobotLocomotion/drake #673: frame error in runMixedIntegerSimpleForest

RussTedrake, 2015-01-02:
Site: drake007
Build Name: master-RobotLocomotion-master
Build Server Directory: /tmp/drake-master-1644/drake
CDash: http://kobol.csail.mit.edu/cd/index.php?project=Drake&date=20150102

Most recent commits to drake included in this build:
```
364335f Merge pull request #666 from RussTedrake/pulley
fa19104 add numerical test on length for the pulleys
afd7caf Merge branch 'pulley' of https://github.com/RussTedrake/drake into
pulley
01f7d1e renamed function CablesAndPulleys to CableLength.  Minor tweaks to
example
a02e099 Merge pull request #15 from kuindersma/fix-torque-calculation
```
examples/Quadrotor/runMixedIntegerSimpleForest (unknown issue)
```
error_id = 
error_message = frame dimension does not match number of inputs
is_segfault = 0
```
```Matlab

                           < M A T L A B (R) >
                 Copyright 1984-2014 The MathWorks, Inc.
                   R2014a (8.3.0.532) 64-bit (glnxa64)
                            February 11, 2014

To get started, type one of these: helpwin, helpdesk, or demo.
For product information, visit www.mathworks.com.

To reproduce this test use rng(47517574,'twister')

[Warning: Disabling autosave for simulink blocks (to avoid generating a
lot of
*.mdl.autosave files in your directory.  If you aren't a regular Simulink
user
and don't want this disabled, comment out this section in addpath_drake.
To disable this warning in the future, add
  warning('off','Drake:DisablingSimulinkAutosave')
to your matlab startup.m] 
[> In addpath_drake at 74
 In addpath_drake at 4] 
Added the lcm jar to your javaclasspath (found via cmake)
[Warning: Currently all of your LCM traffic will be broadcast to the
network,
because:
 Loopback interface is not set to multicast.
 Consider running: sudo
 /tmp/drake-master-1644/drake/util/setup_loopback_multicast.sh
] 
[> In checkDependency at 63
 In checkDependency at 74
 In runMixedIntegerEnvironment at 16
 In runMixedIntegerSimpleForest at 21
 In fevalPackageSafe at 21] 
Added the lcmgl jar to your javaclasspath (found via cmake)
Calling addpath_iris
Calling addpath_mosek

MOSEK Version 7.1.0.12 (Build date: 2014-12-12 07:25:47)
Copyright (c) 1998-2014 MOSEK ApS, Denmark. WWW: http://mosek.com
Platform: Linux/64-X86

For help type: help mosekopt.
LCM: Disabling IPV6 support
LCM: TTL set to zero, traffic will not leave localhost.
Running mixed-integer convex program for 3rd-degree polynomial...
Calling addpath_yalmip

[... 245 lines cut ...]
 In fevalPackageSafe at 21] 
Calling addpath_snopt
done!
Error using DynamicalSystem/setInputFrame (line 432)
frame dimension does not match number of inputs

Error in runMixedIntegerEnvironment (line 146)
v =
v.setInputFrame(sys.getOutputFrame().getFrameByName('quadrotorPosition'));

Error in runMixedIntegerSimpleForest (line 21)
runMixedIntegerEnvironment(r, start, goal, lb, ub, seeds, degree,
n_segments, n_regions);

Error in fevalPackageSafe (line 21)
 feval(f);

<test_name>examples/Quadrotor/runMixedIntegerSimpleForest</test_name>
<error_id></error_id> <error_message>frame dimension does not match number
of inputs</error_message> 

```

<matlab_test>strcmp(error_message,'frame dimension does not match number
of inputs') && strcmp(test_name,'examples/Quadrotor/runMixedIntegerSimpleForest')</matlab_test>

rdeits, 2015-01-02:
This does not occur for me on RobotLocomotion/drake master on my machine. I'll try to reproduce it on another machine.

rdeits, 2015-01-02:
Both my mac and my DRC laptop run the example without encountering this error on drake/master.

rdeits, 2015-01-02:
This also doesn't occur for me at 364335f which is the commit at which the attached error message was generated. I have no idea why it shows up on the build servers and not on my machines...

RussTedrake, 2015-01-02:
i'll take a look on my machines.

RussTedrake, 2015-01-02:
can you try running it after calling

```
global g_disable_visualizers; g_disable_visualizers=true;
```

RussTedrake, 2015-01-02:
have you seen this one?

```
Simulating the system...
Warning: system has constraints, but they aren't enforced in the simulink model
yet. 
> In WarningManager>WarningManager.warnOnce at 21
  In DrakeSystem.DrakeSystem>DrakeSystem.getModel at 253
  In DynamicalSystem.simulate at 23
  In DrakeSystem.DrakeSystem>DrakeSystem.simulate at 490
  In runMixedIntegerEnvironment at 136
  In runMixedIntegerSimpleForest at 21 
 Calling addpath_snopt
DrakeSystem S-Function: error when calling ''getInitialState'' with the following arguments:
  FeedbackSystem with properties:

               sys1: [1x1 Quadrotor]
               sys2: [1x1 AffineSystem]
            sys1ind: [12x1 double]
            sys2ind: [0x1 double]
               umin: [4x1 double]
               umax: [4x1 double]
    warning_manager: [1x1 WarningManager]

An error occurred while trying to determine whether "permute" is a function name.

Error in cross (line 64)
    c = ipermute(c,perm);

Error in RigidBodyManipulator/surfaceTangents (line 1914)
      t2 = cross(t1,normal);

Error in RigidBodyManipulator/contactConstraints (line 60)
d = obj.surfaceTangents(normal);

Error in RigidBodyManipulator/compile/nonpenetrationConstraint (line 802)
          [phi,~,~,~,~,~,~,~,dphi] = contactConstraints(model,kinsol,false,model.contact_options);

Error in FunctionHandleConstraint/constraintEval (line 44)
      [varargout{:}] = obj.eval_handle(varargin{:});

Error in Constraint/eval (line 121)
          [varargout{:}] = obj.constraintEval(varargin{:});

Error in NonlinearProgram/objectiveAndNonlinearConstraints (line 568)
          [f(f_count+(1:obj.nlcon{i}.num_cnstr)),G(f_count+(1:obj.nlcon{i}.num_cnstr),obj.nlcon_xind_stacked{i})] = ...

Error in NonlinearProgram/snopt/snopt_userfun (line 1286)
        [f,G] = objectiveAndNonlinearConstraints(obj,x_all);

Error in snoptUserfun (line 6)
  [f,G] = SNOPT_USERFUN(x);

Error in NonlinearProgram/snopt (line 1349)
        [x_free,objval,exitflag,xmul,Fmul] = NonlinearProgramSnoptmex(x0_free, ...

Error in NonlinearProgram/solve (line 946)
          [x,objval,exitflag,infeasible_constraint_name] = snopt(obj,x0);

Error in DynamicalSystem/resolveConstraints (line 325)
      [x,~,exitflag,infeasible_constraint_name] = solve(prog,x0);

Error in DrakeSystem/getInitialState (line 53)
            [x0,success] = resolveConstraints(obj,x0);

Error in FeedbackSystem/getInitialState (line 83)
      x0=encodeX(obj,getInitialState(obj.sys1),getInitialState(obj.sys2));

Error in getInitialState (line 27)

Error in sl (line 15)
[varargout{1:nargout}]=feval(varargin{:});

Error in Simulink.BlockDiagram.getInitialState (line 26)

Error in DynamicalSystem/stateVectorToStructure (line 663)
        obj.structured_x = Simulink.BlockDiagram.getInitialState(mdl);

Error in DynamicalSystem/simulate (line 54)
  x0 = stateVectorToStructure(obj,x0,mdl);

Error in DrakeSystem/simulate (line 490)
        [varargout{:}] = simulate@DynamicalSystem(obj,varargin{:});

Error in runMixedIntegerEnvironment (line 136)
xtraj_sim = simulate(sys,[0 tf],x0);

Error in runMixedIntegerSimpleForest (line 21)
runMixedIntegerEnvironment(r, start, goal, lb, ub, seeds, degree, n_segments, n_regions);
Error using getInitialState (line 27)
Error reported by S-function 'DCSFunction' in
'FeedbackSystem_063587519665577088/DrakeSys':

DrakeSystem S-Function: error MATLAB:err_while_looking_up_function in MATLAB
callback.
See additional debugging information above

Error in sl (line 15)
[varargout{1:nargout}]=feval(varargin{:});

Error in Simulink.BlockDiagram.getInitialState (line 26)

Error in DynamicalSystem/stateVectorToStructure (line 663)
        obj.structured_x = Simulink.BlockDiagram.getInitialState(mdl);

Error in DynamicalSystem/simulate (line 54)
  x0 = stateVectorToStructure(obj,x0,mdl);

Error in DrakeSystem/simulate (line 490)
        [varargout{:}] = simulate@DynamicalSystem(obj,varargin{:});

Error in runMixedIntegerEnvironment (line 136)
xtraj_sim = simulate(sys,[0 tf],x0);

Error in runMixedIntegerSimpleForest (line 21)
runMixedIntegerEnvironment(r, start, goal, lb, ub, seeds, degree, n_segments,
n_regions);
 
```

i'm having it regularly here (but not always??), but it's proving to be a real pain to debug.

rdeits, 2015-01-03:
Ew, no, I haven't seen that one before. Here's a guess: yalmip has an `@ndsdpvar/permute.m` which normally wouldn't be a problem, BUT, `ndsdpvar` has no `classdef` (it's not a full-fledged class, it seems), so perhaps that's confusing things? I've been using yalmip with `tbxmanager`, whereas you're using the yalmip pod, and perhaps those systems set the path in different ways? I'll keep looking...

rdeits, 2015-01-03:
Okay, thanks for the visualizers tip: that's indeed the problem. The issue is that the NullVisualizer requires 12 inputs (6 positions and 6 velocities), but the BotVisualizer expects 6 inputs. Everything seems to work if I just pass the full 12 states into the visualizer.

RussTedrake, 2015-01-03:
that sounds right.  the state coordinate frame has a transform down to the position frame, which gets automatically inserted in the BotVisualizer case.

PR #677 (2015-01-03): fix frame error
closes #673
commit 1db40fe8 (2015-01-03): fix frame error

closes #673
closed 2015-01-03 by RussTedrake