RobotLocomotion/drake #687: When merging welded links, add a frame with the same name

avalenzu, 2015-01-14:
When we merge fixed links we should create a frame at the origin of the link that is merged into its parent. That way `parseBodyOrFrameId(merged_link_name)` will return the correct coordinate system even if the body frame of the merged link is not coincident with the body frame of its parent link.

RussTedrake, 2015-01-15:
Seems like an obvious win, but we want to make sure that it doesn't slow down `doKinematics`.   it could potentially be an option that gets passed into addRobotFromURDF (and passed through to compile).

RussTedrake, 2015-02-01:
@tkoolen - maybe you could add this to your performance testing for the new kinsol?  (just hang a bunch of spurious frames onto some existing bodies and see what sort of performance hit we take)?

tkoolen, 2015-02-01:
I can add it, but I don't iterate over frames in `doKinematics`, just over bodies, so the number of frames attached to a robot's bodies can't really change anything.

RussTedrake, 2015-02-01:
oh.  you're right, of course.  excellent.  then i think we should add kinematic frames without fear.  correct?

tkoolen, 2015-02-01:
Yeah, sounds good to me!

RussTedrake, 2015-04-21:
in c++ we support a "Fixed" joint type.  @tkoolen suggests that we add this to matlab, too, which would resolve this.

RussTedrake, 2016-04-30:
the fixed joint type in c++ is the long term solution.  the matlab classes will soon run through those classes.  resolving this as "won't fix"

closed 2016-04-30 by RussTedrake