RobotLocomotion/drake #689: Pathlcp failure during simulation

nicolapiccinelli, 2015-01-15:
Hi,
i'm against a strange lcppath behaviour. I'm simulating a passive walker with hip actuation. I've build the robot using an urdf definition and i'm using bullet collision with a custom sloped shape acting as ground. The problem appear when the path solver fail successively for 3,4 executions and giving me a segfault from this mex file lcppath.mexa64. I'm using Matlab R2014a on Ubuntu 14.04.

Error using pathlcp (line 163)
Path fails to solve problem

This is the detailed log:

MATLAB crash file:/home/npiccinelli/matlab_crash_dump.28044-1:

```
------------------------------------------------------------------------
       Segmentation violation detected at Thu Jan 15 21:51:42 2015
------------------------------------------------------------------------

Configuration:
  Crash Decoding     : Disabled
  Current Visual     : 0x21 (class 4, depth 24)
  Default Encoding   : UTF-8
  GNU C Library      : 2.19 stable
  MATLAB Architecture: glnxa64
  MATLAB Root        : /usr/local/MATLAB/R2014a
  MATLAB Version     : 8.3.0.532 (R2014a)
  Operating System   : Linux 3.13.0-43-generic #72-Ubuntu SMP Mon Dec 8 19:35:06 UTC 2014 x86_64
  Processor ID       : x86 Family 6 Model 15 Stepping 6, GenuineIntel
  Virtual Machine    : Java 1.7.0_11-b21 with Oracle Corporation Java HotSpot(TM) 64-Bit Server VM mixed mode
  Window System      : The X.Org Foundation (11501000), display :0

Fault Count: 1

Abnormal termination:
Segmentation violation

Register State (from fault):
  RAX = 00000000000000c5  RBX = 0000000000000002
  RCX = 0000000000000404  RDX = 0000000000000404
  RSP = 00007f44b7a57050  RBP = 00007f44b7a57060
  RSI = 0000000000000404  RDI = 00000000000000c5

   R8 = 0000000000000002   R9 = 00007f44a8000078
  R10 = 2530352e3738202c  R11 = 0000000000000025
  R12 = 00007f44b7a577c0  R13 = 00007f4400bdf6a0
  R14 = 00007f44b7a577f8  R15 = 00007f44cabc1560

  RIP = 00007f4406541ebc  EFL = 0000000000010202

   CS = 0033   FS = 0000   GS = 0000

Stack Trace (from fault):
[  0] 0x00007f4406541ebc /home/npiccinelli/tesi/drake-distro/drake/thirdParty/path/lcppath.mexa64+00196284
[  1] 0x00007f440657be8c /home/npiccinelli/tesi/drake-distro/drake/thirdParty/path/lcppath.mexa64+00433804
[  2] 0x00007f440652f7e6 /home/npiccinelli/tesi/drake-distro/drake/thirdParty/path/lcppath.mexa64+00120806
[  3] 0x00007f440652e86c /home/npiccinelli/tesi/drake-distro/drake/thirdParty/path/lcppath.mexa64+00116844
[  4] 0x00007f4406545ec1 /home/npiccinelli/tesi/drake-distro/drake/thirdParty/path/lcppath.mexa64+00212673
[  5] 0x00007f440651475f /home/npiccinelli/tesi/drake-distro/drake/thirdParty/path/lcppath.mexa64+00010079 mexFunction+00001428
[  6] 0x00007f44c1d1d72a     /usr/local/MATLAB/R2014a/bin/glnxa64/libmex.so+00120618 mexRunMexFile+00000090
[  7] 0x00007f44c1d19a94     /usr/local/MATLAB/R2014a/bin/glnxa64/libmex.so+00105108
[  8] 0x00007f44c1d1afb4     /usr/local/MATLAB/R2014a/bin/glnxa64/libmex.so+00110516
[  9] 0x00007f44c1114ad9 /usr/local/MATLAB/R2014a/bin/glnxa64/libmwm_dispatcher.so+00670425 _ZN8Mfh_file11dispatch_fhEiPP11mxArray_tagiS2_+00000697
[ 10] 0x00007f44c01eb20e /usr/local/MATLAB/R2014a/bin/glnxa64/libmwm_interpreter.so+02601486
[ 11] 0x00007f44c01a61d0 /usr/local/MATLAB/R2014a/bin/glnxa64/libmwm_interpreter.so+02318800
[ 12] 0x00007f44c01a81ea /usr/local/MATLAB/R2014a/bin/glnxa64/libmwm_interpreter.so+02327018
[ 13] 0x00007f44c01ab167 /usr/local/MATLAB/R2014a/bin/glnxa64/libmwm_interpreter.so+02339175
[ 14] 0x00007f44c01a926f /usr/local/MATLAB/R2014a/bin/glnxa64/libmwm_interpreter.so+02331247
[ 15] 0x00007f44c01a9ec4 /usr/local/MATLAB/R2014a/bin/glnxa64/libmwm_interpreter.so+02334404
[ 16] 0x00007f44c020730b /usr/local/MATLAB/R2014a/bin/glnxa64/libmwm_interpreter.so+02716427
[ 17] 0x00007f44c1114ad9 /usr/local/MATLAB/R2014a/bin/glnxa64/libmwm_dispatcher.so+00670425 _ZN8Mfh_file11dispatch_fhEiPP11mxArray_tagiS2_+00000697
[ 18] 0x00007f44c03b12b4 /usr/local/MATLAB/R2014a/bin/glnxa64/libmwm_interpreter.so+04461236
[ 19] 0x00007f44c03b2bc9 /usr/local/MATLAB/R2014a/bin/glnxa64/libmwm_interpreter.so+04467657
[ 20] 0x00007f44c03b33fc /usr/local/MATLAB/R2014a/bin/glnxa64/libmwm_interpreter.so+04469756
[... 95 lines cut ...]
[116] 0x00007f44c0dba4b0  /usr/local/MATLAB/R2014a/bin/glnxa64/libmwmcos.so+02016432
[117] 0x00007f44c10c3874 /usr/local/MATLAB/R2014a/bin/glnxa64/libmwm_dispatcher.so+00338036 _ZN13Mfh_MATLAB_fn11dispatch_fhEiPP11mxArray_tagiS2_+00000244
[118] 0x00007f44c0dbb031  /usr/local/MATLAB/R2014a/bin/glnxa64/libmwmcos.so+02019377
[119] 0x00007f44c01eb20e /usr/local/MATLAB/R2014a/bin/glnxa64/libmwm_interpreter.so+02601486
[120] 0x00007f44c01a61d0 /usr/local/MATLAB/R2014a/bin/glnxa64/libmwm_interpreter.so+02318800
[121] 0x00007f44c01a81ea /usr/local/MATLAB/R2014a/bin/glnxa64/libmwm_interpreter.so+02327018
[122] 0x00007f44c01ab167 /usr/local/MATLAB/R2014a/bin/glnxa64/libmwm_interpreter.so+02339175
[123] 0x00007f44c01a926f /usr/local/MATLAB/R2014a/bin/glnxa64/libmwm_interpreter.so+02331247
[124] 0x00007f44c01a9ec4 /usr/local/MATLAB/R2014a/bin/glnxa64/libmwm_interpreter.so+02334404
[125] 0x00007f44c020730b /usr/local/MATLAB/R2014a/bin/glnxa64/libmwm_interpreter.so+02716427
[126] 0x00007f44c1114ad9 /usr/local/MATLAB/R2014a/bin/glnxa64/libmwm_dispatcher.so+00670425 _ZN8Mfh_file11dispatch_fhEiPP11mxArray_tagiS2_+00000697
[127] 0x00007f44c01eb20e /usr/local/MATLAB/R2014a/bin/glnxa64/libmwm_interpreter.so+02601486

This error was detected while a MEX-file was running. If the MEX-file
is not an official MathWorks function, please examine its source code
for errors. Please consult the External Interfaces Guide for information
on debugging MEX-files.

If this problem is reproducible, please submit a Service Request via:
    http://www.mathworks.com/support/contact_us/

A technical support engineer might contact you with further information.

Thank you for your help.
```

RussTedrake, 2015-01-15:
that's no good.  I would try to capture the data being sent into path when it fails, and make sure there is nothing obviously wrong with it (nans, etc).

nicolapiccinelli, 2015-01-15:
That's the output of pathlcp error:
```
Warning: Contact is not supported by the dynamics methods of this class.  Consider using TimeSteppingRigidBodyManipulator or HybridPlanarRigidBodyManipulator 
> In RigidBodyManipulator.RigidBodyManipulator>RigidBodyManipulator.compile at 800
  In Loop at 38
  In LoopRun at 26 
Warning: t=3.646: missed 2 constraints.  resolving lcp. 
> In TimeSteppingRigidBodyManipulator>TimeSteppingRigidBodyManipulator.solveLCP at 489
  In TimeSteppingRigidBodyManipulator>TimeSteppingRigidBodyManipulator.update at 158
  In FeedbackSystem>FeedbackSystem.update at 67
  In DynamicalSystem.simulate at 96
  In DrakeSystem.DrakeSystem>DrakeSystem.simulate at 488
  In Loop at 60
  In LoopRun at 26 
Warning: t=3.646: missed 2 constraints.  resolving lcp. 
> In TimeSteppingRigidBodyManipulator>TimeSteppingRigidBodyManipulator.solveLCP at 489
  In TimeSteppingRigidBodyManipulator>TimeSteppingRigidBodyManipulator.update at 158
  In FeedbackSystem>FeedbackSystem.update at 67
  In DynamicalSystem.simulate at 96
  In DrakeSystem.DrakeSystem>DrakeSystem.simulate at 488
  In Loop at 60
  In LoopRun at 26 
Warning: t=3.647: missed 5 constraints.  resolving lcp. 
> In TimeSteppingRigidBodyManipulator>TimeSteppingRigidBodyManipulator.solveLCP at 489
  In TimeSteppingRigidBodyManipulator>TimeSteppingRigidBodyManipulator.update at 158
  In FeedbackSystem>FeedbackSystem.update at 67
  In DynamicalSystem.simulate at 96
  In DrakeSystem.DrakeSystem>DrakeSystem.simulate at 488
  In Loop at 60
  In LoopRun at 26 
Warning: t=3.647: missed 1 constraints.  resolving lcp. 
> In TimeSteppingRigidBodyManipulator>TimeSteppingRigidBodyManipulator.solveLCP at 489
  In TimeSteppingRigidBodyManipulator>TimeSteppingRigidBodyManipulator.update at 158
  In FeedbackSystem>FeedbackSystem.update at 67
  In DynamicalSystem.simulate at 96
  In DrakeSystem.DrakeSystem>DrakeSystem.simulate at 488
  In Loop at 60
  In LoopRun at 26 
DrakeSystem S-Function: error when calling ''update'' with the following arguments:
  FeedbackSystem with properties:

               sys1: [1x1 TimeSteppingRigidBodyManipulator]
               sys2: [1x1 WalkerController]
            sys1ind: [14x1 double]
            sys2ind: [0x1 double]
               umin: [3x1 double]
               umax: [3x1 double]
    warning_manager: [1x1 WarningManager]

    3.6480

   1.0e+09 *

   -0.0019
    0.0033
   -0.0000
    0.0075
   -0.0000
   -0.0082
   -0.0000
   -1.8532
    3.2587
   -0.0000
    7.4976
    0.0000
   -8.2195
   -0.0000

     0
     0
     0

Error using lcppath
Solution not finite.

Error in pathlcp (line 158)
[status, ttime] = lcppath(n+m, nnzJ, z, l, u, M, q);

Error in TimeSteppingRigidBodyManipulator/solveLCP (line 478)
            z(active) = pathlcp(M(active,active),w(active));

Error in TimeSteppingRigidBodyManipulator/update (line 158)
        [obj,z,Mqdn,wqdn] = solveLCP(obj,t,x,u);

Error in FeedbackSystem/update (line 67)
      if (obj.sys1.getNumDiscStates()) xdn=[xdn;update(obj.sys1,t,x1,sat1(obj,y2+u))]; end

Error in DynamicalSystem/simulate (line 96)
simout = sim(mdl,pstruct);

Error in DrakeSystem/simulate (line 488)
        [varargout{:}] = simulate@DynamicalSystem(obj,varargin{:});

Error in Loop (line 60)
            [xtraj, ~, ~] = simulate(sys,[0 8.0],ci);

Error in LoopRun (line 26)
        [status,counter,increment_e] = Loop(counter + 1,increment_e + delta_e);
Error during simulation loop: 4.Warning: Contact is not supported by the dynamics methods of this class.  Consider using TimeSteppingRigidBodyManipulator or HybridPlanarRigidBodyManipulator 
> In RigidBodyManipulator.RigidBodyManipulator>RigidBodyManipulator.compile at 800
  In Loop at 38
  In LoopRun at 26 
```

RussTedrake, 2015-01-15:
Your state variables are 1e9.  The system has gone unstable, so pathlcp should not really be expected to find a good solution.  We should probably stop the simulation earlier (before things fail), but something has gone very wrong with your robot/controller to get you here.

nicolapiccinelli, 2015-01-15:
Yes i know, i'm exploring some limit robot configuration and i don't have any saturation on the output of the controller, my mistake :). But probably stop the simulation earlier it's a good thing for be able to do batch simulations without supervision or something else...

RussTedrake, 2015-05-11:
will close this one (the big debug spew is complicated my issue parsing on the build servers).

closed 2015-05-11 by RussTedrake