RobotLocomotion/drake #696: drake_viewer only uses one scale factor for the x-, y- and z-directions

steinachim, 2015-01-19:
When displaying meshes from urdf-files (e.g. using RigidBodyManipulator.constructVisualizer), the drake_viewer currently only uses the first scale-factor for all three dimensions and ignores the others. 

This leads to problems if the urdf depends on different scalings in different directions:
```xml
<geometry>
  <mesh filename="test.stl" scale="-1.0 1.0  1.0"/>
</geometry>
```

I do have a patch ready and working, but I'm unsure how to proceed with committing it as I'm both new to github and Drake. A little help would be appreciated ;)

Best wishes,
Achim

RussTedrake, 2015-01-20:
Excellent.  

The easiest way to contribute (and get credit for your contribution!) is to fork the drake repository on your github account webpage.  Then clone your forked copy, make the change, commit the change to your forked copy (e.g. push back to your github fork).  Then back on the website you will create a "pull request" back to RobotLocomotion/drake.   
https://help.github.com/articles/using-pull-requests/

It's a lot of steps the first time, but once it's set up it should be very easy to contribute.  Thanks!

If you'd prefer to just send a one-file change to fix this one, then simply email it to me and I can apply it.

steinachim, 2015-01-20:
Thanks, you've got yourself a pull request!

commit 821bdb86 (2015-01-20): Fix #696: Use all available scale factors in drake_viewer
commit e567413f (2015-01-20): Merge pull request #697 from steinachim/master

Fix #696: Use all available scale factors in drake_viewer
closed 2015-01-20 by RussTedrake