RobotLocomotion/drake #701: LeggedRobotPlanningProblem

manuelli, 2015-01-20:
Attemps to reference the field 'constraints' in obj.support_struct(i) which is non-existent. Maybe it should be changed to obj.constraint_struct.constraint. Could @avalenzu take a look?

-Lucas

The matlab error is:

Reference to non-existent field 'constraints'.

Error in LeggedRobotPlanningProblem/addSupportConstraintsToPlanner (line 412)
        if ~isempty(obj.support_struct(i).constraints)

Error in LeggedRobotPlanningProblem/generateQuasiStaticPlanner (line 304)
      prog = obj.addSupportConstraintsToPlanner(prog);

avalenzu, 2015-01-21:
Sorry about that. We ought to have a test for `LeggedRobotPlanningProblem` in drake. We didn't catch this in DRC because we're adding the quasistatic constraint directly.

Actually, I think that call to `addSupportConstraintsToPlanner` is unnecessary. Can you try commenting out line 304 in `generateQuasiStaticPlanner`?

manuelli, 2015-01-21:
Yes I figured that out yesterday. Commenting out that line seems to work.

commit 508edd17 (2015-01-22): Merge pull request #704 from manuelli/fixLeggedRobotPlanningProblem

resolves #701
commit 9d58e3f2 (2015-01-22): resolves #701
closed 2015-01-22 by RussTedrake