RobotLocomotion/drake #705: Release binaries built on 14.04 or just against newer Boost

stevegray954, 2015-01-21:
I'm using the latest binary release on an Ubuntu 14.04 LTS system and am noticing that the C++ components have been built against an older version of Boost (1.46).  I'm hoping to use this with other code that relies on features in Boost 1.54 (the oldest packaged version available for 14.04). Would it be possible to release binaries that have been built on a 14.04 system?  

./urdf_kin_test: error while loading shared libraries: libboost_filesystem.so.1.46.1: cannot open shared object file: No such file or directory

Thanks,
Steve

RussTedrake, 2015-01-22:
Hi Steven,

Yes - I'll roll a new version today.

- Russ

stevegray954, 2015-01-23:
Thanks Russ!  Looking forward to it.

RussTedrake, 2015-01-24:
Give these a shot and let me know?
https://github.com/RobotLocomotion/drake-distro/releases/tag/0.9.8

- Russ

stevegray954, 2015-01-26:
Thanks for putting this up.  With the previous release, the Drake libraries were automatically found when I ran the test executables.  This time I had to add the library path to LD_LIBRARY_PATH.

testConstraintConstructor, urdf_collision_test, and urdf_kin_test fail; it's something about BulletModel.  Without debug symbols this is all I get from GDB:
```
./urdf_collision_test ../../examples/Atlas/urdf/atlas_minimal_contact.urdf

Program received signal SIGSEGV, Segmentation fault.
0x00002aaaac8b8e14 in DrakeCollision::BulletModel::BulletModel() ()
   from /home/steve/Downloads/drake-precompiled-0.9.8-2fce1a1-linux/drake/pod-build/lib/libdrakeCollision.so
```
I've got libbullet2.82, 2.82-dev, and libbulletcollision2.81. It's the version that gets installed with ROS Indigo + drcsim.

testDrakeGradientUtil and testDrakeGeometryUtil succeed.
testIK, testIKpointwise, and testIKtraj were present in 0.9.7, but were not built in 0.9.8.

Thanks,
Steve

RussTedrake, 2015-01-27:
I will look at why you need the LD_LIBRARY_PATH.  I thought we cleaned that up so you would not.

My guess is that these tests will pass again if you put drake-distro/build/lib on your LD_LIBRARY_PATH *before* the bullet libs installed from ROS.  We have some local mods to bullet, and should be distributing those libs in the build folder.

stevegray954, 2015-01-27:
Even with drake-distro/build/lib first in my LD_LIBRARY_PATH, I'm still getting a segfault in the BulletModel constructor.

Should the modified bullet libs be present, or just built-in to the drake libraries?

avalenzu, 2015-01-27:
The modified bullet libs should be present in drake-distro/build/lib. Can you verify that you have our bullet submodule (it should be at drake-distro/bullet)? 

The reason for the incompatibility with the ROS bullet libraries is that we use a double-precision version of bullet in order to get the accuracy we need for gradient-based optimization.

stevegray954, 2015-01-27:
I don't even have a drake-distro directory.  The precompiled 0.9.8 tarball has build and drake directories at the top level.  The libraries are present in drake/pod-build/lib and in build/lib.

Is the precompiled version not self-contained?

RussTedrake, 2015-01-27:
we call the root directory drake-distro.  so we've been referring to what you're calling build/lib

stevegray954, 2015-01-27:
Makes sense.  Here's what is in drake-distro/build/lib.  Do I need to build drake-distro/bullet from source to get the bullet libraries, or should they be included in the precompiled release?

[image]

RussTedrake, 2015-01-27:
ah.  doh.  i'm slow.

the precompiled source does not include bullet integration yet (because I don't have it building on windows yet -- though it shouldn't be too hard).  unfortunately, this is the expected behavior.

avalenzu, 2015-01-27:
For what it's worth, [compiling drake (and friends) from source](https://github.com/RobotLocomotion/drake/wiki/FromSource) is pretty painless on Ubuntu 14.04 (at least, that's been my experience). I'd recommend it.

stevegray954, 2015-01-27:
I think that's what I'm going to do.  I'm in the process of buying SNOPT right now.

RussTedrake, 2015-01-27:
will close this one

closed 2015-01-27 by RussTedrake

stevegray954, 2015-01-27:
As expected,  the precompiled binaries were fine once I built bullet-pod and made sure those libraries were in the path.  Thanks for the help.

RussTedrake, 2015-01-27:
great.  glad that worked.