RobotLocomotion/drake #708: running SNOPT more than once crashes MATLAB on my windows machines

taylorott, 2015-01-23:
Hi Everyone,

I make use of SNOPT through the Nonlinear program interface. Everything runs smoothly on my mac machines. However, when I try to run the same exact program on one of my windows64 machines, MATLAB generally crashes after SNOPT is called a second time. Here is an example of a program that I can run multiple times on my mac, but has a 60% chance of crashing MATLAB on my windows machine (It generally crashes the 2nd or 3rd time I run it):

function res=snopt_test01()
nlp1 = NonlinearProgram(3);
nlp1 = nlp1.setCheckGrad(true);
nlp1 = nlp1.setSolver('snopt');

%%%%%%%%%%%
% x1^2+4*x2^2<=4
% (x1-2)^2+x2^2<=5
% x1 >= 0
cnstr1 = FunctionHandleConstraint([-inf(2,1);0],[4;5;inf],2,@cnstr1_userfun);
xind1 = [2;3];
nlp1 = nlp1.addConstraint(cnstr1,xind1);
x1 = [1;2;1];

x=nlp1.solve(x1)

cnstr1_userfun(x(xind1'))

end

function [c,dc] = cnstr1_userfun(x)
c = [x(1)^2+4*x(2)^2;(x(1)-2)^2+x(2)^2;x(1)];
dc = [2*x(1) 8*x(2);2*(x(1)-2) 2*x(2);1 0];
end

Is there something wrong in the way that I'm calling SNOPT through the nonlinear program interface? Or is there just something bonked somewhere? I'm not sure what's going on.

Occasionally (but not always), I'll get a crash dump like this (I have no clue what to make of it):

MATLAB crash file:C:\Users\TAYLOR~1\AppData\Local\Temp\matlab_crash_dump.8264-1:

------------------------------------------------------------------------
          std::terminate() detected at Fri Jan 23 13:35:41 2015
------------------------------------------------------------------------

Configuration:
  Crash Decoding     : Disabled
  Default Encoding   : windows-1252
  Graphics card 1    : NVIDIA ( 0x10de ) NVIDIA Quadro K2100M  Version 9.18.13.1269
  Graphics card 2    : Intel Corporation ( 0x8086 ) Intel(R) HD Graphics Family Version 10.18.10.3412
  MATLAB Architecture: win64
  MATLAB Root        : C:\Program Files\MATLAB\R2014b
  MATLAB Version     : 8.4.0.150421 (R2014b)
  Operating System   : Microsoft Windows 7 Professional 
  Processor ID       : x86 Family 6 Model 60 Stepping 3, GenuineIntel
  Software OpenGL    : 0
  Virtual Machine    : Java 1.7.0_11-b21 with Oracle Corporation Java HotSpot(TM) 64-Bit Server VM mixed mode
  Window System      : Version 6.1 (Build 7601: Service Pack 1)

Fault Count: 1

Abnormal termination:
std::terminate()

Register State (captured):
  RAX = 00000000043eea01  RBX = 000000000401b420
  RCX = 000000000401ade0  RDX = 0000000000000000
  RSP = 000000000401a950  RBP = 000000000401bbd0
  RSI = 00000000043dea60  RDI = 00000000043eeab0
 
   R8 = 000007fffffde000   R9 = 000007feeabf0000
  R10 = 00000000043deab0  R11 = 00000000043deab0
  R12 = 00000000fd7f3f01  R13 = 000000000401b9e8
  R14 = 0000000000000000  R15 = 0000000000000000
 
  RIP = 000000000421432a  EFL = 00000206
[... 119 lines cut ...]
[115] 0x00000000777411f5                      C:\Windows\SYSTEM32\ntdll.dll+00332277 KiUserCallbackDispatcher+00000031
[116] 0x00000000774e908a                     C:\Windows\system32\USER32.dll+00102538 PeekMessageW+00000186
[117] 0x00000000774e50fe                     C:\Windows\system32\USER32.dll+00086270 GetKeyState+00000238
[118] 0x00000000774e3a6f                     C:\Windows\system32\USER32.dll+00080495 PeekMessageA+00000087
[119] 0x00000000fab0a26b   C:\Program Files\MATLAB\R2014b\bin\win64\uiw.dll+00172651 UIW_ShowMenuItem+00003995
[120] 0x00000000fab5dce2   C:\Program Files\MATLAB\R2014b\bin\win64\uiw.dll+00515298 UIW_SuspendAttachThreadInput+00000466
[121] 0x0000000100118223 C:\Program Files\MATLAB\R2014b\bin\win64\libmwservices.dll+01212963 services::system_events::PpeDispatchHook::dispatchOne+00032291
[122] 0x0000000100123c95 C:\Program Files\MATLAB\R2014b\bin\win64\libmwservices.dll+01260693 sysq::addProcessPendingEventsUnitTestHook+00005813
[123] 0x0000000100123d50 C:\Program Files\MATLAB\R2014b\bin\win64\libmwservices.dll+01260880 sysq::addProcessPendingEventsUnitTestHook+00006000
[124] 0x0000000100125b75 C:\Program Files\MATLAB\R2014b\bin\win64\libmwservices.dll+01268597 sysq::getCondition+00004629
[125] 0x00000001001274ba C:\Program Files\MATLAB\R2014b\bin\win64\libmwservices.dll+01275066 svWS_ProcessPendingEvents+00000138
[126] 0x00000000fd0a783d   C:\Program Files\MATLAB\R2014b\bin\win64\mcr.dll+00161853 mcr::runtime::setInterpreterThreadToCurrent+00023677
[127] 0x00000000fd0a84aa   C:\Program Files\MATLAB\R2014b\bin\win64\mcr.dll+00165034 mcr::runtime::setInterpreterThreadToCurrent+00026858

This error was detected while a MEX-file was running. If the MEX-file
is not an official MathWorks function, please examine its source code
for errors. Please consult the External Interfaces Guide for information
on debugging MEX-files.

If this problem is reproducible, please submit a Service Request via:
    http://www.mathworks.com/support/contact_us/

A technical support engineer might contact you with further information.

Thank you for your help.

taylorott, 2015-01-26:
The new drake that was posted 3 days ago fixes this problem. I don't know how it did it, but it does. Thanks!

closed 2015-01-26 by taylorott

RussTedrake, 2015-01-26:
Excellent!  Thanks for letting me know.

- Russ