RobotLocomotion/drake #712: is there any way to provide snopt for people compiling from source

RussTedrake, 2015-01-27:
the obvious answer is no.  we are only allowed to redistribute the snopt libraries linked statically into drake (not callable by themselves).  but perhaps we can think of something clever -- for instance having an easy way for someone to download a few precompiled binaries, but otherwise build from source?

brandonjdehart, 2015-01-28:
Not sure if this is helpful or not, but I was running up against issues when trying to use studentsnopt to run Atlas examples (> 300 constraints) so I hacked together a solution to avoid using fmincon.

To get around not having the snopt C libraries, I copied the NonlinearProgramSnopt mex file from the latest drake-distro release into the drake/solvers folder. By also making a one line change to NonlinearProgram.m to check for snopt *or* the mex when choosing the default solver, I no longer have any issues.

Obviously this won't fix anywhere else in the code that might be calling snopt directly, but if mex-ing the calling functions/classes isn't too much work, that could be a way forward?

RussTedrake, 2015-01-28:
excellent.  that's exactly what i have been recommending.  glad to hear it works for you!

if that one line change is something you think everyone should have, it would be great if you could send it in (as a pull request if you want credit for it, or as a copy-paste in this issue if that's easier).

brandonjdehart, 2015-01-28:
Would have been easier to copy-paste, but this is my first time working with GitHub so I figured practice with pull requests would be a good idea. I've made a pull request (#720) to add the change. The mex file for the appropriate platform still needs to be copied into the solvers folder, but at least no code has to change.

RussTedrake, 2015-01-28:
Fantastic.  Will review tonight.  Thanks for contributing.

- Russ

brandonjdehart, 2015-02-12:
Not sure if this is related, but I'm trying to get the runAtlasWalking example to work and ran into some more snopt issues. I changed a couple of files so that the example uses studentsnopt instead of crashing, but now the robot flings its foot into the air after shifting its weight.

I know a previous version of the runAtlasWalking example worked fine with studentsnopt, so I'm wondering if something snopt-dependent has changed in the last couple weeks or if this is the current behaviour in the master branch due to some other issues.

RussTedrake, 2015-02-12:
Hi Brandon,

This is news to me.  Ideally you shouldn’t have to change any lines for the studentsnopt to work now — but admittedly we don’t have any build servers checking that.   Could you post the crash report?  

Anything you can do to help us recreate the problem here would help.

- Russ

brandonjdehart, 2015-02-17:
Hey Russ,

Fixed the runAtlasWalking problem when using student snopt in PR #781. Basically just removed a pointer lookup, as it is already happening down the chain when necessary.

brandonjdehart, 2015-03-10:
It's not built as a pod (as requested in #245), but I've got a wrapper for student snopt setup as a repo (along with gloptipoly3 and sedumi), along with modified forks of drake and drake-distro that use them. Not sure if this will help you get pods going faster or not, but thought I should let you know as I've been building from source using student snopt, sedumi, and gloptipoly3 for a month or so now with no issues so far.

PR #1305 (2015-09-02): adds cmake option to install drake's snopt precompiled binaries
resolves #712 .

uses cmake external projects to grab a tgz of the NonlinearProgramSnoptmex.mex* and put them into the correct directory.  this should make life much better for folks without the snopt license.
closed 2015-09-02 by RussTedrake