RobotLocomotion/drake #713: Tilqr controller uses frames to handle u0 and x0

blandry, 2015-01-27:
The way tilqr currently produces a controller is by creating a linear system with a D matrix only, and adding a transform from the controller's state frame that substracts x0 (desired) and another one from the controller's input frame that adds u0. This works fine if the next thing the user does is create a feedback with the plant and simulate. But it can throw you off if you use the controller for other things, like as a feedback with a real system using LCM coders (the user has to figure out he needs to add u0 and substract x0 himself).

Could this be done better by instead using more than just the D of the controller's system (like having a non-zero y0)?

RussTedrake, 2015-01-27:
i think this is the correct behavior.  You had to label the input frame of the lcm block, too, as either the relative or absolute coordinate frame.  so i think everything should be clear.

RussTedrake, 2015-02-01:
will close this, but reopen if you disagree.

closed 2015-02-01 by RussTedrake

blandry, 2015-02-01:
Sounds good. I wasn't sure if this was actually up for discussion sorry. I
can bring it back up offline if it seems necessary.