RobotLocomotion/drake #717: clean up mex interface to RBM methods

RussTedrake, 2015-01-28:
@tkoolen commented on this in #716.  @avalenzu and I talked about it briefly the other day.  He prefers having named mex entry functions (as opposed to `RigidBodyManipulatormex(int command, ...)` because it keeps things readable, but it makes for a lot of code duplication, longer build times, and larger object files.  I think that our tradition of having a matlab method which encapsulates the different commands is a pretty good alternative.

tkoolen, 2015-01-28:
Ah, actually in #716 I meant utility functions like eigenToMatlab or matlabToEigen, of which copies exist in every mex entry function that I wrote for lack of a good place to put those functions. I'm fine with having separate mex entry functions and agree that it keeps things more readable.

tkoolen, 2015-02-14:
I guess I'm actually OK with it, as long as the single entry function's only job is to switch on the function name input and call the current mex functions (which will no longer be compiled as mex functions). I'm just opposed to having a single, 2000 line file that contains the method name switch as well as the and the input and output processing and RigidBodyManipulator call.

psiorx, 2015-02-20:
@RussTedrake Is this issue still relevant in light of the lessons learned by rigidBodyManipulatormex?  Are there any other ideas in the air for cleaning up the interface?

RussTedrake, 2015-02-20:
i was thinking that we could automate the construction of the mex interface methods.  Perhaps labeling a method in the header file with a MEX_INTERFACE tag, and then having something like lcm-gen which writes the appropriate *mex.cpp method.  that would also reduce bugs and reduce the barrier to making new methods accessible.  but this is not a high priority.

RussTedrake, 2015-04-21:
resolving this as duplicate of #805 and #551

closed 2015-04-21 by RussTedrake