RobotLocomotion/drake #718: Avoid self-collisions during IK trajectory generation

steinachim, 2015-01-28:
Hi everyone,

when using the inverseKinTraj function to generate trajectories, the resulting paths contain self-collisions of the robot that make them infeasible in practice. 

I tried using the AllBodiesClosestDistanceConstraint (with [0, inf] as bounds), but although it worked as expected, the calculation took  6 minutes (as opposed to 5 seconds normally). Is there some middle-ground I am missing? Perhaps some way to ignore bodies that can not collide anyways?

Thanks, 
Achim

RussTedrake, 2015-01-28:
That sounds right.  You can make sure you :  
1) use simple collision geometries (as opposed to meshes) whenever possible
2) set up the collision_filter_groups so that you're only checking for real potential collisions.

- Russ

steinachim, 2015-01-28:
Thanks for clearing that up. Is there by chance some function to automatically import the collision filter groups from an SRDF file? If not, I might be tempted to write one...

Achim

RussTedrake, 2015-01-28:
from urdf yes, but not from sdf yet.

RussTedrake, 2015-01-28:
to be clear.  we do have an sdf parser, but it's just not as fully implemented as the urdf parser (yet).  if you're interested in adding that support, it would be looking at the collision filter group parsing in addRobotFromURDF and reimplementing it in addRobotFromSDF (shouldn't be hard).

steinachim, 2015-01-28:
The thing is, I was not talking about SDF, but SRDF (http://wiki.ros.org/srdf) - our robot models are a curious mixture of URDF and SRDF. Though from what you're saying that is probably unsupported right now.

avalenzu, 2015-01-28:
Another way to speed things up is to use the `MinDistanceConstraint` class instead of `AllBodiesClosestDistance`. The `MinDistanceConstraint` only adds one non-linear constraint to your nonlinear program, rather than the approximately n^2/2 (where n is the number of collision geometries) added by `AllBodiesClosestDistance`. It does this at the cost of being unable to impose an upper bound on the distance between bodies, but in my experience that's not really useful anyway. Usually we just want to ensure no pair of geometries gets closer than some lower bound. The evaluation of the constraint should take about the same amount of time, but the NLP will be much smaller.

steinachim, 2015-01-28:
Thanks, I must have overlooked that one. I'll try that tomorrow.

steinachim, 2015-01-29:
Hm, for some reason using `MinDistanceConstraint` screws up several other constraints (e.g. foot position and orientation) and still tells me it can't be fulfilled... Is there a reason why the minimum distance cannot be set to 0? And does `MinDistanceConstraint` make use of collision groups? Because I don't see an increase in performance (whereas with `AllBodiesClosestDistanceConstraint` there was a significant boost).

avalenzu, 2015-01-29:
Do you have terrain in your `RigidBodyManipulator`?

Re. collision groups: There was a bug in the C++ implementation of `MinDistanceConstraint` that was fixed in de4b9b39dadbb87459a5c59ab92d7ae2024afd58. Prior to that the collision filter groups were not being parsed correctly (or at all really). Can you make sure that you're at least at that revision?

tri-ltyyu, 2019-03-22:
Use case is likely outdated. Closing for now. File new issue if needed.

closed 2019-03-22 by tri-ltyyu

jwnimmer-tri, 2023-07-27:
FYI See https://drake.mit.edu/doxygen_cxx/group__multibody__parsing.html#tag_drake_collision_filter_group for how specify filter groups in a Drake-specific URDF and/or SDFormat.

It's always possible that we'd implement the SDRF input format for Drake at some point in the future, but so far we haven't seen a lot of interest in that file format.

See https://stackoverflow.com/questions/76783635/how-to-allow-collisions-based-on-srdf-file for some related code.