RobotLocomotion/drake #719: Create trajectory with linear interpolation between world points

steinachim, 2015-01-28:
Hi everyone,

I'm looking for a way to receive an IK trajectory that interpolates linearly between world positions wherever possible. 
Using WorldPositionConstraints and WorldQuatConstraints I can reach the specified points at their respective times, but the paths between them are... interesting ;) 

Is there a simple way to accomplish this that I am overlooking? 

Thanks again,
Achim

avalenzu, 2015-01-28:
Are you using `RigidBodyManipulator/inverseKinTraj` or the `InverseKinematicsTrajectory` class? While the latter is part of our new mathematical programming class hierarchy, there's one feature of the former that hasn't been ported over yet which could be useful for you.

The `IKoptions` class, which gets passed to `inverseKinTraj` as its final argument, has a property called `additional_tSamples`. If you add times to this property, any active constraints will be checked at those times. That is, adding additional time samples increases the number of constraints in your NLP, but not the number of decision variables (the state of the robot at the additional times is determined by the spline through the knot points). 

How does this help in your example? If you know the desired start and end positions for the end-effector, you can add a `Point2LineSegDistConstraint` that requires your end-effector to be within some distance of the line between starting and ending positions at all times and then add additional time samples to your `IKoptions` object before calling `inverseKinTraj`. You'll have to figure out the correct trade-off between how close the end-effector stays to the line and how many knot points are necessary to make it do so.

steinachim, 2015-01-29:
I'm currently using `RigidBodyManipulator/inverseKinTraj`, so this might work, thank you. Are there any major disadvantages to using the class interface?

avalenzu, 2015-01-29:
The only major disadvantage at this point is speed. `inverseKinTraj` is was heavily used by the MIT DRC team in the VRC and Trials and has a C++/mex implementation. The new classes don't have a mex implementation yet.

RussTedrake, 2015-01-30:
Thanks for the responses @avalenzu .  Will close this for now -- feel free to reopen if more discussion is needed.

closed 2015-01-30 by RussTedrake