RobotLocomotion/drake #721: Best way to create external wrenches (for disturbance rejection simulations)?

brandonjdehart, 2015-01-29:
I'm wondering what might be the best way to implement external wrenches into a simulation using the Drake toolbox. Specifically, I'm looking for a way to push a biped with a known force on a given link, to test disturbance rejection controllers in simulation before moving them to the real world.

Obviously the controller needs to be defined as some form of DrakeSystem and be connected to the robot through a cascade or feedback connection, as shown in the examples. However, the method of including external wrenches is somewhat unclear. Any help you can provide would be very welcome!

RussTedrake, 2015-01-30:
Hi Brandon,

We often do robustness tests by setting initial velocities in simulation (imagine it as the velocity of the system immediately after a disturbance impact).  

You could also add an actuator to e.g. the floating base joints on the robot if you want to apply, e.g., a linear force to the x coordinate by cascading a force disturbance block with the plant.

I've got a more comprehensive answer for you coming soon in #551, which I intend to implement very soon.

closed 2015-01-30 by RussTedrake

brandonjdehart, 2015-01-30:
Thanks for the feedback, I will definitely try these two options out to see what will work better for my controller testing.

It looks like once #551 is in, the toolbox will definitely allow some more freedom to play with custom forces, actuators, sensors, etc. Let me know if there is anything I can do to contribute.

RussTedrake, 2015-01-30:
Thanks.  #551 has been compared here to open-heart surgery; I think it's on me.  But i'll definitely let you know when there is something to test.

- Russ