RobotLocomotion/drake #723: runDRCPracticeTerrain errors out with  probability

RussTedrake, 2015-01-30:
Site: drake007
Build Name: master-RobotLocomotion-master
Build Server Directory: /tmp/drake-master-16136/drake
CDash: http://kobol.csail.mit.edu/cd/index.php?project=Drake&date=20150130

Most recent commits to drake included in this build:
```
2abbbf7 Merge pull request #720 from brandonjdehart/master
a5f62e8 Merge pull request #714 from scottviteri/master
36c09a4 Merge pull request #1 from rdeits/shift-blocks-slightly
1fa84b9 shift the block piles down slightly to get them all on the ground
ffd9a58 final fix of the DRC terrain heights
```

examples/Atlas/runDRCPracticeTerrain (unknown issue)
```
error_id = 
error_message = Out of range.  Min val: -8.087023e-01 but needed 4Out of
range.  Min val: -1.500000e+00 but needed 
is_segfault = 0
```
```Matlab

                           < M A T L A B (R) >
                 Copyright 1984-2014 The MathWorks, Inc.
                   R2014a (8.3.0.532) 64-bit (glnxa64)
                            February 11, 2014

To get started, type one of these: helpwin, helpdesk, or demo.
For product information, visit www.mathworks.com.

To reproduce this test use rng(289325293,'twister')

[Warning: Disabling autosave for simulink blocks (to avoid generating a
lot of
*.mdl.autosave files in your directory.  If you aren't a regular Simulink
user
and don't want this disabled, comment out this section in addpath_drake.
To disable this warning in the future, add
  warning('off','Drake:DisablingSimulinkAutosave')
to your matlab startup.m] 
[> In addpath_drake at 74
 In addpath_drake at 4] 
Calling addpath_iris
Calling addpath_mosek
Calling addpath_snopt
[Warning: materials not implemented yet] 
[> In WarningManager>WarningManager.warnOnce at 21
 In RigidBodyManipulator.addRobotFromSDF>parseVisual at 323
 In RigidBodyManipulator.addRobotFromSDF>parseLink at 244
 In RigidBodyManipulator.addRobotFromSDF>parseSDFModel at 133
 In RigidBodyManipulator.addRobotFromSDF>parseSDFWorld at 75
 In RigidBodyManipulator.addRobotFromSDF at 44
 In
RigidBodyManipulator.RigidBodyManipulator>RigidBodyManipulator.RigidBodyManipulator at 71
 In runDRCPracticeTerrain at 15
 In fevalPackageSafe at 21] 
Breaking early because start point is no longer contained in polytope
Breaking early because start point is no longer contained in polytope
Breaking early because start point is no longer contained in polytope
Added the lcm jar to your javaclasspath (found via cmake)
Added the lcmgl jar to your javaclasspath (found via cmake)
Calling addpath_gurobi
Calling addpath_spotless

ans =

  -0.8087
  -0.1242
   0.8706
   0.0006
   0.0001
  -0.3781

Breaking early because start point is no longer contained in polytope
Breaking early because start point is no longer contained in polytope
Breaking early because start point is no longer contained in polytope
Breaking early because start point is no longer contained in polytope
Breaking early because start point is no longer contained in polytope
[Warning: The footstep planning problem is infeasible. Returning just the
initial footstep poses] 
[> In Biped.planFootsteps at 98
 In runAtlasWalkingPlanning at 44
 In runDRCPracticeTerrain at 46
 In fevalPackageSafe at 21] 
Error using rangecheck (line 15)
Out of range.  Min val: -8.087023e-01 but needed 4Out of range.  Min val:
-1.500000e+00 but needed 

Error in runDRCPracticeTerrain (line 72)
rangecheck(options.initial_pose(1:2), [4;-1.5], [inf; 1.5]);

Error in fevalPackageSafe (line 21)
 feval(f);

<test_name>examples/Atlas/runDRCPracticeTerrain</test_name>
<error_id></error_id> <error_message>Out of range.  Min val: -8.087023e-01
but needed 4Out of range.  Min val: -1.500000e+00 but needed
</error_message> 

```

RussTedrake, 2015-01-30:
note - this has been failing for some time on the build servers... i was just slow adding it to github as a known issue.

rdeits, 2015-01-30:
I believe #727 may help with this issue (it's possible the build servers were using an out-of-date copy of the terrain map from the cached data).

RussTedrake, 2015-01-30:
the build servers do a clean checkout to a fresh directory every time.  so they should never cache.

jwnimmer-tri, 2017-06-19:
Closing as won't-fix, because #6369 will be removing the relevant software.

If I'm confused and this should in fact stay open, please re-open an clarify the remaining unsolved problem(s) in a new comment here.

closed 2017-06-19 by jwnimmer-tri