RobotLocomotion/drake #728: convert_to_obj doesn't exist

avalenzu, 2015-01-30:
The only occurrence of the string "convert_to_obj" in the repo (as far as I can find) is where it's called in `RigidBodyMesh/convertToObj`.

RussTedrake, 2015-02-01:
it's true.  that's why I check if it exists.  I started the process of automatically converting mesh files for #629 here:
https://github.com/RobotLocomotion/meshConverters
but haven't tested enough to push into the repo.  Do let me know if you try it.

RussTedrake, 2015-02-01:
In case others read this and want to try - meshConverters is pod-compliant, so you would clone it into  your drake-distro and add it to tobuild.txt to get that functionality.  I think it should be pretty easy to add more filetypes.

closed 2015-02-01 by RussTedrake