RobotLocomotion/drake #731: c++ doKinematics, HandC, etc should take VectorXd's (properly templated) instead of double*

RussTedrake, 2015-02-02:
in the RigidBodyManipulator class.

@avalenzu , @hongkai-dai , @tkoolen - can you think of any reason to avoid this?  I think it's just vestigial (those were the first methods we implemented).

tkoolen, 2015-02-02:
No, no real reason I think.

hongkai-dai, 2015-02-02:
No, I don't think we need VectorXd

RussTedrake, 2015-02-02:
i'm proposing that DO use VectorXd so that everything is consistent.

hongkai-dai, 2015-02-02:
I see, I think that should work. And we can also add the size check to make
sure size(q) is consistent with the robot num_dof;

avalenzu, 2015-02-02:
I'm definitely in favor of using VectorXd.

psiorx, 2015-02-16:
I'm also in favor of this.

@RussTedrake Can you please assign this to me?

RussTedrake, 2015-02-16:
will do!  but you might wait until my next pull request comes through (i'm restructuring the mex interface to RBM).

psiorx, 2015-02-16:
@RussTedrake Ok.  I already have getCMM and HandC done.  I'll wait before finishing up doKinematics*().

PR #780 (2015-02-20): templated VectorXd arguments in RBM
Resolves #731
closed 2015-02-20 by RussTedrake