RobotLocomotion/drake #743: invKinTraj and terrain collisions

steinachim, 2015-02-05:
Hi,

I'm trying to get my robot to respect the terrain geometry (currently a simple `RigidBodyHeightMapTerrain`) when planning trajectories using the `inverseKinTraj` function. As far as I understand it, the `MinDistanceConstraint`and `AllBodiesClosestDistanceConstraint` do not take the terrain into account as they only check the robot body distances to each other.
Without further constraints it seems the terrain is simply ignored (leading to collisions) and I have not found anything to actually make it known to the trajectory planner. 

Am I overlooking something or is the terrain only used by the footstep-planning algorithms?

Thanks,
Achim

avalenzu, 2015-02-05:
No, you haven't overlooked anything. We require (mostly) smooth functions for our constraints which is why all of our distance queries look at the distance from one convex geometry to another - concave geometries, like a heightmap can have discontinuous gradients.

Depending on how complicated your terrain is, however, you could incorporate it into the trajectory planners by modeling it as a collection of convex geometries. You can add geometries (instances of one of the child classes of `RigidBodyGeometry`) to the 'world' link of your `RigidBodyManipulator` by calling the `addGeometryToBody` method. Then `MinDistanceConstraint` will check the distance between the robot links and these geometries.

rdeits, 2015-02-05:
If you're interested in exploring this further, I'd suggest the approximate convex decomposition algorithms like this one: http://sourceforge.net/projects/hacd/ which can convert a mesh (like your terrain) into a set of approximately convex pieces which you could use as new collision geometries. I don't think any of us has much experience working with HACD, but I don't know of a better tool for that task.

steinachim, 2015-02-06:
Alright, thank you for clearing that up. I'll try switching to the `RigidBodyGeometry` class where it is truly necessary :)

closed 2015-02-06 by steinachim