RobotLocomotion/drake #747: NonlinearProgramSnoptmex segfaults (at least on mac) when nx=0

RussTedrake, 2015-02-10:
To reproduce:
```
prog = NonlinearProgram(0);
solve(prog,[]);
```
I know it's a ridiculous case, but it currently happens when you call getInitialState(), and therefore resolveConstraints(), on a manipulator with no DOF.  We should definitely not crash.

@hongkai-dai - could you patch this up?

hongkai-dai, 2015-02-10:
> Assigned #747 <https://github.com/RobotLocomotion/drake/issues/747> to

I suppose you do not really want to call SNOPT when the program is empty?
Since in snopt when I call snmema_ to allocate the memory for snopt, if the
variable nx, length of my decision variable, is 0, then it will always
crash.

We can bypass the call to optimization solver if nx = 0, what do you think?

hongkai-dai, 2015-02-13:
Sorry I forgot to mention this issue in my pull request, can we close it?

closed 2015-02-13 by RussTedrake