RobotLocomotion/drake #748: bodies within a single collision filter group generate collision constraints

RussTedrake, 2015-02-10:
Having set up a collision filter group in urdf, I still get collision constraints between some members of the same group.  @avalenzu or @manuelli - could you take a look? 

To reproduce, run the following code:

```
w = warning('off','Drake:RigidBodyManipulator:ReplacedCylinder');
options.terrain = RigidBodyFlatTerrain();
options.floating = true;
r = TimeSteppingRigidBodyManipulator('car.urdf',.01,options);
warning(w);

manip = r.getManipulator();
manip.collision_filter_groups('Russ').members()

manip.body.linkname

[~,~,prog]=resolveConstraints(r,zeros(18,1));
prog.nlcon{1}
```
using the urdf below.  I see
```
10 constraints on 9 variables:
  0.000000 <= non-penetration: body 1 <-> body 2 <= Inf
  0.000000 <= non-penetration: body 1 <-> body 3 <= Inf
  0.000000 <= non-penetration: body 1 <-> body 4 <= Inf
  0.000000 <= non-penetration: body 1 <-> body 5 <= Inf
  0.000000 <= non-penetration: body 1 <-> body 5 <= Inf
  0.000000 <= non-penetration: body 3 <-> body 4 <= Inf
  0.000000 <= non-penetration: body 3 <-> body 5 <= Inf
  0.000000 <= non-penetration: body 3 <-> body 5 <= Inf
  0.000000 <= non-penetration: body 4 <-> body 5 <= Inf
  0.000000 <= non-penetration: body 4 <-> body 5 <= Inf
```
but would not expect to have any constriants between bodies 2-5.  (there is also a duplicate constraint).  I think that only the first four constraints should be there.

Here is the urdf, which you should name `car.urdf`

```

<robot name="car" xmlns="http://drake.mit.edu"
 xmlns:xsi="http://www.w3.org/2001/XMLSchema-instance"
 xsi:schemaLocation="http://drake.mit.edu ../../../../drake-distro/drake/doc/drakeURDF.xsd">

    <material name="tire"><color rgba=".1 .1 .1 0"/></material>
    
    <link name="chassis">
        <inertial>
            <mass value="10"/>
            <inertia ixx="10" iyy="10" izz="10"/>
        </inertial>
        <visual>
            <geometry>
                <box size=".5 .2 .1"/>
            </geometry>
        </visual>
        <collision>
            <geometry>
                <box size=".5 .2 .1"/>
            </geometry>
        </collision>
    </link>
    
    <link name="front_right_wheel">
        <inertial>
            <mass value="1"/>
            <inertia ixx="1" iyy="1" izz="1"/>
        </inertial>        
        <visual>
            <material name="tire"/>
            <origin rpy="pi/2 0 0"/>
[... 225 lines cut ...]
                <color rgba=".9 .9 .9 0"/>
            </material>
            <geometry>
                <box size=".001 .001 .01"/>
            </geometry>
        </visual>    
    </link>

    <joint name="rear_axle" type="continuous">
        <origin xyz="-.2 0 -.05"/>
        <axis xyz="0 1 0"/>
        <parent link="chassis"/>
        <child link="rear_wheels"/>
        <dynamics damping="5"/>
    </joint>

    <collision_filter_group name="Russ">
        <member link="chassis"/>
        <member link="front_right_wheel"/>
        <member link="front_left_wheel"/>
        <member link="rear_wheels"/>
    </collision_filter_group>

</robot>
```

RussTedrake, 2015-02-10:
correction - the two collision constraints between 1 and 5 is the correct behaviour.  Body 5 has two collision elements.

avalenzu, 2015-02-10:
Collision filter groups do not cause collisions between members to be
ignored unless the group is set to ignore itself. This behavior should be
unchanged by @manuelli's recent changes. Adding an
'ignored_collision_filter_group' tag inside the collision filter group tag
should give you the desired behavior.

RussTedrake, 2015-02-10:
Thanks!  Could we document that better? (I looked for it in the drakeURDF.xsd)

But, in that case, why is there no penetration constraint between 2<-->3, 2<-->4, 2<-->5, etc?

RussTedrake, 2015-02-10:
btw - adding `ignored_collision_filter_group` definitely did the trick:
```
5 constraints on 9 variables:
  0.000000 <= non-penetration: body 1 <-> body 2 <= Inf
  0.000000 <= non-penetration: body 1 <-> body 3 <= Inf
  0.000000 <= non-penetration: body 1 <-> body 4 <= Inf
  0.000000 <= non-penetration: body 1 <-> body 5 <= Inf
  0.000000 <= non-penetration: body 1 <-> body 5 <= Inf
```

manuelli, 2015-02-10:
The reason that there are no penetration constraints between 2<-->3,
2<-->4, 2<-->5, etc is that they are adjacent. I believe that 2 is the
parent of 3,4 and 5. Is this a behavior we might want to change? It
wouldn't be too difficult to modify it. I can go into the drakeURDF.xsd
later and add some documentation.

-Lucas

RussTedrake, 2015-02-10:
that makes sense, thanks.  I'm not sure if it should really be the default behavior, but i can see why it might be.  Let's definitely document it better in the xsd file... thanks!

manuelli, 2015-02-10:
For now the collision logic is as follows.

By default all bodies are in checked against each other. Let A and B be the
bodies. They will remain in collision unless

1) Those two bodies are adjacent (unless one of them is the world)

2) A is a member of a collision filter group 'cfg_A', B is a member of
collision filter group 'cfg_B' and A \in
cfg_B.ignored_collision_filter_group.

3) The same as 2) with A and B reversed.

So the behaviour is that everything will be checked against eachother
unless you specifically tell it not to.

-Lucas

avalenzu, 2015-03-30:
This was resolved by #830.

closed 2015-04-27 by RussTedrake