RobotLocomotion/drake #749: Better default behavior for collisions between world and attached bodies

avalenzu, 2015-02-10:
The current collision filtering based on adjacency in the kinematic tree has an exception for the world:

> By default all bodies are in checked against each other. Let A and B be the bodies. They will remain in collision unless
> 
> 1) Those two bodies are adjacent (unless one of them is the world)

(from @manuelli's comment on #748 )

This is good for floating-base robots, since the floating base is a child link of the world. For fixed base robots, however, it can cause an unresolvable collision between the world and the next link.

Proposed resolution:
Check if a link is a floating base when adding it to the collision model. Add a `is_floating` member to `DrakeCollision::Body` that get's checked in `DrakeCollision::Body::collidesWith`. If a link is not a floating base, then adjacency to the world link causes potential collisions with the world link to be ignored.

amcastro-tri, 2016-05-25:
#2299 implements the capability to add collision elements (or bodies with all their collision elements) to "cliques". The use of "clique" instead of "group" is chosen so that this concept does not collide with the concept of groups in the Matlab API. This notion adds more flexibility to specify how things collide (or not).

tri-ltyyu, 2019-03-22:
This is likely specific to RBP. If this is applicable to MBP/SG, please file new issue. Closing for now.

closed 2019-03-22 by tri-ltyyu