RobotLocomotion/drake #763: Compatibility checklist from source or binary installation,

bankh, 2015-02-14:
Drake is an amazing tool to implement complex control algorithms. I really appreciate the efforts to build and develop that. However, I had several issues during the implementation of the binaries and  installation via the available "Installation and Quickstart Guide."
I downloaded available binaries and run the example codes (e.g. runLQR). Afterwards, I got an error such as "Undefined variable "Simulink" or class "Simulink.Mask.create"." AFAIK, that is related to the version of the MATLAB and it is not available the current version I have (MATLAB 2009b). 
I also tried to follow the installation steps for the following settings: Windows 7 Enterprise Ver 6.1 Build 7601 Service Pack 1, Cygwin 64 bits 1.7.34-6, Make 4.0.-2, Matlab 2009b, Microsoft Visual C++ 2008 Express. I have an error on the "Build the Collection" step as 
\n-------------------------------------------
-- lcm
-------------------------------------------
make[1]: *** No rule to make target 'all'.  Stop.
Makefile:22: recipe for target 'all' failed
make: *** [all] Error 2
I think that I would also have problem on the version of the packages or software I am currently using. My question are:
- What is your opinion regarding these errors that I've got both from the binary or source installation of the software?
- Is there a compatibility checklist for available software under Windows 7? 

Thank you,

RussTedrake, 2015-02-14:
We've put effort into making drake work (in many cases) without the simulink dependency, and i'm happy to help if there are places where missing simulink causes a problem.  But i'm afraid we only support matlab >= 2012, because we depend heavily on some object oriented features that are not available in the 2009 version.  You will need to have an updated version to use it.

- Russ

bankh, 2015-02-15:
Thank you for the quick response. I can say that Matlab 2014 is compatible I've tried runLQR. Although I've still some problems with the source based installation. I got some error messages while I was following the Installation and Quickstart guide.
The first bump is the makefile. I've followed the instructions and installed the Cygwin 64 bits 1.7.34-6, Make 4.0.-2. I still get error on the cygwin when I tried to kick the $make. The error is the same as the earlier post 
\n-------------------------------------------
-- lcm
make[1]: *** No rule to make target 'all'. Stop.
Makefile:22: recipe for target 'all' failed
make: *** [all] Error 2
Other possible command $make VERBOSE=1 return similar errors for each of the cloned packages as following:
\n-------------------------------------------
-- gloptipoly3
-------------------------------------------
make[1]: Entering directory '/cygdrive/c/2014-2015/drake-distro/gloptipoly3'
make[1]: *** No rule to make target '1.SILENT'.  Stop.
make[1]: Leaving directory '/cygdrive/c/2014-2015/drake-distro/gloptipoly3'

I didn't see anyone has similar problem. I hope I don't repeat a former issue.

Thank you,
Sinan

RussTedrake, 2015-02-15:
Lcm and gloptipoly are not supported yet on windows.  You should checkout the "precompiled" distro, not "rigidbody"

- Russ

bankh, 2015-02-15:
Let me step back. Is that means, it is not an appropriate way of using Windows for Fromsource installation at all? Because I've another problem that is similar to error #565 (/bin/sh: line 6: /cygdrive/c/Program: No such file or directory) while I am using "rigidbody." The suggestion was using eigen collection to that post #565 . I've tried to use eigen collection but it seems there isn't an eigen collection on the git.
$git clone https://github.com/RobotLocomotion/drake-distro.git --recursive -b eigen

The reason that I insist on Fromsource is I want to use Drake on my ROS platform. I want to see what is happening on the lower-level of drake while I am implementing a control algorithm. But fromSource installation would be irrelevant to that goal so I might be wrong to insist on that. I've seen there is a plan to integrate Moveit based solution to Drake (e.g. Issue #459). I think it is a good idea even though I am not using moveIt node. To that end, one of my limit is the available MATLAB licenses for installation. I need to make a license request for Linux platform and wait for that if it is better to do for FromSource Installation. In the meantime, I will try "precompiled" distro in order to see some results from the examples (e.g. acrobot, quadrotor etc.) as suggested. 

On a slightly separate note, the control-theory part is straightforward if the user couple drake with the Edx or MIT OCW courses -or the user has a non-linear control background. So far, I really like the pipeline of the course -on Edx- and it helped me to brush some of my non-linear control knowledge. However, the installation of Drake seems a little problematic for Windows. On the other hand, I agree regarding the complexity of the software development for the cross-platform systems and hard to keep everyone happy.

I really appreciate your quick response. I hope I would have better questions in advance.
Thank you,
Sinan

RussTedrake, 2015-02-15:
My suggestion is to use
> $git clone https://github.com/RobotLocomotion/drake-distro.git --recursive -b precompiled
> 
you're correct - we've zapped eigen.  (precompiled is eigen + a little more).

Going from source is great.  Developing on windows is not.  Everything will work easily on mac or linux, but the windows setup is definitely a pain.

- Russ

closed 2015-04-21 by RussTedrake