RobotLocomotion/drake #767: Snopt7 problems and external dependensies for ATLAS

Ahmedatcsir, 2015-02-16:
Hi

I'm trying to simulate the ATLAS humanoid on Drake but I can't to get it to work.
I've built the Rgidbody distro (previously got the constructModelMex.mexa64 error), and run some of the other examples without a problem (although I had to set the fortran compiler variable to gfortran in some of the CMakeCache.txt files in the inner folders).

I've installed snopt7 and am now getting the following error when I try executing runAtlasWalking.

******************************
>> runAtlasWalking
Undefined function or variable "A".

Error in snopt (line 176)
[x,F,inform,xmul,Fmul] = snoptmex( solveopt, x, ...

Error in NonlinearProgram/snopt (line 1377)
        [x_free,objval,exitflag,xmul,Fmul] = snopt(x0_free, ...

Error in NonlinearProgram/solve (line 947)
          [x,objval,exitflag,infeasible_constraint_name] = snopt(obj,x0);

Error in DynamicalSystem/resolveConstraints (line 325)
      [x,~,exitflag,infeasible_constraint_name] = solve(prog,x0);

Error in TimeSteppingRigidBodyManipulator/resolveConstraints (line 901)
      [varargout{:}] = resolveConstraints(obj.manip,x0,varargin{:});

Error in Atlas (line 47)
        obj = obj.setInitialState(obj.resolveConstraints(zeros(obj.getNumStates(),1)));

Error in runAtlasWalking (line 26)
r = Atlas(fullfile(getDrakePath,'examples','Atlas','urdf','atlas_minimal_contact.urdf'),options);

*****************************

I got the trial version of snopt7, precompiled mex file. Snopt.m has this definition "function [x,F,inform,xmul,Fmul] = snopt(x, xlow, xupp, xmul, xstate, Flow, Fupp, Fmul, Fstate, userfun, varargin); there are 4 ways of calling snopt with exaclty 10, 12, 15 and 17 input arguments (from the snopt m file). In NonlinearProgram/snopt (line 1377) "[x_free,objval,exitflag,xmul,Fmul] = snopt(x0_free, x_lb_free,x_ub_free, lb,ub, 'snoptUserfun',0,1,Avals,iAfun,jAvar,iGfun_free,jGvar_free);" here you're passing 13 arguments. After a little digging I think you're missing the xmul, xstate, Fmul and Fstate arguments. 

What are the correct inputs for these?

I searched for xmul and Fmul and didn't find it anywhere else in the file, so to see what would happen I set all missing arguments to empty matrices [], and called snopt with all 17 arguments and all in the correct positions. I then got the following error "Maximum row size of 300 exceeded," which is indeed a limitation of the trial version (300*300). Maybe the empty matrices pushes it past its limit? I am not a student and I work at a research institute so I can't get the student edition. Also when I leave out the path to snopt it says the following:
"Error using RigidBodyManipulator/inverseKinBackend (line 33) inverseKinBackend requires SNOPT. Either try install SNOPT, or try using InverseKinematics (capital case) or InverseKinematicsTrajectory (capital case) and set solver to fmincon or ipopt". What do I pass to InverseKinematics or InverseKinematicsTrajectory and how do I set the solver to fmincon?

I have MATLAB's optimization toolbox. Is it possible to get rid of the dependencies on Gurobi which I also have a trial for (which will expire soon and I haven't even tested out the humanoid as yet), as well as snopt (seems like its possible from the above), and only use MATLAB's toolbox for everything? 

I just want to modify the walking algorithms, and implement my own. I don't need speed, high accuracy or high end optimation. What I'm working on is just for simulation and will not be implemented on any real robot.

hongkai-dai, 2015-02-16:
This has something to do with the version of SNOPT, the current trial version of SNOPT requires the input 'xmul', 'xstate','Fmul' and 'Fstate'. We use an older version of SNOPT, that does not require these input arguments. You can set those arguments to zero by yourself.

@RussTedrake , I think it makes sense for us to switch to the newer version of SNOPT in the call to `snopt()`, as that is the most accessible version on the internet. What do you think?

stevegray954, 2015-02-16:
I recently bought SNOPT version 7.2-13, unaware that the interface had changed.  Another difference is the new C/C++ interface that doesn't need f2c at all.  Unfortunately, that also means "snfilewrapper.hh" no longer exists, which poses problems for the C++ IK components.  I'd love it if you guys would support this.  If not, could you tell me what version you use?

Thanks,
Steve

hongkai-dai, 2015-02-16:
The SNOPT we use is 7.2-8.

Ahmedatcsir, 2015-02-17:
Hi

I've set those arguments to 0 and still end up with the Snopt "maximum row size of 300 exceeded" error.

What should I pass to InverseKinematics or InverseKinematicsTrajectory and how do I set the solver to fmincon?

I also want to get rid of the Gurobi dependency, is that possible?

hongkai-dai, 2015-02-17:
You have more than 300 rows of constraints, exceeding the limit of the trail version of SNOPT (I think the student version also has the 300 x 300 limit).

To use `InverseKinematics`, you can refer to the test file in drake/systems/test/testIK.m. It has a very similar interface as `inverseKin`. You will need to give a  bunch of `RigidBodyConstraint` objects, the nominal posture. You can refer to the documentation of `InverseKinematics`. `InverseKinematics

brandonjdehart, 2015-02-18:
I'm also using the trial version of SNOPT, but I've got runAtlasWalking working on my machine, as discussed in #712. Then you don't need to mess with the alternative version of InverseKinematics.

Ahmedatcsir, 2015-02-18:
I stuck the "NonlinearProgramSnoptmex.mexa64" file from a precomiled, binary version of Drake into my built version, I also added the OR check for "NonlinearProgramSnoptmex" via chechDependency and added obj.which_snopt=1; on the default option.  

Original:
function obj = setSolver(obj,solver)
...
elseif(strcmp(solver,'default'))
        if(checkDependency('snopt'))
          obj = obj.setSolver('snopt'); 
        elseif  

modified:
function obj = setSolver(obj,solver)
....
elseif(strcmp(solver,'default'))
        if(checkDependency('snopt') || checkDependency('NonlinearProgramSnoptmex'))
          obj = obj.setSolver('snopt');     
          obj.which_snopt = 1;
        elseif

I got past the 300x300 limitation of the Snopt trial, however now I'm getting an incorrect row dimension error ... 

Error using snoptmex
jAvar has incorrect row dimension   192.  Should be length   304

Error in snsolve (line 176)
[x,F,inform,xmul,Fmul] = snoptmex( solveopt, x, ...

Error in footstepPlanner.nonlinearCollocation (line 178)
  [xstar, fval, ~, ~, exitflag] = snsolve(x0,xlow,xupp,xmul,xstate,    ...

Error in footstepPlanner.alternatingMIQP (line 37)
  [plan, t] = footstepPlanner.nonlinearCollocation(obj, miqp_plan, weights, goal_pos);

Error in Biped/planFootsteps (line 95)
  [plan, solvertime] = options.method_handle(obj, plan, weights, goal_pos);

Error in runAtlasWalking (line 52)
footstep_plan = r.planFootsteps(x0(1:nq), goal_pos);

brandonjdehart, 2015-02-18:
Ah yes, sorry. You also need to modify the snsolve function call in nonlinearCollocation, as the trial version of Snopt has a different calling structure for snsolve than what Drake is using. You will need to move the user function variable ('snoptUserFun') from the last item in the list and place it between the Fstate and ObjAdd variables.

Ahmedatcsir, 2015-02-18:
I made the changes and got more errors....

Original:
[xstar, fval, ~, ~, exitflag] = snsolve(x0,xlow,xupp,xmul,xstate,    ...
               Flow,Fupp,Fmul,Fstate,      ...
               ObjAdd,ObjRow,A_sn(iAndx),iAfun,jAvar,...
               iGfun,jGvar,'snoptUserfun');           

Modified:
[xstar, fval, ~, ~, exitflag] = snsolve(x0,xlow,xupp,xmul,xstate,    ...
               Flow,Fupp,Fmul,Fstate,      ...
               'snoptUserfun',ObjAdd,ObjRow,A_sn(iAndx),iAfun,jAvar,...
               iGfun,jGvar);  

When I ran it agin I got the exact same error as mentioned previously, the function call was missing the xmul, xstate, Fmul and Fstate arguments. 

Undefined function or variable "A".

Error in snopt (line 179)
[x,F,inform,xmul,Fmul] = snoptmex( solveopt, x, ...

Error in RigidBodyManipulator/inverseKinBackend (line 230)
    [x_sol,F,info(i)] = snopt(x0,xlow(:,i),xupp(:,i),Flow,Fupp,'snoptUserfun',0,1,A,iAfun,jAvar,iGfun,jGvar);

I set those to 0 as was suggested. 

Original:
function varargout = inverseKinBackend(obj,mode,t,q_seed,q_nom,varargin)
...
[x_sol,F,info(i)] = snopt(x0,xlow(:,i),xupp(:,i),Flow,Fupp,'snoptUserfun',0,1,A,iAfun,jAvar,iGfun,jGvar);

Modified:
function varargout = inverseKinBackend(obj,mode,t,q_seed,q_nom,varargin)
...
xmul=0;
xstate=0;
Fmul=0; 
Fstate=0;
[x_sol,F,info(i)] = snopt(x0,xlow(:,i),xupp(:,i),xmul,xstate,Flow,Fupp,Fmul,Fstate,'snoptUserfun',0,1,A,iAfun,jAvar,iGfun,jGvar);
    

Another error...

Error using snoptmex
xmul has incorrect row dimension     1.  Should be length    34

Error in snopt (line 179)
[x,F,inform,xmul,Fmul] = snoptmex( solveopt, x, ...

Error in RigidBodyManipulator/inverseKinBackend (line 229)
    [x_sol,F,info(i)] =
    snopt(x0,xlow(:,i),xupp(:,i),xmul,xstate,Flow,Fupp,Fmul,Fstate,'snoptUserfun',0,1,A,iAfun,jAvar,iGfun,jGvar);

I kept changing the size of the xmul, xstate, Fmul and Fstate arguments as suggested by the errors and ended up with...
xmul=zeros(34,1);
xstate=zeros(34,1);
Fmul=zeros(1,1); 
Fstate=zeros(1,1);

Then I got...
 
Error using QPController (line 171)
can't find QPControllermex.  did you build it?

Solved this by copying the QPControllermex from the prebuilt binaries into my build location.

Then got the error...

Invalid MEX-file '/home/miaslab/Documents/drake-distro/drake/systems/controllers/QPControllermex.mexa64':
libdrakeQP.so: cannot open shared object file: No such file or directory

Searched the directories and it was missing, found it in the prebuilt version and copied it over to both the corresponding locations:
/home/miaslab/Downloads/drake-distro-precompiled/drake/pod-build/lib/libdrakeQP.so
/home/miaslab/Downloads/drake-distro-precompiled/build/lib/libdrakeQP.so

And the last error which I can't do anything about...

Invalid MEX-file '/home/miaslab/Documents/drake-distro/drake/systems/controllers/QPControllermex.mexa64':
libgurobi56.so: cannot open shared object file: No such file or directory

I can't find that shared lib anywhere, not in the prebuilt binaries, nor in the Gurobi install folder... 

I'm working with the Drake-Distro-Rigidbody collection.

Ahmedatcsir, 2015-02-18:
I have libgurobi60.so
/opt/gurobi600/linux64/lib/libgurobi.so.6.0.0
/opt/gurobi600/linux64/lib/libgurobi60.so

Could someone email the libgurobi56.so lib to me please...
ashaik@csir.co.za

Where would I stick that file?  Would "/opt/gurobi600/linux64/lib/" work? 

Or indicate how I can build QPControllermex.mexa64, linking it to libgurobi60.so.
As can be seen make is skipping required builds....

brandonjdehart, 2015-02-19:
If you clone the Gurobi pod from RobotLocomotion into your drake-distro folder and add gurobi to the tobuild.txt file (before drake), you should get an appropriate version of gurobi. If you then want to build the QPControllermex, then you will need to run make again in the drake-distro folder. That should build you an appropriate version of the QP Controller mex file.

Ahmedatcsir, 2015-02-20:
I tried something different, I realize interfaces etc could have changed as had happend with Snopt but I tried a simlink and got the simulation to continue. 

Create symlink where libgurobi56.so points to libgurobi60.so
ln -s /path/to/file /path/to/symlink
ln -s /opt/gurobi600/linux64/lib/libgurobi.so.6.0.0 /opt/gurobi600/linux64/lib/libgurobi56.so

It got to a point where the robot starts shifting its mass, but throws up one leg in the air and falls over. I halted the simulation and got the smoothts error (needs the Financial Toolbox) and used post #476 to change the function to smooth but looks like the file is different (post refers to runAtlasWalking.m) mine is in ".../drake-distro/drake/examples/Atlas/+atlasUtil/plotWalkingTraj.m"

ORIGINAL:
function [com, rms_com] = plotWalkingTraj(r, traj, walking_plan_data)
....
x_smooth=smoothts(traj.eval(tts),'e',150);
dtraj = fnder(PPTrajectory(spline(tts,x_smooth)));

MODIFIED:
function [com, rms_com] = plotWalkingTraj(r, traj, walking_plan_data)
....
nx = r.getNumStates();
x_smooth=zeros(nx,length(tts));
x_breaks = traj.eval(tts);
for i=1:nx
  x_smooth(i,:) = smooth(x_breaks(i,:),15,'lowess');
end
dtraj = fnder(PPTrajectory(spline(tts,x_smooth)));

Ran it again and got the following error:
Error using chckxy (line 51)
The data sites should be distinct.

Error in pchip (line 58)
[x,y,sizey] = chckxy(x,y);

Error in atlasUtil.plotWalkingTraj (line 40)
rfoottraj = PPTrajectory(pchip(walking_plan_data.link_constraints(1).ts,...

Error in runAtlasWalking (line 60)
[com, rms_com] = atlasUtil.plotWalkingTraj(r, traj, walking_plan_data);

#712 Also mentions this happening so I modified solve_for_pose as suggested....

% ORIGINAL - FROM MY CLONED COPY
function [pose, q0] = solve_for_pose(constraints, q0)
  constraint_ptrs = {};
  for k = 1:length(constraints)
    constraint_ptrs{end+1} = constraints{k}.mex_ptr;
  end
  [q0, info] = inverseKin(biped,q0,q0,constraint_ptrs{:},ikoptions);
  if info >= 10
    error('Drake:planSwingPitched', 'The foot pose IK problem could not be solved. This should not happen and likely indicates a bug in the constraints.');
  end
  if DEBUG
    v.draw(0, q0);
  end
  kinsol = biped.doKinematics(q0);
  pose = biped.forwardKin(kinsol, swing_body_index, [0;0;0], 1);
end

% MODIFIED VERSION - FROM POST #781
function [pose, q0] = solve_for_pose(constraints, q0)
  [q0, info] = inverseKin(biped,q0,q0,constraints{:},ikoptions);
  if info >= 10
    % try again?
    [q0, info] = inverseKin(biped,q0,q0,constraints{:},ikoptions);
    if info < 10
       warning('Whoa...tried inverseKin again and got a different result...');
    else
       error('Drake:planSwingPitched', 'The foot pose IK problem could not be solved. This should not happen and likely indicates a bug in the constraints.');
    end
   end
   if DEBUG
       v.draw(0, q0);
   end
   kinsol = biped.doKinematics(q0);
   pose = biped.forwardKin(kinsol, swing_body_index, [0;0;0], 1);
end

When I ran it this time I got the next error...

Error using snoptmex
Fmul has incorrect row dimension     1.  Should be length     5

Error in snopt (line 179)
[x,F,inform,xmul,Fmul] = snoptmex( solveopt, x, ...

Error in RigidBodyManipulator/inverseKinBackend (line 229)
    [x_sol,F,info(i)] =
    snopt(x0,xlow(:,i),xupp(:,i),xmul,xstate,Flow,Fupp,Fmul,Fstate,'snoptUserfun',0,1,A,iAfun,jAvar,iGfun,jGvar);
    
Error in RigidBodyManipulator/inverseKin (line 75)
  [q,info,infeasible_constraint]= inverseKinBackend(obj,1,[],q_seed,q_nom,varargin{:},ikoptions);

Error in TimeSteppingRigidBodyManipulator/inverseKin (line 806)
      [varargout{:}]=inverseKin(obj.manip,varargin{:});

Error in Biped/planSwingPitched/solve_for_pose (line 179)
  [q0, info] = inverseKin(biped,q0,q0,constraints{:},ikoptions);

Error in Biped/planSwingPitched (line 226)
[pose, q_latest] = solve_for_pose(constraints, q_latest);

Error in Biped/planZMPTraj (line 92)
  [new_foot_knots, new_zmp_knots] = planSwingPitched(biped, st, sw0, sw1, initial_hold);

Error in Biped/planWalkingZMP (line 17)
[zmptraj,link_constraints, support_times, supports] = planZMPTraj(obj, q0, footstep_plan.footsteps);

Error in runAtlasWalking (line 54)
walking_plan_data = r.planWalkingZMP(x0(1:r.getNumPositions()), footstep_plan);

I set Fmul=zeros(5,1) and Fstate=zeros(5,1) and SUCCESS :) Atlas started walking.... when I halted the simulation however I got the following error :(

Error using valuecheck (line 78)
Values don't match.  Expected
[1.5;0;0.9]
 but got
[0.900322779228072;0;1.01420628053719]

Error in runAtlasWalking (line 68)
valuecheck(com(1:3,end), [navgoal(1:2); 0.9], 0.2);

brandonjdehart, 2015-02-20:
I simply set my xmul and xstate to zeros(size(xlow)) and my Fmul and Fstate to zeros(size(Flow)) everywhere and haven't had any issues. That's also how they are set in the snopt examples.

The valuecheck error is just for the Drake unit testing, so don't worry about it if you halt the simulation part way through. It is simply letting you know the robot didn't end up where it was told to walk to.

Congrats on getting it working!

gnowzil, 2015-02-21:
Hi, I'm the one in charge of all the interfaces to SNOPT.  

Just fyi, xmul, xstate, Fmul, Fstate are the initial guesses for the optimal multipliers and states of the variables and constraints going into SNOPT.  Usually, they are just set to 0, indicating that you don't have any additional information to provide SNOPT.  The interface also expects column vectors (not row vectors), hence the error you received about incorrect row dimension for Fmul.

Ahmedatcsir, 2015-02-23:
Thanks everyone, especially Brandon. 
The toolbox is working fine now, and the Atlas simulation is impressive.

closed 2015-02-23 by Ahmedatcsir

brandonjdehart, 2015-02-23:
Congrats again, happy to hear it!