RobotLocomotion/drake #776: C++ IK functionality

stevegray954, 2015-02-17:
Is the C++ IK functional?  I've been trying it out by modifying testIK.cpp, but it justs adjusts base_x,y,z to satisfy the WorldCoMConstraint and will make sure any joint angles are within bounds.  It seems to ignore WorldFixedBodyPoseConstraint and WorldPositionConstraint when I try to pin the feet to their starting location. Here are my modifications to the code:

```
  VectorXd q0 = VectorXd::Zero(model->num_dof);
  // The state frame of cpp model does not match with the state frame of MATLAB model, since the dofname_to_dofnum is different in cpp and MATLAB
  q0(2) = 0.8;
  model->doKinematics(q0.data());
  Vector3d com_lb = Vector3d::Zero(); 
  Vector3d com_ub = Vector3d::Zero(); 
  com_lb(2) = 0.9;
  com_ub(2) = 1.0;
  WorldCoMConstraint* com_kc = new WorldCoMConstraint(model,com_lb,com_ub,tspan);

 int l_foot;
  int r_foot;
  for(int i = 0;i<model->num_bodies;i++)
  {
    if(model->bodies[i]->linkname == "l_foot")
    {
      l_foot = i;
    }
    else if(model->bodies[i]->linkname == "r_foot")
    {
      r_foot = i;
    }
  }
  WorldFixedBodyPoseConstraint* kc_lfoot = new WorldFixedBodyPoseConstraint(model, l_foot, tspan);
  WorldFixedBodyPoseConstraint* kc_rfoot = new WorldFixedBodyPoseConstraint(model, r_foot, tspan);

  int num_constraints = 3;
  RigidBodyConstraint** constraint_array = new RigidBodyConstraint*[num_constraints];
  constraint_array[0] = com_kc;
  constraint_array[1] = kc_lfoot;
  constraint_array[2] = kc_rfoot;
  IKoptions ikoptions(model);
  VectorXd q_sol(model->num_dof);
  int info;
  vector<string> infeasible_constraint;
  inverseKin(model,q0,q0,num_constraints,constraint_array,q_sol,info,infeasible_constraint,ikoptions);
  printf("INFO = %d\n",info);
  if(info != 1)
  {
    return 1;
  }
  model->doKinematics(q_sol.data());
  for (const auto &jm : model->dof_map) {
    for (const std::pair<std::string, int> &pair : jm) {
      printf("%s [%d] went from  %f to %f\n", pair.first.c_str(), pair.second, q0[pair.second], q_sol[pair.second]);
    }
  }
```

And the results:

```
back_bkx [8] went from  0.000000 to 0.000000
back_bky [7] went from  0.000000 to 0.003193
back_bkz [6] went from  0.000000 to 0.000000
base_pitch [4] went from  0.000000 to 0.001918
base_roll [3] went from  0.000000 to 0.000000
base_x [0] went from  0.000000 to -0.019368
base_y [1] went from  0.000000 to -0.000000
base_yaw [5] went from  0.000000 to 0.000000
base_z [2] went from  0.800000 to 0.539609
l_arm_elx [12] went from  0.000000 to 0.000000
l_arm_ely [11] went from  0.000000 to 0.000000
l_arm_mwx [14] went from  0.000000 to 0.000000
l_arm_shx [10] went from  0.000000 to -0.009245
l_arm_usy [9] went from  0.000000 to 0.000593
l_arm_uwy [13] went from  0.000000 to 0.000000
l_leg_akx [20] went from  0.000000 to 0.000000
l_leg_aky [19] went from  0.000000 to 0.000000
l_leg_hpx [16] went from  0.000000 to 0.000000
l_leg_hpy [17] went from  0.000000 to -0.000217
l_leg_hpz [15] went from  0.000000 to 0.000000
l_leg_kny [18] went from  0.000000 to 0.000134
neck_ay [21] went from  0.000000 to 0.000000
r_arm_elx [25] went from  0.000000 to 0.000000
r_arm_ely [24] went from  0.000000 to 0.000000
r_arm_mwx [27] went from  0.000000 to 0.000000
r_arm_shx [23] went from  0.000000 to 0.009245
r_arm_usy [22] went from  0.000000 to 0.000593
r_arm_uwy [26] went from  0.000000 to 0.000000
r_leg_akx [33] went from  0.000000 to 0.000000
r_leg_aky [32] went from  0.000000 to 0.000000
r_leg_hpx [29] went from  0.000000 to 0.000000
r_leg_hpy [30] went from  0.000000 to -0.000217
r_leg_hpz [28] went from  0.000000 to 0.000000
r_leg_kny [31] went from  0.000000 to 0.000134
```

Am I doing something wrong?  

Also, why does WorldPositionConstraint take a 4xn matrix for pts?  It is supposed to be locations in the frame of the specified rigid body?

Thanks,
Steve

hongkai-dai, 2015-02-17:
Hi Steve,

The `WorldFixedBodyPoseConstraint` is used for `inverseKinTraj`. The usages of this constraint is that we want to require that some body (for example, left foot) does not move in the whole trajectory. In your case, you just want to use `inverseKin` to find a single-shot posture, so `WorldFixedBodyPoseConstraint` is ignored. If you know where you want to pin your left/right foot, then you can call the `WorldPositionConstraint` to constrain the left/right feet to be at the desired position. You will need to compute the starting position first.
```
model->doKinematics(q0.data());
lfoot_position0 = model->forwardKin(l_foot,lfoot_pts,0);
rfoot_position0 = model->forwardKin(r_foot,rfoot_pts,0);
lfoot_cnstr = WorldPositionConstraint(model,l_foot,lfoot_pts,lfoot_position0,lfoot_position0);
rfoot_cnstr = WorldPositionConstraint(model,r_foot,rfoot_pts,rfoot_position0,rfoot_position0);
```

The last row of `lfoot_pts` and `rfoot_pts` are all 1, and the first three rows are the position in the body frame. We use the homogeneous transformation matrix `T` in the kinematics computation. `T` is a 4 x 4 matrix
```
T = [ R p]
    [ 0  1]
```
where `R` is the 3 x 3 body rotation matrix, and `p` is the 3 x 1 vector, for translation of body origin. To compute the world position of a point on the body frame with coordinate `body_position`, we use `[world_position;1] = T*[body_position;1]`, where that `[body_position;1]` is a 4 x 1 vector. That is why we use 4 x n matrix for pts.

Hongkai

stevegray954, 2015-02-17:
Thanks for the super fast answer, Hongkai.  I've got a couple more questions, then hopefully I'll be all set.  In the C++ code, it looks like forwardKin takes in (int body, MatrixXd pts, int rotation type, MatrixXd x);  What should I use for rotation type and what is the matrix x?  Or should I be using forwardKinNew?

Thanks,
Steve

stevegray954, 2015-02-17:
Whoops, I see now that fowardKin is a void, so x must be the result.  And I see the rotation types defined in drakeGeometryUtil.  I guess my only remaining question is whether I should use forwardKinNew?

hongkai-dai, 2015-02-17:
It is better to use `forwardKin` now, the `forwardKinNew` will be ready in a couple of weeks.

hongkai-dai, 2015-02-17:
You are right, `forwardKin` has void return type, `x` is the result. Sorry for my error in the code in the previous post.

If it is all set, could you close this issue? Thanks!

stevegray954, 2015-02-17:
Awesome, that worked.  Thank you!

closed 2015-02-17 by stevegray954