RobotLocomotion/drake #777: Consider using Matlab inputParser class instead of options structs

rdeits, 2015-02-17:
I'm making a new issue to continue the discussion started in #771 about the use of inputParser classes instead of options structs. 

I think the main advantage of the inputParser class is that it enables optional and keyword arguments to be passed to our functions just like they are to built-in Matlab functions, which should improve consistency between our code and the Matlab ecosystem. For example, we currently must do:

```
v = r.constructVisualizer(struct('slider', true));
```

whereas an inputParser would let us write:

```
v = r.constructVisualizer('slider', true)
```

which is more concise and consistent with the rest of Matlab. By default, the inputParser will also accept a struct of arguments, so the original syntax will still work. It also lets us do things like:

```
p.KeepUnmatched = true;
p.parse(varargin{:});
call_my_subfunction(p.Unmatched); % this calls my subfunction, passing in only the arguments that were *not* consumed by the current input parser
```

to hand extra parameters to a sub-function. 

## Performance
On the other hand, there may be a serious performance hit from using the inputParser. @avalenzu notes that inputParser is [slower for checking optional inputs](http://stackoverflow.com/a/26635564/727415). I modified the script given in that post to try comparing it with our `if isfield(options, 'foo')` method, and found that inputParser is substantially slower in that case as well. My script and its results follow. 

```
function t = testOptionsParsing()
    args = {'option1', 'option2'};
    fcns = {
        @foo_isfield ;
        @foo_inputparser
    };

    % parameters sweep
    [f,k] = ndgrid(1:numel(fcns), 0:numel(args));
    f = f(:); k = k(:);

    % test combinations of functions and number of input args
    t = cell(numel(f), 3);
    for i=1:size(t,1)
        t{i,1} = func2str(fcns{f(i)});
        t{i,2} = k(i);
        if k(i) == 0
          t{i,3} = timeit(@() feval(fcns{f(i)}), 2);
        elseif k(i) == 1
          t{i,3} = timeit(@() feval(fcns{f(i)}, struct('option1', 1)), 2);
        elseif k(i) == 2
          t{i,3} = timeit(@() feval(fcns{f(i)}, struct('option1', 1, 'option2', 2)), 2);
        end
    end

    % format results in table
    t = cell2table(t, 'VariableNames',{'func','nargs','time'});
    
    figure(2);
    tt = unstack(t, 'time', 'func');
    names = tt.Properties.VariableNames(2:end);
    bar(tt{:,2:end}.')
    set(gca, 'XTick',1:numel(names), 'XTickLabel',names, 'YGrid','on')
    legend(num2str(tt{:,1}, 'nargin=%d'))
    ylabel('Time [sec]'), xlabel('Functions')
end

function [aa,bb] = foo_isfield(options)
  if nargin < 1, options = struct(); end
  if ~isfield(options, 'option1'); options.option1 = 1; end;
  if ~isfield(options, 'option2'); options.option2 = 1; end;
  aa = options.option1;
  bb = options.option2;
end

function [aa,bb] = foo_inputparser(varargin)
  p = inputParser;
  p.addParamValue('option1', 1);
  p.addParamValue('option2', 1);
  p.parse(varargin{:});
  aa = p.Results.option1;
  bb = p.Results.option2;
end
```

```
>> testOptionsParsing

ans = 

          func           nargs       time   
    _________________    _____    __________

    'foo_isfield'        0         3.614e-05
    'foo_inputparser'    0        0.00015898
    'foo_isfield'        1        5.3772e-05
    'foo_inputparser'    1        0.00019222
    'foo_isfield'        2        4.7881e-05
    'foo_inputparser'    2        0.00018348
```

[image]

rdeits, 2015-02-17:
[abandonmatlab](https://abandonmatlab.wordpress.com/2009/07/31/no-named-arguments/) is also (unsurprisingly) rather pessimistic about the inputParser class

rdeits, 2015-02-18:
@tkoolen suggested a dedicated options class, which can include default values and seems like a good idea. I've benchmarked that here:

[image]

rightmost two columns are as before, but the left columns introduce an options class (a non-handle class on the right, and a handle class on the left). The nargin=0 case includes constructing the default options class at each call to the function.

rdeits, 2015-02-18:
On further discussion, we've decided that handle classes are probably not a good idea here, since they would make it possible for a child function to alter options used in its parent. That's actually already caused bugs for me in the past, because LCM messages are pass-by-reference.

rdeits, 2015-02-18:
We next tried benchmarking against positional arguments with `nargin`. Positional arguments are pretty much always faster, but not necessarily by much. With 8 arguments, we get the following results:

[image]

`foo_args` uses only positional arguments with `nargin`, others are as above.

rdeits, 2015-02-18:
I think that for reliability speed, the options class is the winner. It's not quite as fast as positional arguments, but it's vastly easier to use. 

The primary issue I see is that if the user does not pass the options class in, there's some overhead involved in constructing it within the function. We tried adding a `default_instance` as a constant method of the class, but it gave little to no improvement. 

On the other hand, we can use the `persistent` directive to make the defaults stick around. The code for that looks like:

```
function [aa,bb,cc,dd,ee,ff,gg,hh] = foo_class_persistent(options)
  persistent defaults
  if isempty(defaults)
    defaults = OptionsClassNew();
  end
  if nargin < 1
    options = defaults;
  end
  aa = options.option1;
  bb = options.option2;
  cc = options.option3;
  dd = options.option4;
  ee = options.option5;
  ff = options.option6;
  gg = options.option7;
  hh = options.option8;
end
```

This actually removes the class construction overhead on repeated calls:

[image]

RussTedrake, 2015-02-18:
Thanks Robin.

In all of this, though, I’m seeing that the class approach is not actually faster than the isfield approach (which has the virtue that we’ve been using it everywhere else).   Is there any reason beyond the additional clarity / safety to adopt the class approach?

If not, then I might vote for using isfield for now, and making a general cleaning pass to change to an options class at some future (less time critical) date.

rdeits, 2015-02-18:
The thing that worries me about struct() and isfield() is that a misspelled option will just silently cause the default to be used, and we have no way of catching that.

RussTedrake, 2015-02-18:
totally agree.  but that is a problem that we have everywhere in the code.  

i’m happy to address it, but i think the short term solution would be to have forwardKin conform to the existing standard, and then make a more significant cleanup later?

rdeits, 2015-02-18:
Ok, makes sense.

RussTedrake, 2016-04-30:
realistically, this is not going to happen anytime soon.  (robin has probably already implemented something better in julia, though).  resolving as "won't fix"

closed 2016-04-30 by RussTedrake