RobotLocomotion/drake #801: Gurobi and Matlab's Optimization Toolbox

Ahmedatcsir, 2015-02-23:
Hi 

We already have Matlab's Optimization Toolbox. Could someone tell me if I could get rid of the Gurobi dependency by using Matlab's Optimization Toolbox?

Is Gurobi used because it has specific functions that are not in Matlab or is it due to performance?
How difficult would it be to swap the toolboxes?

Also I'm new to all of this. Could you recommend a good resource to learn about optimization and walking humanoids (MOOC courses, books, etc)?

RussTedrake, 2015-02-23:
There are a few examples where we use gurobi for it's mixed-integer solvers.  I don't believe the matlab toolbox can replace those.  But in many places, we use gurobi just for it's QP solver for the sake of speed, and we have already written a QuadraticProgram function class that would do the right thing (call gurobi if it exists, otherwise fall back to the matlab solvers).  The Atlas examples have been optimized for speed, so we don't use this class due to it's small overhead, but you could probably use it easily enough.

We do have a MOOC version of our graduate robotics class: http://tiny.cc/mitx-underactuated .  We plan to offer it online again starting this june.  The course notes (also linked off the drake documentation page) are here:  http://people.csail.mit.edu/russt/underactuated/underactuated.html

- Russ

rdeits, 2015-02-23:
If you are interested in running some of the mixed-integer optimizations (which are, for example, used in the footstep planner), there do exist free solvers for such problems, but in my experience they tend to be extremely slow, and we don't currently support them in the code (but that could change if it's necessary)

closed 2015-03-22 by RussTedrake

waqas2079, 2015-11-24:
hi, can any one tell me how to optimize my model using gurobi in matlab ?

rdeits, 2015-11-24:
I would recommend starting here: http://www.gurobi.com/documentation/6.0/refman/matlab_api_overview.html