RobotLocomotion/drake #803: Master doesn't compile on my machine

tkoolen, 2015-02-24:
This is when compiling under OSX with Matlab 2014a. Here's the error I'm getting:

```
/Users/twan/code/MIT/drake-distro/drake/systems/plants/contactConstraintsmex.cpp:224:37: error: non-constant-expression cannot be narrowed from type 'int' to 'mwSize' (aka 'unsigned long') in initializer list [-Wc++11-narrowing]
      const mwSize cellDims[] = {1, 2*nk};
                                    ^~~~
/Users/twan/code/MIT/drake-distro/drake/systems/plants/contactConstraintsmex.cpp:224:37: note: override this message by inserting an explicit cast
      const mwSize cellDims[] = {1, 2*nk};
                                    ^~~~
                                    static_cast<mwSize>( )
```

@psiorx Looks like you last touched these lines.
Why didn't the build servers catch this?

tkoolen, 2015-02-24:
My clang --version:
```
Apple LLVM version 6.0 (clang-600.0.56) (based on LLVM 3.5svn)
Target: x86_64-apple-darwin14.0.0
Thread model: posix
```

RussTedrake, 2015-02-24:
the mac build server is way underpowered and is way behind.  (drake001).  the macpro is the intended solution, which I will bring up soon.

psiorx, 2015-02-24:
@tkoolen Looks like clang doesn't like the implicit cast.  I'll fix this in my latest pull request.

tkoolen, 2015-02-24:
I'm a little held up until the fix is in, so if the pull request doesn't get accepted quickly, could you submit a new one fixing just that line?

psiorx, 2015-02-24:
Yes. Sorry didn't see this. https://github.com/RobotLocomotion/drake/pull/806

closed 2015-02-25 by tkoolen