RobotLocomotion/drake #804: RigidBodyManipulator: 'world' link interferes with joint limits

steinachim, 2015-02-24:
Hi,

I'm currently working with a `RigidBodyManipulator` and try to set `PostureConstraint` constraints. 

The problem I have run into is the following: 
`PostureConstraint` uses the joint (body) index for setting joint limits, but reads from  `RigidBodyManipulator.joint_limits_min` and `RigidBodyManipulator.joint_limits_max` for limit checking. Unfortunately the indices are different since `RigidBodyManipulator.body(1)` is the global world link and missing in the limit vectors. Therefore the wrong limits are checked. 

I've tried narrowing it down, but I'm not comfortable trying to fix this. There is too much I might break without noticing... As a work-around I tried to manually add a virtual set of joint limits for the `world` link, but this failed with some internal error because the number of joint limits did not match up with the action number of joints.

Best,
Achim

hongkai-dai, 2015-02-24:
Hi Achim,

`PostureConstraint` uses joint index for setting the joint limits, it has nothing to do with the body index. (body index and joint index are different, the index of the joint that connects body `i` to its parent link, can be obtained by `RigidBodyManipulator.getBody(i).position_num`).  If your robot has a floating body, then there is no joint for the world link, so you do not need to set its joint limits. For example, if I construct my robot as
```
p = RigidBodyManipulator([getDrakePath,'/examples/Atlas/urdf/atlas_minimal_contact.urdf'],struct('floating',true));
```
Then the joint for the first body (world) is `p.getBody(1).jointname` is empty, and its joint index `p.getBody(1).position_num` is 0.

If you want to find the joint limits of the joint that links body `i` to its parent link, it is
`p.joint_limit_min(p.getBody(i).position_num)` and `p.joint_limit_max(p.getBody(i).position_num)`

steinachim, 2015-02-24:
Thank you! From the naming I just assumend `RigidBodyManipulator.getJointId` would give me the joint index I need for the `PostureConstraint`. Now everything works fine :)

closed 2015-02-24 by steinachim