RobotLocomotion/drake #805: Idea: Use function handles instead of branching for choosing method implementation

psiorx, 2015-02-24:
While looking at some of the profile results, I noticed a disproportionate amount of time being spent checking for the presence of mex_model_ptrs and branching to two separate implementations.  This was especially significant in tight inner loops that are called tens of thousands of times during the course of a short simulation.

At first, we thought the profiler might be giving spurious results but commenting out these checks does indeed give a real speed up.  I've thought of some ways around this and put together a performance comparison to evaluate the different methods. 

The big idea is to use function handles to store the current implementation.  By default, the function handle points to the MATLAB implementation.  When a mex model is compiled, the function handle is pointed to the mex implementation.  This swap only happens once.  At runtime, the current implementation is called without the need to ever check for a mex_model_ptr and branch.

But will a function handle call outperform a check-and-branch for tight inner loops?  Let's see.

1. branching - This is how Drake is currently handling mex implementations.  First, a condition is checked(the presence of a mex_model_ptr) and, depending on its value, the code forks into one of two branches.

2. wrapper - The class contains a wrapper method which calls an internal function handle and forwards the objects 'self' property to it.  (i.e. myobj.wrapper(obj) calls myobj.funchandle(obj))

3. handle - Calling the function handle directly, and explicitly passing it an instance of the object as the first argument. (i.e. myobj.funchandle(myobj))

[image]

It looks like the function handle approach has the potential to be a winner.  This could also help to force a uniform interface for mex implementations and their MATLAB counterpart since the function handles would need to have identical signatures for this to work.

rdeits, 2015-02-24:
I like this function handle idea. I'm imagining something like the following pseudo-code:

```
% When debugging:
c = AtlasController('pd_function', function_check(@doPD, @doPDmex));

% When running in the real world
c = AtlasController('pd_function', @doPDmex)
```

rdeits, 2015-02-24:
One side effect of this is that we need to make the mex and matlab implementations take *exactly* the same signature. That's not currently possible because the mex functions all want the model's mex pointer, not the model itself. 

I can easily remedy this by making mex_model_ptr a public attribute of TSRBM (as it is for RBM) which is set whenever TSRBM.compile() is called. That way the mex functions can take in the TSRBM or RBM directly and then retrieve the pointer themselves.

RussTedrake, 2015-02-24:
what about using inheritance.  it could get messy fast, but if we had (in matlab) effectively an RBMmex class which derives from RBM (we could do something to make sure all the existing code calling RBM() constructors get the mex versions as appropriate).  then the matlab call stack would implement all of the function handle logic for us.

rdeits, 2015-02-24:
Yeah, we discussed that too. But I think it's pretty messy. If RBMmex is a subclass of RBM, then we would want the RBM constructor to invoke the RBMmex constructor. But a superclass cannot invoke methods on a subclass. In addition, the RBMmex constructor would have to call obj@RBM(), which would in turn invoke the RBMmex constructor, and thus create an infinite recursive loop (I've just verified this myself).

rdeits, 2015-02-24:
On the other hand, making the mex model pointer a public read-only attribute is already working nicely for me: I now have bodyMotionControl and bodyMotionControlmex which use *exactly* the same signature.

RussTedrake, 2015-02-24:
No, i think we could handle it. Especially since we are planning to separate out the rigidbodytree from the dynamics (so could leave the existing constructors intact)

- Russ

RussTedrake, 2015-02-24:
Great.  I still think we'll need this - since the idea would be to have the mex class overload simply by placing the mex functions into the class directory.

- Russ

psiorx, 2015-02-24:
I like that.  During compile, the presence of the corresponding mex file dictates whether or not the function handle is overridden.

RussTedrake, 2015-02-24:
Exactly.  But i don't think we should do this today - rather synchronize it with the rigidbodytree idea.  I cojld take that next...

- Russ

anoushnajarian, 2015-02-25:
@psiorx, could you please share the code snippets you used to generate the above histogram, and, if possible, the application-level code case that led you to looking at those snippets?  Seeing that will help me comment on this performance optimization further!  Also, what OS are you focusing on for these experiments?

RussTedrake, 2015-02-25:
Thanks Anoush!  We're currently focused on ubuntu 14.04 and Matlab R2014a.

- Russ

RussTedrake, 2015-02-26:
Just so everyone is on the same page -- the reason that we don't currently simple write a mex file with the same name next to the matlab implementation is that we sometimes need to call the matlab implementation:
  case 1: we want to do symbolic processing.  this case could potentially be detected on the mex side by checking isnumeric on the input data -- we could then do a mexCallMATLAB to get back to the m version.  
  case 2: we are trying to do unit test comparisons between the matlab and mex versions of the code to make sure they compute the same things.  since this really only happens in the unit tests, we could potentially handle this particular case by, e.g., temporarily moving the mex file, computing the non-mex version, then putting it back.  But it's not pretty.
  case 3: we do have a few cases in which the c++ is a partial implementation.  e.g. the c++ version computes some gradients, but the matlab version computes more.  This could also be easily detected from mex as well as matlab.  Or would could enforce that this is never the case.

So one possibility might be to have the mex get called by default (if it exists) and then branch back into the matlab if necessary. Is there a way to tell matlab to call the matlab version of a file if the mex file also exists?

psiorx, 2015-02-26:
@anoushnajarian 
The simple class I used to test this is shown below:
[image]

The graph was shown by calling the following lines 100,000 times in a tic-toc block.

    myObj.branching_method()
    myObj.wrapper_method()
    myObj.function_handle(myObj)

The actual implementation of the function1 and function2 was left blank to highlight the overhead of the call itself.

Here's a snippet of actual code that we're using that illustrates the motivation for doing these tests:
[image]

We have code like this all over the place. In tight loops where the branch condition needs to be checked tens of thousands of times over the course of a few seconds of simulation, the overhead really starts to add up.

psiorx, 2015-02-26:
@RussTedrake Regarding case 2 above:

For unit tests that compare functionality, we can also just call the underlying implementations directly instead of going through the function handle on an object.

e.g.
compare_outputs(@functionMatlab, @functionMex, args)

Where compare_outputs is a generalization of valuecheck that works on functions.

RussTedrake, 2015-02-26:
the big question I have is whether we can get away with calling the mex function the SAME name as the matlab function — that’s what would avoid the overhead.

psiorx, 2015-02-26:
Why do they need to be the same name?  The actual function name would be hidden behind the function handle.

Using contactConstraints as an example:

At construction time:

    obj.contactConstraints = @contactConstraintsMATLAB

Then in compile(or whenever we want to start using the mex version):

    obj.contactConstraints = @contactConstraintsMEX

All clients of the interface just call it using 

    obj.contactConstraints()

RussTedrake, 2015-02-26:
My latest proposal was to skip the function handles, and let the matlab call stack figure it out.

- Russ

psiorx, 2015-02-26:
Ah, I see.  I just did a quick test of this and although you can't have a mex file that has the same name, you can actually overload the methods in RBM with function handles in RBMmex.

example:
In RBM there would be a method called contactConstraints implemented in matlab.
In RBMmex < RBM there would be a public function handle property called contactConstraints that points to @contactConstraintsmex.

Any function that gets called on RBMmex that doesn't have an overloading function handle falls back to the method implementation in RBM.

RussTedrake, 2015-02-26:
that's right.  You could also have the same name in the same class, but would have to move the mex file away to call the matlab version.

Did you check the performance of the version you just tried?

- Russ

psiorx, 2015-02-26:
Performance seems just as good as the function_handle method above.  However, there is a problem.  The object reference will not be automatically passed into the function handle call as the first argument like it will for methods.

This is the reason I needed the wrapper originally.  There might be a workaround for this?

RussTedrake, 2015-04-21:
will address this alongside #551 and the overall conversion to c++

closed 2015-04-21 by RussTedrake