RobotLocomotion/drake #815: CDash: can't see test output

tkoolen, 2015-02-26:
Expected behavior:
1. Go to a build on Kitware's CDash for CMake, e.g. https://open.cdash.org/viewTest.php?buildid=3711230
2. Click on the word 'Passed' in the Status column for the first test (kwsys.testIOS)
3. See the following: https://open.cdash.org/testDetails.php?test=133238316&build=3711230, i.e. shows test output.

For our CDash instance:
1. Go to a build on our CDash, e.g. http://kobol.csail.mit.edu/cd/viewTest.php?onlyfailed&buildid=11111
2. Click on the word 'Failed' in the Status column for the first test (examples/Acrobot/runPassiveURDF)
3. See the following: http://kobol.csail.mit.edu/cd/testDetails.php?test=1375686&build=11111, i.e. shows a badly formatted site with no test output shown.

It would be extremely helpful to be able to see the test output for tests that fail intermittently (e.g. #774).

RussTedrake, 2015-02-26:
I also noticed this but haven't debugged it yet...

- Russ

RussTedrake, 2015-03-05:
resolved it by updating cdash.   thanks @mpearrow !

closed 2015-03-05 by RussTedrake

tkoolen, 2015-03-06:
Awesome, thank you!!