RobotLocomotion/drake #838: testForwardJacDotV can fail to meet tolerance

RussTedrake, 2015-03-09:
Site: drake006
Build Name: master-RobotLocomotion-master
Build Server Directory: /tmp/drake-master-22834/drake
CDash: http://kobol.csail.mit.edu/cd/viewTest.php?onlyfailed&buildid=11621

Most recent commits to drake included in this build:
```
4c8ce90 Merge pull request #823 from hongkai-dai/rotate_ABB_base
be59b57 forgot to rotate the base of the convex hull model in my previous
commit
b6a0f07 Merge pull request #821 from hongkai-dai/rotate_ABB_base
5deae73 rotate the ABB base by 90 degrees, fix a bug in ATI sensor urdf
b1fd26c Merge pull request #799 from RussTedrake/no_boost
```

systems/plants/test/testForwardJacDotTimesV (unknown issue)
```
error_id = Drake:ValueCheck
error_message = Values don't match.
   Index       Value       Desired Value
  -------     --------    ---------------
   (4,19)    -0.000000       -0.000000
   (6,19)    -0.000000       -0.000000
  (10,19)    -0.000000       -0.000000
  (12,19)    -0.000000       -0.000000
  (16,19)    -0.000000       -0.000000
  (18,19)    -0.000000       -0.000000
  (22,19)    -0.000000       -0.000000
  (24,19)    -0.000000       -0.000000
  (28,19)    -0.000000       -0.000000
  (30,19)    -0.000000       -0.000000
  (34,19)    -0.000000       -0.000000
  (36,19)    -0.000000       -0.000000
  (40,19)    -0.000000       -0.000000
  (42,19)    -0.000000       -0.000000
   (4,20)    -0.000000       -0.000000
   (6,20)    -0.000000       -0.000000
  (10,20)    -0.000000       -0.000000
  (12,20)    -0.000000       -0.000000
  (16,20)    -0.000000       -0.000000
  (18,20)    -0.000000       -0.000000
  (22,20)    -0.000000       -0.000000
  (24,20)    -0.000000       -0.000000
  (28,20)    -0.000000       -0.000000
  (30,20)    -0.000000       -0.000000
  (34,20)    -0.000000       -0.000000
  (36,20)    -0.000000       -0.000000
  (40,20)    -0.000000       -0.000000
  (42,20)    -0.000000       -0.000000

is_segfault = 0
```
```Matlab

                           < M A T L A B (R) >
                 Copyright 1984-2012 The MathWorks, Inc.
                   R2012b (8.0.0.783) 64-bit (glnxa64)
                             August 22, 2012

To get started, type one of these: helpwin, helpdesk, or demo.
For product information, visit www.mathworks.com.

To reproduce this test use rng(617542319,'twister')

[Warning: Disabling autosave for simulink blocks (to avoid generating a
lot of
*.mdl.autosave files in your directory.  If you aren't a regular Simulink
user
and don't want this disabled, comment out this section in addpath_drake.
To disable this warning in the future, add
  warning('off','Drake:DisablingSimulinkAutosave')
to your matlab startup.m] 
[> In addpath_drake at 74
 In addpath_drake at 4] 
Error using valuecheck (line 78)
Values don't match.
   Index       Value       Desired Value
  -------     --------    ---------------
   (4,19)    -0.000000       -0.000000
   (6,19)    -0.000000       -0.000000
  (10,19)    -0.000000       -0.000000
  (12,19)    -0.000000       -0.000000
  (16,19)    -0.000000       -0.000000
  (18,19)    -0.000000       -0.000000
  (22,19)    -0.000000       -0.000000
  (24,19)    -0.000000       -0.000000
  (28,19)    -0.000000       -0.000000
  (30,19)    -0.000000       -0.000000
  (34,19)    -0.000000       -0.000000
  (36,19)    -0.000000       -0.000000
  (40,19)    -0.000000       -0.000000
  (42,19)    -0.000000       -0.000000
   (4,20)    -0.000000       -0.000000
   (6,20)    -0.000000       -0.000000
  (10,20)    -0.000000       -0.000000
  (12,20)    -0.000000       -0.000000
  (16,20)    -0.000000       -0.000000
  (18,20)    -0.000000       -0.000000
  (22,20)    -0.000000       -0.000000
  (24,20)    -0.000000       -0.000000
  (28,20)    -0.000000       -0.000000
  (30,20)    -0.000000       -0.000000
  (34,20)    -0.000000       -0.000000
  (36,20)    -0.000000       -0.000000
  (40,20)    -0.000000       -0.000000
  (42,20)    -0.000000       -0.000000

Error in testForwardJacDotTimesV>checkMex (line 119)
   valuecheck(dJdot_times_v_mex, dJdot_times_v, 1e-10);

Error in testForwardJacDotTimesV (line 16)
 checkMex(robot, rotation_type);

Error in fevalPackageSafe (line 21)
 feval(f);

<test_name>systems/plants/test/testForwardJacDotTimesV</test_name>
<error_id>Drake:ValueCheck</error_id> <error_message>Values don't match.
   Index       Value       Desired Value
  -------     --------    ---------------
   (4,19)    -0.000000       -0.000000
   (6,19)    -0.000000       -0.000000
  (10,19)    -0.000000       -0.000000
  (12,19)    -0.000000       -0.000000
  (16,19)    -0.000000       -0.000000
  (18,19)    -0.000000       -0.000000
  (22,19)    -0.000000       -0.000000
  (24,19)    -0.000000       -0.000000
  (28,19)    -0.000000       -0.000000
  (30,19)    -0.000000       -0.000000
  (34,19)    -0.000000       -0.000000
  (36,19)    -0.000000       -0.000000
  (40,19)    -0.000000       -0.000000
  (42,19)    -0.000000       -0.000000
   (4,20)    -0.000000       -0.000000
   (6,20)    -0.000000       -0.000000
  (10,20)    -0.000000       -0.000000
  (12,20)    -0.000000       -0.000000
  (16,20)    -0.000000       -0.000000
  (18,20)    -0.000000       -0.000000
  (22,20)    -0.000000       -0.000000
  (24,20)    -0.000000       -0.000000
  (28,20)    -0.000000       -0.000000
  (30,20)    -0.000000       -0.000000
  (34,20)    -0.000000       -0.000000
  (36,20)    -0.000000       -0.000000
  (40,20)    -0.000000       -0.000000
  (42,20)    -0.000000       -0.000000
</error_message> 

```

<matlab_test>strcmp(test_name,'systems/plants/test/testForwardJacDotTimesV') && strcmp(error_id,'Drake:ValueCheck')</matlab_test>

RussTedrake, 2015-03-09:
just realized this should be resolved by #831

RussTedrake, 2015-03-15:
still failing.

RussTedrake, 2015-03-21:
still failing.  

Site: drake001.csail.mit.edu
Build Name: minimal-RobotLocomotion-master
Build Server Directory: /private/tmp/drake-minimal-7406/drake
CDash: http://kobol.csail.mit.edu/cd/viewTest.php?onlyfailed&buildid=11961

Most recent commits to drake included in this build:
```
0bef673 Merge pull request #874 from psiorx/jc-positionConstraintsmex
5305810 Merge pull request #877 from psiorx/jc-fix-ik-tests
e09dd56 Merge pull request #876 from
avalenzu/upstream-hotfix-raycast-margins
9ed9c1a Merge pull request #868 from
tkoolen/testForwardJacDotTimesV-tolerances
79ac36f Merge pull request #865 from rdeits/rd-unified-plans-upstream
```

368 tests run:
278 passed
81 cancelled (with missing dependencies  SDPSolver:9, SNOPT:3, avl:4,
bullet:11, gloptipoly3:3, gurobi:7, iris:3, lcm:26, mosek:2, octomap:1,
rigidbodyconstraint_mex:5, sedumi:3, snopt:1, testQuatmex:1, yalmip:2)
1 timed out
8 failed
 4 <a href="https://github.com/RobotLocomotion/drake/issues/374">known
issue #374</a>: failing imu unit tests
 1 <a href="https://github.com/RobotLocomotion/drake/issues/466">known
issue #466</a>: implement Jdot*v for drakeFunction.kinematic methods
 1 <a href="https://github.com/RobotLocomotion/drake/issues/468">known
issue #468</a>: non-bullet builds can throw logical error in
active_collision parsing in collisionDetect
 2 unknown

systems/plants/test/testForwardJacDotTimesV (unknown issue)
```
error_id = Drake:ValueCheck
error_message = Values don't match.
   Index       Value       Desired Value
  -------     --------    ---------------
    (4,4)     0.000000       -0.000000
    (6,4)     0.000000       -0.000000
   (10,4)     0.000000       -0.000000
   (12,4)     0.000000       -0.000000
   (16,4)     0.000000       -0.000000
   (18,4)     0.000000       -0.000000
   (22,4)     0.000000       -0.000000
   (24,4)     0.000000       -0.000000
   (28,4)     0.000000       -0.000000
   (30,4)     0.000000       -0.000000
   (34,4)     0.000000       -0.000000
   (36,4)     0.000000       -0.000000
   (40,4)     0.000000       -0.000000
   (42,4)     0.000000       -0.000000
   (46,4)     0.000000       -0.000000
   (48,4)     0.000000       -0.000000
   (4,29)    -0.000023       -0.000023
   (6,29)    -0.000023       -0.000023
  (10,29)    -0.000023       -0.000023
  (12,29)    -0.000023       -0.000023
  (16,29)    -0.000023       -0.000023
  (18,29)    -0.000023       -0.000023
  (22,29)    -0.000023       -0.000023
  (24,29)    -0.000023       -0.000023
  (28,29)    -0.000023       -0.000023
  (30,29)    -0.000023       -0.000023
  (34,29)    -0.000023       -0.000023
  (36,29)    -0.000023       -0.000023
  (40,29)    -0.000023       -0.000023
  (42,29)    -0.000023       -0.000023
  (46,29)    -0.000023       -0.000023
  (48,29)    -0.000023       -0.000023
   (4,30)    -0.000023       -0.000023
   (6,30)    -0.000023       -0.000023
  (10,30)    -0.000023       -0.000023
  (12,30)    -0.000023       -0.000023
  (16,30)    -0.000023       -0.000023
  (18,30)    -0.000023       -0.000023
  (22,30)    -0.000023       -0.000023
  (24,30)    -0.000023       -0.000023
  (28,30)    -0.000023       -0.000023
  (30,30)    -0.000023       -0.000023
  (34,30)    -0.000023       -0.000023
  (36,30)    -0.000023       -0.000023
  (40,30)    -0.000023       -0.000023
  (42,30)    -0.000023       -0.000023
  (46,30)    -0.000023       -0.000023
  (48,30)    -0.000023       -0.000023
   (4,31)    -0.000023       -0.000023
   (6,31)    -0.000023       -0.000023
  (10,31)    -0.000023       -0.000023
  (12,31)    -0.000023       -0.000023
  (16,31)    -0.000023       -0.000023
  (18,31)    -0.000023       -0.000023
  (22,31)    -0.000023       -0.000023
  (24,31)    -0.000023       -0.000023
  (28,31)    -0.000023       -0.000023
  (30,31)    -0.000023       -0.000023
  (34,31)    -0.000023       -0.000023
  (36,31)    -0.000023       -0.000023
  (40,31)    -0.000023       -0.000023
  (42,31)    -0.000023       -0.000023
  (46,31)    -0.000023       -0.000023
  (48,31)    -0.000023       -0.000023

is_segfault = 0
```

tkoolen, 2015-03-23:
Ugh, looking into it again. I did 100 random tests earlier and didn't get a failure... I'm glad you implemented the random seed message.

tkoolen, 2015-04-03:
OK, I think I finally have this one. It seems to happen when we're asking for rpy rotation_type and the rotation from end effector to base (the rotation part of x from forwardKin) has pitch close to pi/2.

PR #868 (2015-03-20): Fix testForwardJacDotV can fail to meet tolerance
Fixes #838.
Reduced tolerance a bit. Also lower bounding the absolute tolerance.
PR #955 (2015-04-04): Fix intermittent testForwardJacDotTimesV failures
Fixes #838.
The answer was to reject test inputs that happen to result in singular forwardKin rpy.
closed 2015-04-04 by RussTedrake