RobotLocomotion/drake #841: Compatibility issue with the new SNOPT interface: Error in snwrapper

zamzami, 2015-03-09:
I cannot run my student version of SNOPT with the examples in drake (e.g KneedCompassGait, contactImplicitBrick) even after editing NonlinearProgram.m as suggested in the issue #608 i got a new error in the snwrapper file. 
So my guess is that there is still a passing variable issue to the new interface. Any suggestion on how to fix it ? 
*Thanks in advance*
<pre><code>
Subscript indices must either be real positive integers or logicals.

Error in snwrapper (line 25)
      [obj] = myobj(x);
Error in snopt (line 176)
[x,F,inform,xmul,Fmul] = snoptmex( solveopt, x, ...
Error in NonlinearProgram/snopt (line 1300)
      [x_free,objval,exitflag,xmul,Fmul] = snopt(x0_free, ...
Error in NonlinearProgram/solve (line 907)
          [x,objval,exitflag,infeasible_constraint_name] = snopt(obj,x0);
Error in DirectTrajectoryOptimization/solveTraj (line 171)
      [z,F,info] = obj.solve(z0);
Error in ContactImplicitTrajectoryOptimization/solveTraj (line 269)
      [xtraj,utraj,z,F,info] = solveTraj@DirectTrajectoryOptimization(obj,t_init,traj_init);
Error in testNewTrajOpt (line 87)
[xtraj,utraj,ltraj,ljltraj,z,F,info] = traj_opt.solveTraj(t_init,traj_init);
</pre></code>

**Side Question**: I'm interested in Posa's work  *"A Direct Method for Trajectory Optimization of Rigid Bodies Through Contact"*  The trajectory optimization framework presented using contact forces as decision variables seems to be a promising approach without any à priori references. I was hoping that you could pinpoint out for me the code source for a working example (SpringFlamingo) or the implemented algorithm in Drake. 
Thanks again

hongkai-dai, 2015-03-09:
Can you add a break point at line 176 of snopt to make sure that you are passing in the correct argument? It seems that in snoptmex function your input is an array instead of a function name in the argument that supposed to be the `userfun`

zamzami, 2015-03-10:
I guess you're right, the sequence order of the passing arguments are not the same. But instead of tackling all the different modifications, I chose to solve the problem using the suggestion presented in the issue #767 & #712. And you are right about having SNOPT statically linked in Drake as a Milestone. 

I'm really sorry for the inconvenience but is it possible to get a response for my side question @mposa @RussTedrake    

Thank you again for all your efforts and especially your *open science* spirit.

closed 2015-03-10 by zamzami

RussTedrake, 2015-03-10:
Hi,

The examples you found (kneedcompassgait and the contactimplicitbrick) are both good examples of the results from @mposa's paper.  You should be able to find the trajectory optimization class from that.  Do open up a new issue if you have any specific questions.

- Russ

mposa, 2015-03-10:
Just to second what Russ said--those two examples are the ones we've ported over to the released version of Drake. The stack trace in your first post here shows the proper class (ContactImplicitTrajectoryOptimization) as well.

zamzami, 2015-03-10:
Thank you for your answers. I appreciate it. 
I wasn't sure if the Github issues section was the appropriate place to ask such question (not an enhancement or bug related question). As the Drake community is growing it would be nice to have a Q&A forum as with ROS & Gazebo.