RobotLocomotion/drake #848: eval() and fasteval() treat t as 0 if it's a Matlab `single`

rdeits, 2015-03-12:
try it:

```
K>> traj = PPTrajectory(foh([0, 1], [0, 1]));

K>> traj.eval(0.5)        

ans =

    0.5000

K>> traj.eval(single(0.5))

ans =

  5.2221e-315

K>> traj.eval(double(single(0.5)))

ans =

    0.5000
```

This is made much worse by the fact that `single` + `double` = `single` for some crazy reason, so having one `single` around will break all sorts of things in silent and scary ways. We end up with `single`s when we load LCM types with `float`s.

rdeits, 2015-03-12:
I think the reason this happens is that PPTmex does the following:

```
double *t = mxGetPr(prhs[2]);
```

but if prhs[2] isn't a Matlab `double`, then all sorts of stupid bad stuff can happen (like your `single` becoming a `double` which is very very close to 0). Replacing that with:

```
double t = mxGetScalar(prhs[2]);
```

seems to fix the problem. Is that the right solution?

RussTedrake, 2015-03-12:
Wow.  Yikes.  Good catch!

- Russ

RussTedrake, 2015-03-12:
Yes, looks like that will do the conversion as necessary.  http://www.mathworks.com/help/matlab/apiref/mxgetscalar.html

Arrays will be harder.  Is there a good way to watch/check for this happening in other places?

- Russ

rdeits, 2015-03-12:
We could write a `myGetPr` which checks `mxIsDouble` before assigning the pointer and throws an error otherwise

RussTedrake, 2015-03-12:
It’s a good idea.  I might call it mxGetPrSafe (or getPrSafe) and would put it in drakeUtil.cpp

rdeits, 2015-03-12:
Ah, I was trying to keep the analogy we have in controlUtil with `myGetField` (which is a safer version of `mxGetField`) and `myGetProperty` vs. `mxGetProperty`. Should we try to keep a consistent scheme for these utils:

RussTedrake, 2015-03-12:
Definitely yes to being consistent.  I know controlutil was different and would love to make everything agree.  I'm not wed to either naming convention.

- Russ

PR #865 (2015-03-20): Unified walking, standing, and manip plans
This brings in all of my drake changes from the past week, which brings drake up to the version that was used for walking in sim and in the real world with the new split InstaQP controller. The big change is that WalkingPlanData, WalkingControllerData, ConfigurationTraj, etc have all been unified as `QPLocomotionPlan`. This new class provides useful constructors for getting various types of plans:

* `QPLocomotionPlan.from_standing_state()`
* `QPLocomotionPlan.from_biped_footstep_plan()`
* `QPLocomotionPlan.from_configuration_traj()`

It should also m [... cut]
closed 2015-03-20 by RussTedrake