RobotLocomotion/drake #849: C++ code should store points as 3xn instead of 4xn

RussTedrake, 2015-03-12:
talking to @mattantone , we think that changing the matrix4d transforms to eg isometry3d would allow the multiplication to happen right out of the box (without appending and removing ones everywhere).

tkoolen, 2015-03-12:
I agree. Also, there's the `.homogeneous()` method that adds a one to the end of a vector (without actually making a copy I believe, .homgeneous() returns a special type that implies that the last element is a 1 without storing it in memory). You can also handle 3 x N matrices using .colwise().

See http://eigen.tuxfamily.org/dox-devel/classEigen_1_1VectorwiseOp.html#a7cd745fb2ab988918bff73f0863bb7e1

But it's better still to just use Isometry3d, where the multiplication with a 3xN matrix does the right thing out of the box.

tkoolen, 2015-08-14:
This was fixed in https://github.com/RobotLocomotion/drake/pull/1190.

closed 2015-08-19 by RussTedrake