RobotLocomotion/drake #854: runDRCDoorTask fails w/ probability citing 'Out of range'

RussTedrake, 2015-03-15:
Site: drake006
Build Name: master-pull-avalenzu-akv-cpp-mesh-handling-into-RobotLocomotion-master
Build Server Directory: /tmp/drake-master-15446/drake
Pull Request: #846
CDash: http://kobol.csail.mit.edu/cd/viewTest.php?onlyfailed&buildid=11833

Most recent commits to drake included in this build:

ce7311e Merge branch 'akv-cpp-mesh-handling' of https://github.com/avalenzu/drake
5085531 Address @RussTedrake's comments.
65edc91 Fix alignment of vectors of VisualElements.
7d2d598 Add Eigen align macro to RBody::CollisionElement
1033abe Added DLLEXPORT_RBM to RigidBody::CollisionElement

examples/Atlas/runDRCDoorTask (unknown issue)
```
error_id = 
error_message = Out of range.  Min val: -3.093353e-02 but needed 1Out of range.  Min val: -1 but needed 
is_segfault = 0
                            < M A T L A B (R) >
                  Copyright 1984-2012 The MathWorks, Inc.
                    R2012b (8.0.0.783) 64-bit (glnxa64)
                              August 22, 2012

To get started, type one of these: helpwin, helpdesk, or demo.
For product information, visit www.mathworks.com.

To reproduce this test use rng(671609202,'twister')

[Warning: Disabling autosave for simulink blocks (to avoid generating a lot of
*.mdl.autosave files in your directory.  If you aren't a regular Simulink user
and don't want this disabled, comment out this section in addpath_drake.
To disable this warning in the future, add
   warning('off','Drake:DisablingSimulinkAutosave')
to your matlab startup.m] 
[> In addpath_drake at 74
  In addpath_drake at 4] 
 Calling addpath_iris
 Calling addpath_mosek

MOSEK Version 7.1.0.24 (Build date: 2015-2-20 06:44:16)
Copyright (c) 1998-2015 MOSEK ApS, Denmark. WWW: http://mosek.com
Platform: Linux/64-X86

For help type: help mosekopt.
 Calling addpath_snopt
[Warning: Contact is not supported by the dynamics methods of this class.
Consider using TimeSteppingRigidBodyManipulator or
HybridPlanarRigidBodyManipulator] 
[> In RigidBodyManipulator.RigidBodyManipulator>RigidBodyManipulator.compile at 882
  In RigidBodyManipulator.addRobotFromURDF at 59
  In runDRCDoorTask at 15
  In fevalPackageSafe at 21] 
Breaking early because start point is no longer contained in polytope
 Added the lcm jar to your javaclasspath (found via cmake)
 Added the lcmgl jar to your javaclasspath (found via cmake)
 Calling addpath_gurobi
[Warning: not implemented yet] 
[> In PlanShiftData>PlanShiftData.verifyControllerData at 12
  In ControllerData>ControllerData.ControllerData at 19
  In PlanShiftData>PlanShiftData.PlanShiftData at 8
  In QPWalkingPlan>QPWalkingPlan.QPWalkingPlan at 141
  In QPWalkingPlan>QPWalkingPlan.from_biped_foot_and_zmp_knots at 49
  In QPWalkingPlan>QPWalkingPlan.fro

**This post was going to be too long and has been truncated**
```

<matlab_test>strcmp(testname,'examples/Atlas/runDRCDoorTask') && ~isempty(strfind(error_message,'Out of range'))</matlab_test>

RussTedrake, 2016-01-07:
.

RussTedrake, 2016-04-03:
presumably related to #723 and #855

jwnimmer-tri, 2017-06-19:
Closing as won't-fix, because #6369 will be removing the relevant software.

If I'm confused and this should in fact stay open, please re-open an clarify the remaining unsolved problem(s) in a new comment here.

closed 2017-06-19 by jwnimmer-tri