RobotLocomotion/drake #866: testJointCalibration fails to meet tolerance (w/ probability)

RussTedrake, 2015-03-17:
Site: drake002
Build Name: master-RobotLocomotion-master
Build Server Directory: /tmp/drake-master-11786/drake
CDash: http://kobol.csail.mit.edu/cd/viewTest.php?onlyfailed&buildid=11888

Most recent commits to drake included in this build:
```
2c14a79 Merge pull request #861 from
avalenzu/akv-speed-up-updateCollisionElements
046e1aa Speed up updateCollisionElements
b45a007 Merge pull request #856 from
RussTedrake/compile_old_kinematics_support
526064e Merge pull request #857 from avalenzu/akv-fix-model-test
c767a8a yikes.  fixed bad typo bug in isinf.
```

systems/robotInterfaces/calibration/test/testJointCalibration (unknown
issue)
```
error_id = Drake:ValueCheck
error_message = Values don't match.  Expected 
[10380.9013136671;6165.62389999596;9088.03684010126;13086.155432052;8525.32114997679;14568.8910436703]
but got 
[10421.2044207871;6161.1442288046;9217.89881448806;12762.9312559189;8434.06137558476;14933.3141577657]
is_segfault = 0
```
```Matlab

                           < M A T L A B (R) >
                 Copyright 1984-2014 The MathWorks, Inc.
                   R2014a (8.3.0.532) 64-bit (glnxa64)
                            February 11, 2014

To get started, type one of these: helpwin, helpdesk, or demo.
For product information, visit www.mathworks.com.

To reproduce this test use rng(687500677,'twister')

[Warning: Disabling autosave for simulink blocks (to avoid generating a
lot of
*.mdl.autosave files in your directory.  If you aren't a regular Simulink
user
and don't want this disabled, comment out this section in addpath_drake.
To disable this warning in the future, add
  warning('off','Drake:DisablingSimulinkAutosave')
to your matlab startup.m] 
[> In addpath_drake at 74
 In addpath_drake at 4] 

                               Norm of      First-order 
Iteration        f(x)          step          optimality   CG-iterations
    0            13792.1                       1.7e+03                
    1            5605.38             10       3.45e+03           3
    2            834.088        2.08156            443           3
    3            304.915            2.5            315           3
    4            226.174            2.5            271           3
    5            226.174            2.5            271           2
    6             104.93       0.516522            177           0
    7            74.5014          0.625             61           3
    8            62.2613           1.25           36.2           4
    9             52.809            2.5           25.9           5
   10             52.809            2.5           25.9           8
   11            48.5541          0.625           25.8           0
   12            42.1228           1.25           36.4           7
   13            42.1228            2.5           36.4           7
   14            38.1495          0.625             33           0
   15            30.2118           1.25           30.8           6
   16            25.0835            2.5           51.7           5
   17            18.5538            2.5           41.4           9
   18            18.5538            2.5           41.4           5
[... 117 lines cut ...]
 364.4231

Error using valuecheck (line 78)
Values don't match.  Expected 
[10380.9013136671;6165.62389999596;9088.03684010126;13086.155432052;8525.32114997679;14568.8910436703]
but got 
[10421.2044207871;6161.1442288046;9217.89881448806;12762.9312559189;8434.06137558476;14933.3141577657]

Error in testJointCalibration>testJointStiffnessCalibration (line 72)
valuecheck(k_estimated, k_actual, k_nominal / 20);

Error in testJointCalibration (line 4)
testJointStiffnessCalibration();

Error in fevalPackageSafe (line 21)
 feval(f);

<test_name>systems/robotInterfaces/calibration/test/testJointCalibration</test_name> <error_id>Drake:ValueCheck</error_id> <error_message>Values don't match.  Expected 
[10380.9013136671;6165.62389999596;9088.03684010126;13086.155432052;8525.32114997679;14568.8910436703]
but got 
[10421.2044207871;6161.1442288046;9217.89881448806;12762.9312559189;8434.06137558476;14933.3141577657]</error_message> 

```

<matlab_test>strcmp(test_name,'systems/robotInterfaces/calibration/test/testJointCalibration') && strcmp(error_id,'Drake:ValueCheck')</matlab_test>

RussTedrake, 2015-03-17:
actually, it seems like it's happened on all? of the builds last night for some reason??

avalenzu, 2015-03-17:
Just tested this with master on my machine and it passes. The test sets the `rng` seed to a fixed value, though, which makes this very strange.

psiorx, 2015-03-17:
This also passes on my machine, but the displayed values of the floating states function stops matching the build server's output at iteration 19.

avalenzu, 2015-03-17:
FWIW, my floating states optimization diverges from the build server's at iteration 26 - which means drake002, John's machine, and my machine all get different results.

tkoolen, 2015-03-17:
Seems to only fail on drake002 and drake004
http://kobol.csail.mit.edu/cd/testSummary.php?project=1&name=systems%2FrobotInterfaces%2Fcalibration%2Ftest%2FtestJointCalibration&date=2015-03-17
The annoying thing is that those machines couldn't be more different.
drake002.csail.mit.edu - (slow) ubuntu 12.04LTS with MATLAB 2014a and g++ (virtual machine)
drake004.csail.mit.edu - 32-bit windows 7 with MATLAB 2012a and Visual Studio 2010 Express

It also failed 3 times in a row on my machine (OSX 10.10, 2014a).

tkoolen, 2015-03-17:
Nope, I take that back, it also failed on drake003 and drake006
http://kobol.csail.mit.edu/cd/testSummary.php?project=1&name=systems%2FrobotInterfaces%2Fcalibration%2Ftest%2FtestJointCalibration&date=2015-03-15

tkoolen, 2015-03-17:
Looks like the first time the test started failing with this particular error message was this build:
http://kobol.csail.mit.edu/cd/testDetails.php?test=1626802&build=11841

PR #943 (2015-04-03): Proposed fix for testJointCalibration failures + speedup
Fixes #866.

testJointCalibration started failing reliably on drake004 on March 15. Although the test continued to pass on other machines, the exact values going into the valuecheck also changed on other machines around March 15.

testJointCalibration used a fixed random seed to try to get repeatable test results. We have already established that despite fixing the random seed, behavior can still differ between versions of Matlab and/or the OS.

I believe that a commit around March 15 probably introduced additional calls to one of Matlab [... cut]
closed 2015-04-03 by RussTedrake