RobotLocomotion/drake #871: A particular set of inputs causes a segfault in pathlcp on drake006, but not on my machines

rdeits, 2015-03-18:
This is the cause of the segfault observed in #865 

@psiorx and I stepped through the test and captured the exact inputs to pathlcp, which can be found here: https://www.dropbox.com/s/yje1b1wyrj0k5ce/bad_path_inputs.mat?dl=0

To reproduce, just load that file and do:

```
z = pathlcp(M, w, lb, ub)
```

On drake006 this causes a reliable segfault with the following (truncated) stack trace: 

```
Stack Trace (from fault):
[  0] 0x00007ffd134cc1de    /usr/local/MATLAB/R2012b/bin/glnxa64/libmwfl.so+00516574 _ZN2fl4diag15stacktrace_base7captureERKNS0_14thread_contextEm+000158
[  1] 0x00007ffd134cd4b2    /usr/local/MATLAB/R2012b/bin/glnxa64/libmwfl.so+00521394
[  2] 0x00007ffd134ceffe    /usr/local/MATLAB/R2012b/bin/glnxa64/libmwfl.so+00528382 _ZN2fl4diag13terminate_logEPKcRKNS0_14thread_contextE+000174
[  3] 0x00007ffd127bb093   /usr/local/MATLAB/R2012b/bin/glnxa64/libmwmcr.so+00557203 _ZN2fl4diag13terminate_logEPKcPK8ucontext+000067
[  4] 0x00007ffd127b7b9d   /usr/local/MATLAB/R2012b/bin/glnxa64/libmwmcr.so+00543645
[  5] 0x00007ffd127b9835   /usr/local/MATLAB/R2012b/bin/glnxa64/libmwmcr.so+00550965
[  6] 0x00007ffd127b9a55   /usr/local/MATLAB/R2012b/bin/glnxa64/libmwmcr.so+00551509
[  7] 0x00007ffd127ba0fe   /usr/local/MATLAB/R2012b/bin/glnxa64/libmwmcr.so+00553214
[  8] 0x00007ffd127ba295   /usr/local/MATLAB/R2012b/bin/glnxa64/libmwmcr.so+00553621
[  9] 0x00007ffd10cb1cb0              /lib/x86_64-linux-gnu/libpthread.so.0+00064688
[ 10] 0x00007ffd10950794                    /lib/x86_64-linux-gnu/libc.so.6+00452500 fclose+000004
[ 11] 0x00007ffc35f369c3 /tmp/drake-master-4954/drake/thirdParty/path/lcppath.mexa64+00010691 mexFunction+002040
[ 12] 0x00007ffd0949a69a     /usr/local/MATLAB/R2012b/bin/glnxa64/libmex.so+00112282 mexRunMexFile+000090
[ 13] 0x00007ffd094964e9     /usr/local/MATLAB/R2012b/bin/glnxa64/libmex.so+00095465
[ 14] 0x00007ffd0949733c     /usr/local/MATLAB/R2012b/bin/glnxa64/libmex.so+00099132
[ 15] 0x00007ffd12509a4b /usr/local/MATLAB/R2012b/bin/glnxa64/libmwm_dispatcher.so+00596555 _ZN8Mfh_file11dispatch_fhEiPP11mxArray_tagiS2_+000539
[ 16] 0x00007ffd11d9de56 /usr/local/MATLAB/R2012b/bin/glnxa64/libmwm_interpreter.so+02264662
```

On my machines it works just fine.

rdeits, 2015-03-18:
For a more specific test, these are the exact data which are passed to `lcppath`: https://www.dropbox.com/s/77flkpps9xexlgc/bad_lcppath_inputs.mat?dl=0

To reproduce, do

```
[status, ttime] = lcppath(n+m, nnzJ, z, l, u, M, q);
```

RussTedrake, 2015-03-18:
This error happens even if path is the first and only thing called from matlab?

- Russ

psiorx, 2015-03-18:
Yes, but only on the old ubuntu and old mac machine.  It might be a bug in an older c std lib.

RussTedrake, 2016-05-01:
sadly, i think this is a "won't fix" unless it rears its head again or we get a new lead.

closed 2016-05-01 by RussTedrake