RobotLocomotion/drake #873: URDF transmission parsing doesn't search child elements

mauricefallon, 2015-03-19:
mitdrc/drc recently incorporated the new URDF parsing in drake. It has a knock on effect on the KUKA arm I recently added as well as the Valkyrie. Both previously loaded into ddapp and don't now.

Below are some example transmission blocks:
* Atlas and IRB140 parse fine in ddapp
* Schunk fails as the type field is a FirstChildElement and not an Attribute
* Valkyrie has a very different transmission.

Two questions:
* Should you support the Schunk case?
* Are the tags used in the Valkyrie case common? It came from NASA, I don't know their origin.
I see the ROS pages that Transmission elements aren't well defined. Is there no standard approach?

In passing, the Valkyrie URDF has a PushrodTransmission transmission, which apparently isn't supported yet.

 ( Thanks for the heads up @patmarion . As a work around I can manually edit the URDFs to get by this, but its not the right solution.)

1. Atlas:  ... parses fine
  <transmission name="r_arm_shx_trans" type="pr2_mechanism_model/SimpleTransmission">
    <actuator name="r_arm_shx_motor"/>
    <joint name="r_arm_shx"/>
    <mechanicalReduction>1</mechanicalReduction>
  </transmission>

2. IRB140: ... parses fine
  <transmission name="joint1_trans" type="SimpleTransmission">
    <joint name="joint1"/>
    <actuator name="joint1_motor"/>
    <mechanicalReduction>1</mechanicalReduction>
  </transmission>

3. Schunk Hand: ... loader fails to find the 'type' subelement within the transmission element
  <transmission name="sdh_knuckle_trans">
    <type>transmission_interface/SimpleTransmission</type>
    <joint name="sdh_finger_21_joint"/>
    <actuator name="sdh_knuckle_motor">
      <hardwareInterface>EffortJointInterface</hardwareInterface>
      <mechanicalReduction>1</mechanicalReduction>
    </actuator>
  </transmission>

4. Valkyrie: ... loader cannot find joint name. This transmission element is quite different
  <transmission name="LeftKneeExtensorTransmission" type="loadable_transmissions/SimpleTransmission">
    <mechanicalReduction>1</mechanicalReduction>
    <jointOffset>0</jointOffset>
    <interface type="ActuatorToJointStateInterface">
      <joint data="data" name="l_leg_kny"/>
      <actuator data="data" name="left_leg/j4"/>
    </interface>
    <interface type="JointToActuatorPositionInterface">
      <joint data="positionCommand" name="l_leg_kny"/>
      <actuator data="positionCommand" name="left_leg/j4"/>
    </interface>
    <interface type="JointToActuatorVelocityInterface">
      <joint data="velocityCommand" name="l_leg_kny"/>
      <actuator data="velocityCommand" name="left_leg/j4"/>
    </interface>
    <interface type="JointToActuatorEffortInterface">
      <joint data="effortCommand" name="l_leg_kny"/>
      <actuator data="effortCommand" name="left_leg/j4"/>
    </interface>
    <interface type="JointToActuatorStateInterface">
      <joint data="data" name="l_leg_kny"/>
      <actuator data="data" name="left_leg/j4"/>
    </interface>
    <interface type="ActuatorToJointPositionInterface">
      <joint data="positionCommand" name="l_leg_kny"/>
      <actuator data="positionCommand" name="left_leg/j4"/>
    </interface>
    <interface type="ActuatorToJointVelocityInterface">
      <joint data="velocityCommand" name="l_leg_kny"/>
      <actuator data="velocityCommand" name="left_leg/j4"/>
    </interface>
    <interface type="ActuatorToJointEffortInterface">
      <joint data="effortCommand" name="l_leg_kny"/>
      <actuator data="effortCommand" name="left_leg/j4"/>
    </interface>
  </transmission>

RussTedrake, 2015-03-19:
i looked at the valkyrie transmission tag w/ andres yesterday and didn't recognize it.  
i hadn't seen the schunk failure.  
you're not actually using the transmission tag for anything in ddapp (yet).  as a short term solution, what about if we take a boolean option for whether or not we parse the transmission elements, which you will set to false?  (we're about to add similar options for parsing visual elements and collision elements, like the matlab parser has).

mauricefallon, 2015-03-19:
I think the appropriate solution is to have the urdf provide valid transmission tags.
Having a bit that needs to be flipped for each robot on the application layer would be problematic.
Instead, in the short term, I'm happy to fix the urdf to match drake's expectations.

I think Valkyrie's tag isn't standard and appending the joint name wouldn't conflict.

For the Schunk tag, I think its on you to parse that?

RussTedrake, 2015-03-19:
Can you point me to the schunk urdf?

- Russ

mauricefallon, 2015-03-19:
It's here:
https://github.com/drcbot/drc/blob/mfallon-kuka-bridge/software/models/common_components/hand_factory/schunk_hand.urdf
Currently not merged into mitdrc/drc. Pat will merge today I expect.

This version includes my hacks to allow it to load. Before it was:
<transmission name="sdh_finger_12_trans">
    <type>transmission_interface/SimpleTransmission</type>
    <joint name="sdh_finger_12_joint"/>
    <actuator name="sdh_finger_12_motor">
      <hardwareInterface>EffortJointInterface</hardwareInterface>
      <mechanicalReduction>1</mechanicalReduction>
    </actuator>
  </transmission>

Not it is:
<transmission name="sdh_finger_12_trans" type="SimpleTransmission">
    <type>transmission_interface/SimpleTransmission</type>
    <joint name="sdh_finger_12_joint"/>
    <actuator name="sdh_finger_12_motor">
      <hardwareInterface>EffortJointInterface</hardwareInterface>
      <mechanicalReduction>1</mechanicalReduction>
    </actuator>
  </transmission>

The original xacro source:
https://github.com/drcbot/drc/blob/mfallon-kuka-bridge/software/models/common_components/schunk_description/urdf/sdh/

Upsteam maintainers - perhaps?
https://github.com/ipa320/schunk_modular_robotics/tree/indigo_dev/schunk_description/urdf/sdh

RussTedrake, 2015-03-19:
Ok, i confirm that the urdf transmission spec has changed :

http://wiki.ros.org/urdf/XML/Transmission

And that the schunk urdf is compliant.  Will add support (soonish)

- Russ

mauricefallon, 2015-03-19:
That's great, thanks!

RussTedrake, 2015-04-21:
i believe this is resolved with #998

closed 2015-04-21 by RussTedrake