RobotLocomotion/drake #889: testCOM can fail on minimal builds

RussTedrake, 2015-03-23:
Site: thrashcan (aka drake005)
Build Name: minimal-RobotLocomotion-master
Build Server Directory: /tmp/drake-minimal-17225/drake
CDash: http://kobol.csail.mit.edu/cd/viewTest.php?onlyfailed&buildid=wrong
site

Most recent commits to drake included in this build:
```
16c0d92 Merge pull request #887 from
psiorx/jc-fix-testPositionConstraintsmex
84aff10 Merge pull request #814 from tkoolen/cmm-com-cleanup
c15584e Revert change to getCOM.
b3076fa Got rid of robotnum logic that is already handled in
isBodyPartOfRobot.
0320955 Fix tests, robotnum logic.
```

systems/plants/test/testCOM (unknown issue)
```
error_id = MATLAB:nonExistentField
error_message = Reference to non-existent field 'dTdqdot'.
is_segfault = 0
```
```Matlab

                           < M A T L A B (R) >
                 Copyright 1984-2014 The MathWorks, Inc.
                   R2014a (8.3.0.532) 64-bit (glnxa64)
                            February 11, 2014

To get started, type one of these: helpwin, helpdesk, or demo.
For product information, visit www.mathworks.com.

[Warning: Executing startup failed in matlabrc.
This indicates a potentially serious problem in your MATLAB setup, which
should
be resolved as soon as possible.  Error detected was:
MATLAB:run:FileNotFound
/software/build/config/drc_control_setup.m not found.] 
[> In matlabrc at 192] 
To reproduce this test use rng(739164675,'twister')

[Warning: Disabling autosave for simulink blocks (to avoid generating a
lot of
*.mdl.autosave files in your directory.  If you aren't a regular Simulink
user
and don't want this disabled, comment out this section in addpath_drake.
To disable this warning in the future, add
  warning('off','Drake:DisablingSimulinkAutosave')
to your matlab startup.m] 
[> In addpath_drake at 74
 In addpath_drake at 4] 
[Warning: This model has no mex pointer. Only checking collisions between
terrain contact points and terrain] 
[> In WarningManager>WarningManager.warnOnce at 21
 In RigidBodyManipulator.collisionDetect at 101
 In RigidBodyManipulator.RigidBodyManipulator>RigidBodyManipulator.compile
at 835
 In RigidBodyManipulator.addRobotFromURDF at 59
 In
RigidBodyManipulator.RigidBodyManipulator>RigidBodyManipulator.RigidBodyManipulator at 73
 In testCOM at 8
 In fevalPackageSafe at 21] 
Check the CoM with robot only
Reference to non-existent field 'dTdqdot'.

Error in RigidBodyManipulator/forwardJacDot (line 27)
 if isempty(kinsol.dTdqdot{1})

Error in testCOM (line 25)
Jdot_mex = forwardJacDot(r,kinsol,0);

Error in fevalPackageSafe (line 21)
 feval(f);

<test_name>systems/plants/test/testCOM</test_name>
<error_id>MATLAB:nonExistentField</error_id> <error_message>Reference to
non-existent field 'dTdqdot'.</error_message> 

```

RussTedrake, 2015-03-23:
just realized that this might have been due to sourcing drc_control_setup in my startup.m on that machine.  Will reopen if/when it appears again.

RussTedrake, 2015-03-24:
ok, it's real.  just happened on drake006, too. 

Site: drake006
Build Name: minimal-RobotLocomotion-master
Build Server Directory: /tmp/drake-minimal-10362/drake
CDash: http://kobol.csail.mit.edu/cd/viewTest.php?onlyfailed&buildid=12032

Most recent commits to drake included in this build:
```
9b11f69 Merge pull request #879 from rdeits/rd-clear-active-set-upstream
3a1a78c Merge pull request #875 from rdeits/rd-toe-off-upstream
4d3b22c const
16c0d92 Merge pull request #887 from
psiorx/jc-fix-testPositionConstraintsmex
84aff10 Merge pull request #814 from tkoolen/cmm-com-cleanup
```

systems/plants/test/testCOM (unknown issue)
```
error_id = MATLAB:nonExistentField
error_message = Reference to non-existent field 'dTdqdot'.
is_segfault = 0
```
```Matlab

                           < M A T L A B (R) >
                 Copyright 1984-2012 The MathWorks, Inc.
                   R2012b (8.0.0.783) 64-bit (glnxa64)
                             August 22, 2012

To get started, type one of these: helpwin, helpdesk, or demo.
For product information, visit www.mathworks.com.

To reproduce this test use rng(747743136,'twister')

[Warning: Disabling autosave for simulink blocks (to avoid generating a
lot of
*.mdl.autosave files in your directory.  If you aren't a regular Simulink
user
and don't want this disabled, comment out this section in addpath_drake.
To disable this warning in the future, add
  warning('off','Drake:DisablingSimulinkAutosave')
to your matlab startup.m] 
[> In addpath_drake at 74
 In addpath_drake at 4] 
[Warning: This model has no mex pointer. Only checking collisions between
terrain contact points and terrain] 
[> In WarningManager>WarningManager.warnOnce at 21
 In RigidBodyManipulator.collisionDetect at 101
 In RigidBodyManipulator.RigidBodyManipulator>RigidBodyManipulator.compile
at 835
 In RigidBodyManipulator.addRobotFromURDF at 59
 In
RigidBodyManipulator.RigidBodyManipulator>RigidBodyManipulator.RigidBodyManipulator at 73
 In testCOM at 8
 In fevalPackageSafe at 21] 
Check the CoM with robot only
Reference to non-existent field 'dTdqdot'.

Error in RigidBodyManipulator/forwardJacDot (line 27)
 if isempty(kinsol.dTdqdot{1})

Error in testCOM (line 25)
Jdot_mex = forwardJacDot(r,kinsol,0);

Error in fevalPackageSafe (line 21)
 feval(f);

<test_name>systems/plants/test/testCOM</test_name>
<error_id>MATLAB:nonExistentField</error_id> <error_message>Reference to
non-existent field 'dTdqdot'.</error_message> 

```

tkoolen, 2015-03-26:
Looking at this, you can assign it to me.

PR #902 (2015-03-26): Got rid of forwardJacDot calls in testCOM when use_new_kinsol is true
Fixes #889.
closed 2015-03-26 by RussTedrake