RobotLocomotion/drake #893: Inactive Constraints in solveLCP's FastQP formulation are never included

psiorx, 2015-03-23:
The if-statement on line 569 in TimeSteppingRigidBodyManipulator.m sets 

    M_active = z_inactive 

in both of the branches.

Then, on line 600, M_inactive is computed as 

    M_inactive = ~M_active & z_inactive;

Since M_active is always equal to z_inactive, this is a contradiction.

Since M_inactive will be a vector of all zeros, the lines that follow such as 

    Ain_fqp = [-M(M_inactive,z_inactive); -eye(n_z_inactive)];
    bin_fqp = [w(M_inactive); -lb(z_inactive)];

will not actually pull any of the desired constraints out of the LCP problem.

@RussTedrake This also appears to be the case in your lcp_thresholds branch.  Could this have implications on the frequency with which fastQP fails?

psiorx, 2015-03-26:
I ran some tests today to see how much M_inactive affects the success rate of fastQP.  I tried the following four configurations in runAtlasHandControl with various hands attached and recorded the number of fastQP failures (calls to pathlcp) in each trial:

    M_inactive = z_inactive;

    M_inactive = ~z_inactive

    M_inactive = logical(ones(numel(z_inactive), 1))

    M_inactive = logical(zeros(numel(z_inactive), 1));
    ^we are currently using this, basically

It seems like the choice of hand has a much bigger impact on the number of fastQP failures than the choice of M_inactive.  However, it may just be the case that all of the above choices are equally incorrect.

Here's the raw data: (pathlcp calls / fastqp calls)
#### robotiq

    M_inactive = z_inactive;
    635 / 1005

    M_inactive = ~z_inactive;
    636 / 1005

    M_inactive = logical(ones(numel(z_inactive), 1));
    612 / 1005

    M_inactive = logical(zeros(numel(z_inactive), 1));
    628 / 1005

#### robotiq_tendons

    M_inactive = z_inactive;
    36 / 1001

    M_inactive = ~z_inactive;
    66 / 1001

    M_inactive = logical(ones(numel(z_inactive), 1));
    40 / 1001

    M_inactive = logical(zeros(numel(z_inactive), 1));
    37 / 1001

### robotiq_simple

    M_inactive = z_inactive;
    23 / 1001

    M_inactive = ~z_inactive;
    44 / 1001

    M_inactive = logical(ones(numel(z_inactive), 1));
    82 / 1001

    M_inactive = logical(zeros(numel(z_inactive), 1));
    24 / 1001

psiorx, 2015-03-26:
Digging a little deeper, I think I found why this was never causing problems before.  The call to fastQPmex just ignores the inequality constraints completely.  

    [z_,info_fqp] = fastQPmex(QblkDiag,fqp,Ain_fqp,bin_fqp,Aeq,beq,[]);

The last argument passed in above is the set of active constraints.  Since we're always passing in an empty set, all of the inequality constraints are ignored.

kuindersma, 2015-03-27:
By passing the empty set (btw, not sure why this is happening in the code...) we're saying that no inequality constraints are active, but the non-active inequality constraints are still included in the KKT conditions.

PR #953 (2015-04-04): cleaned up the fastQP setup in the matlab implementation of solveLCP
Previously, we were building M_inactive but never using it.  This has been cleaned up. Resolves #893
closed 2015-04-04 by RussTedrake