RobotLocomotion/drake #894: Have to instantiate r_pure before instantiating r_complete, possibly a AtlasState singleton problem

hongkai-dai, 2015-03-24:
In drakeAtlasSimulation, it instantiates two Atlas object, the `r_pure` and `r_complete`. `r_pure` does not have hands. `r_complete` has hands. `r_pure` is instantiated before `r_complete` as
```
r_pure = DRCAtlas([],options)
options.hands = 'robotiq_simple';
r_complete = DRCAtlas([],options);
```
But if we do not instantiate `r_pure` and instantiate `r_complete` directly, namely to comment out line 39 in `drakeAtlasSimulation`, then I get the following error
```
Error using MultiCoordinateFrame/replaceFrameNum
(line 353)
new subframe does not match the dimensions of the
frame you are trying to replace

Error in DRCAtlas/compile (line 130)
        atlas_state_frame =
        replaceFrameNum(atlas_state_frame,1,drcFrames.AtlasState(obj));
        
Error in DRCAtlas (line 77)
      obj = compile(obj);

Error in drakeAtlasSimulation (line 75)
r_complete = DRCAtlas([],options);
```
The `options` for `r_complete` does not change. So I think the place that causes this problem is the singleton object `AtlasState`, that got constructed in `r_pure`.

gizatt, 2015-03-25:
I believe the underlying issue is that the AtlasState figures out what its own coordinates are based on the RBM that gets passed in. Looking at the code, I think that if a similar test was done in pure drake (no DRC), it should work -- atlasFrames.AtlasState explicitely looks for and ignores state for the hands when figure out its coordinates. However, drcFrames.AtlasState within the DRC codebase does not do this, and maybe it should? Either way this might not be a problem within Drake itself, but with our DRC-side extended versions of those frames.

gizatt, 2016-01-25:
I think this problem was fixed in the DRC codebase, but actually reemerged when support for external disturbances was added. Due to (otherwise arbitrary) differences in how the frames are assembled, the bug doesn't exist in Drake.

It's based in the notion that the AtlasState frame wants to be able to infer the state frame from the robot it's being generated for, the first time it's being assembled. This is great, except that extra states that the robot has (for hands, sensors, etc) can accidentally get assimilated into AtlasState if they're in the robot's state frame the first time an AtlasState is constructed. Right now atlasFrames/AtlasState has explicit checks to ensure this doesn't happen (since there are only a few kinds of extra state that can sneak in). It could be tidied up, but the Drake side of things looks like they work OK.

closed 2016-01-26 by RussTedrake