RobotLocomotion/drake #895: Try to Run TrajectorySwingUpAndBalance for LQR-tree

wclin892, 2015-03-24:
In the Pendulum case, which i try to run TrajectorySwingUpAndBalance,but  there is an error message: 
                "Undefined function 'mssprog2sedumi' for input arguments of type 'double'."
if I run TrajectorySwingUpAndBalance for LQR-tree. I try to modify the path to locate mssprog2sedumi. The system shows it can not find the "masspoly" parameter  of the program , and gives me the error meggage:
                "Reference to non-existent field 'v' ". 
I then try to add one statement "pv = msspoly (zeros (0,1))" to the program. The system shows that p.v of mssprog2sedumi is an empty matrix, and responds an erroneous f.

RussTedrake, 2015-04-20:
sorry we didn't respond to this sooner; must have slipped off the stack.  
are you still having the problem?  can you confirm that you've run `addpath_drake` before running the example?

wclin892, 2015-04-21:
Dear Russ Tedrake :

This problem has been resolved for few days ago,
I can run TrajectorySwingUpAndBalance.
Thank for your help ! !

But I have some new error messages.
Recently, I try to add LQR-Trees algorithm for Cart-Pole and make it work.
I use example is " runTrajectorySwingUpAndBalance" on Cart-Pole development
folder, but I am have big problem in it.
In this case, there is an error message:

Error using PolynomialSystem/sampledFiniteTimeVerification (line 133)
infeasible rho. increase options.rho0_tau

Error in CartPolePlant/trajectorySwingUpAndBalance (line 140)

Vtraj=sampledFiniteTimeVerification(psys,xtraj.getBreaks(),Vf,Vtraj,options);

Error in runTrajectorySwingUpAndBalance1 (line 5)
c = trajectorySwingUpAndBalance(p);

Could I ask for you help to give me a guidance to resolve these problems?

Sincerely Yours

David Lin

anirudhamajumdar, 2015-04-22:
Hi David,

I made a few (small) modifications to CartPolePlant.m to make this work. The code (with explanatory comments saying "David") is copied below:

classdef CartPolePlant < Manipulator

  properties
    mc = 10;   % mass of the cart in kg
    mp = 1;    % mass of the pole (point mass at the end) in kg
    l = 0.5;   % length of the pole in m
    g = 9.81;  % gravity m/s^2
    
    xG;
    uG;
  end
  
  methods
    function obj = CartPolePlant
      obj = obj@Manipulator(2,1);
      obj = setInputLimits(obj,-30,30);
      obj = setOutputFrame(obj,obj.getStateFrame);  % allow full-state feedback
      
      obj.xG = Point(obj.getStateFrame,[0;pi;0;0]);
      obj.uG = Point(obj.getInputFrame,0);
    end
        
    function [H,C,B] = manipulatorDynamics(obj,q,qd)
      mc=obj.mc;  mp=obj.mp;  l=obj.l;  g=obj.g;
      s = sin(q(2)); c = cos(q(2));

      H = [mc+mp, mp*l*c; mp*l*c, mp*l^2];
      C = [0 -mp*qd(2)*l*s; 0 0];
      G = [0; mp*g*l*s];
      B = [1; 0];
      
      C = C*qd + G;
    end
    
    function [f,df,d2f,d3f] = dynamics(obj,t,x,u)
      f = dynamics@Manipulator(obj,t,x,u);
      if (nargout>1)
        [df,d2f,d3f]= dynamicsGradients(obj,t,x,u,nargout-1);
      end
    end
    
    function x0 = getInitialState(obj)
      x0 = randn(4,1);
    end
    
  end

  methods 
    function [c,V]=balanceLQR(obj)
      Q = diag([1 50 1 50]);
      R = .2;

      if (nargout<2)
        c = tilqr(obj,obj.xG,obj.uG,Q,R);
      else
        if any(~isinf([obj.umin;obj.umax]))
          error('currently, you must disable input limits to estimate the ROA');
        end
        [c,V] = tilqr(obj,obj.xG,obj.uG,Q,R);
        pp = feedback(obj.taylorApprox(0,obj.xG,obj.uG,3),c);
        options.method='levelSet';
        V=regionOfAttraction(pp,V,options);
      end
    end
    
    function [c,V]=balanceHinf(obj)
      Bw = [zeros(2,2); eye(2)];  %uncertainty affects velocities only
      Q = diag([1 50 1 50]);
      R = .1;
      gamma = 20;
      if (nargout<2)
        c = tiHinf(obj,obj.xG,obj.uG,Q,R,Bw,gamma);
      else
        if any(~isinf([obj.umin;obj.umax]))
          error('currently, you must disable input limits to estimate the ROA');
        end
        [c,V]=tiHinf(obj,obj.xG,obj.uG,Q,R,Bw,gamma);
        pp = feedback(obj.taylorApprox(0,obj.xG,obj.uG,3),c);
        options.method='levelSet';
        V=regionOfAttraction(pp,V,options);
      end        
    end
    
    function [utraj,xtraj]=swingUpTrajectory(obj)
      x0 = zeros(4,1); tf0 = 4; xf = double(obj.xG);
      N = 41; % David: This controls the number of time samples (knot points) used
              % for trajectory optimization. The more time samples you have
              % the more accurate the trajectory will be. But the
              % computation time will increase as you increase this. I've
              % increased the number of time samples to make it more
              % accurate.
      
      obj = setInputLimits(obj,-inf,inf);
      prog = DircolTrajectoryOptimization(obj,N,[2 6]);
      prog = prog.addStateConstraint(ConstantConstraint(x0),1);
      prog = prog.addStateConstraint(ConstantConstraint(xf),N);
      prog = prog.addRunningCost(@cost);
      prog = prog.addFinalCost(@finalCost);
      
      function [g,dg] = cost(dt,x,u)
        R = 1;
        g = sum((R*u).*u,1);
        dg = [zeros(1,1+size(x,1)),2*u'*R];
      end
      
      function [h,dh] = finalCost(t,x)
        h = t;
        dh = [1,zeros(1,size(x,1))];
      end
      
      traj_init.x = PPTrajectory(foh([0,tf0],[x0,xf]));
      for attempts=1:10
        tic
        [xtraj,utraj,z,F,info] = prog.solveTraj(tf0,traj_init);
        toc
        if info==1, break; end
      end
    end
    
    function [tv,Vtraj]=trajectorySwingUpAndBalance(obj)
      [ti,Vf] = balanceLQR(obj);

%      c = LQRTree(ti,Vf);
      [utraj,xtraj]=swingUpTrajectory(obj);  
      
      Q=diag([10,10,1,1]); R=.1;
      [tv,Vtraj] = tvlqr(obj,xtraj,utraj,Q,R,Vf);
      psys = taylorApprox(feedback(obj,tv),xtraj,[],3);
      
      options.rho0_tau = 5; % David: The funnel is defined as the rho(t) sublevel set of the Lyapunov function V(t).
                            % This parameter specifies the initial guess
                            % for rho(t). In particular, the initial rho(t)
                            % is specified as an exponential function with
                            % options.rho0_tau being the exponent (after
                            % time is scaled). See line 119 in
                            % sampledFiniteTimeVerification.m for the exact
                            % function. There's no magic formula for
                            % choosing this. You have to play around with
                            % it a bit (but in general, if a particular
                            % value doesn't work, increasing it is a
                            % reasonable choice).
                            
      options.max_iterations = 3; % David: This controls the maximum number of iterations the bilinear alternations run for.
                                  % It's set relatively low right now.
      
      ts = xtraj.getBreaks();
      ts = linspace(ts(1),ts(end),12);  % David: These are time samples at which the Lyapunov conditions for the funnel are checked.
                                        % I've chosen a slightly coarse
                                        % sampling here to make things
                                        % faster. You can increase it if
                                        % you need more accuracy.
      Vtraj=sampledFiniteTimeVerification(psys,ts,Vf,Vtraj,options);

%      c = c.addTrajectory(tv,Vtraj);
    end  
  end
end

wclin892, 2015-04-23:
Dear Anirudha Majumdar :

Thank you to resolve my problems !!
I have been made modifications on CartPolePlant.m and tried to run
TrajectorySwingUpAndBalance.
I made according your commands to modify and try to make this work.
But I discovered that it can spend lot of time(always busy) on " [prog,info]
= sedumi(prog,-rhointegral,0);" (from sampledFiniteTimeVerification.m) to
run and can not continue.
It always busy so I stop this work (Ctrl+C) and there is an error message :

Operation terminated by user during sedumi (line 463)

In mssprog/sedumi (line 23)
[x,y,info]=sedumi(A,B,optimize*C,K,pars);

In PolynomialSystem/sampledFiniteTimeVerification>optimizeRho (line 477)
  [prog,info] = sedumi(prog,-rhointegral,0); %1,struct());

In PolynomialSystem/sampledFiniteTimeVerification (line 150)
      [rho,rhointegral]=optimizeRho(x,V,Vdot,L,dts,Vmin,rhof,options);

In CartPolePlant/trajectorySwingUpAndBalance (line 145)
      Vtraj=sampledFiniteTimeVerification(psys,ts,Vf,Vtraj,options);

In runTrajectorySwingUpAndBalance1 (line 6)
c = trajectorySwingUpAndBalance(p);

By the way, I have a sedumi folder(from
https://github.com/sqlp/sedumi)  because
I discovered the "function  balanceLQR" needs it.

Could I ask for you help to give me a guidance to resolve these problems?

Sincerely Yours

David Lin

RussTedrake, 2015-04-23:
Sedumi is slow.  Mosek is much faster.  It will take a long time to run.  Don't hit ctrl-c.

wclin892, 2015-04-23:
Dear Russ Tedrake:

First, I use Sedumi to run, but it is much slow and I wait for 30 minutes
and the program can't run to finish, there is an error message:

Error in CartPolePlant>CartPolePlant.trajectorySwingUpAndBalance (line 138)
      [ti,Vf] = balanceLQR(obj);

Error using CartPolePlant/trajectorySwingUpAndBalance
Output argument "c" (and maybe others) not assigned during call to
"C:\Users\yang\Desktop\drake-precompiled-win64\drake\examples\CartPole\CartPolePlant.m>CartPolePlant.trajectorySwingUpAndBalance".

Error in runTrajectorySwingUpAndBalance (line 6)
c = trajectorySwingUpAndBalance(p);

---------------------------------------------------------------------------------------------------------------------------

Then, I try to use Mosek to want to run quickly so I install Mosek
Version7.1 in Matlab2013a,
but it have some new error messages:

MOSEK Version 7.1.0.28 (Build date: 2015-4-14 10:10:55)
Copyright (c) 1998-2015 MOSEK ApS, Denmark. WWW: http://mosek.com
Platform: Windows/64-X86

Error using PolynomialSystem/regionOfAttraction>levelSetMethod (line 334)
Problem looks primal infeasible

Error in PolynomialSystem/regionOfAttraction (line 136)
      V = levelSetMethod(V,f,options);

Error in PendulumPlant/balanceLQR (line 79)
        V=regionOfAttraction(pp,V,options);

Error in PendulumPlant/trajectorySwingUpAndBalance (line 122)
      [ti,Vf] = balanceLQR(obj);

Error in runTrajectorySwingUpAndBalance2 (line 12)
c = trajectorySwingUpAndBalance(pd);

I used Mosek from https://mosek.com/resources/downloads
I don't know that I use Mosek correct or not?

Could I ask for you help to give me a guidance to resolve these problems?

Sincerely Yours

David Lin

wclin892, 2015-04-23:
Dear Russ Tedrake & Anirudha Majumdar :

I have been resolved Mosek error message, and run this program,
but it still is much slow and I wait for 30 minutes and the program can't
run to finish,
there is an error message:

Error in CartPolePlant>CartPolePlant.trajectorySwingUpAndBalance (line 138)
      [ti,Vf] = balanceLQR(obj);

Error using CartPolePlant/trajectorySwingUpAndBalance
Output argument "c" (and maybe others) not assigned during call to
"C:\Users\yang\Desktop\drake-precompiled-3ade858-win64\drake\examples\CartPole\CartPolePlant.m>CartPolePlant.trajectorySwingUpAndBalance".

Error in runTrajectorySwingUpAndBalance (line 6)
c = trajectorySwingUpAndBalance(p);

Could I ask for you help to give me a guidance to resolve these problems?

Sincerely Yours

David Lin

2015-04-23 17:03 GMT+08:00 Wei Chih Lin <wclin892@gmail.com>:

RussTedrake, 2015-04-23:
Ani's code is fine.  I'm afraid you must have a typo in the way you've merged it in; you are trying to return a variable named c but ani's code renamed that to tv.  I will include a working version of it in a pull request in a few moments.

closed 2015-04-23 by RussTedrake

wclin892, 2015-04-23:
Dear Russ Tedrake:

Thank you for responding !!
I will continue to try to run, that I hope make it work.

Sincerely Yours

David Lin

wclin892, 2015-04-30:
Dear Russ Tedrake:

Sorry !
In this cart-pole case, I have some questions.
Because I want to understand LQR-Tree algorithm on cart-pole, I want to try to add it.
If I want to add LQR-Tree algorithm to cart-pole, how should i to make it ?

Sincerely Yours

David

PR #1036 (2015-04-23): updated cart-pole example with working SOS funnels from Ani.  
resolves #895
commit a589f683 (2015-04-23): updated cart-pole example with working SOS funnels from Ani.  resolves #895.