RobotLocomotion/drake #898: lcmt_joint_pd_override compile isseu:  master doesn't compile on mac!

RussTedrake, 2015-03-25:
[ 29%] In file included from /Users/russt/robots/leo/drake/pod-build/lcmgen/lcmtypes/drake_lcmt_qp_controller_input.c:8:
/Users/russt/robots/leo/drake/pod-build/lcmgen/lcmtypes/drake_lcmt_qp_controller_input.h:23:10: fatal error: 'drake_lcmt_joint_pd_override.h' file
      not found
#include "drake_lcmt_joint_pd_override.h"
         ^
Built target drakeShapes
1 error generated.
make[3]: *** [CMakeFiles/drake_lcmtypes.dir/lcmgen/lcmtypes/drake_lcmt_qp_controller_input.c.o] Error 1
make[2]: *** [CMakeFiles/drake_lcmtypes.dir/all] Error 2
make[2]: *** Waiting for unfinished jobs....
[ 29%] [ 29%] Built target testDrakeGeometryUtil
[ 30%] [ 31%] Built target testGeometryGradientsmex
Built target testDrakeGradientUtil
Built target testGeometryConversionFunctionsmex
[ 32%] Built target testQuatmex
warning: [options] bootstrap class path not set in conjunction with -source 1.6
/Users/russt/robots/leo/drake/pod-build/lcmgen/lcmtypes/drake/lcmt_qp_controller_input.java:23: error: cannot find symbol
    public drake.lcmt_joint_pd_override joint_pd_override[];
                ^
  symbol:   class lcmt_joint_pd_override
  location: package drake
/Users/russt/robots/leo/drake/pod-build/lcmgen/lcmtypes/drake/lcmt_qp_controller_input.java:48: error: cannot find symbol
             + drake.lcmt_joint_pd_override._hashRecursive(classes)
                    ^
  symbol:   class lcmt_joint_pd_override
  location: package drake
/Users/russt/robots/leo/drake/pod-build/lcmgen/lcmtypes/drake/lcmt_qp_controller_input.java:140: error: cannot find symbol
        this.joint_pd_override = new drake.lcmt_joint_pd_override[(int) num_joint_pd_overrides];
                                          ^
  symbol:   class lcmt_joint_pd_override
  location: package drake
/Users/russt/robots/leo/drake/pod-build/lcmgen/lcmtypes/drake/lcmt_qp_controller_input.java:142: error: cannot find symbol
            this.joint_pd_override[a] = drake.lcmt_joint_pd_override._decodeRecursiveFactory(ins);
                                             ^
  symbol:   class lcmt_joint_pd_override
  location: package drake
/Users/russt/robots/leo/drake/pod-build/lcmgen/lcmtypes/drake/lcmt_qp_controller_input.java:176: error: cannot find symbol
        outobj.joint_pd_override = new drake.lcmt_joint_pd_override[(int) num_joint_pd_overrides];
                                            ^
  symbol:   class lcmt_joint_pd_override
  location: package drake
5 errors
1 warning

RussTedrake, 2015-03-25:
oops, looks like a sufficiently clean make resolved it.  sorry!

closed 2015-03-25 by RussTedrake

rdeits, 2015-03-25:
Yeah, a `make clean` seems to be required every time we add a new lcmtype