RobotLocomotion/drake #901: TerrainContactPoints in C++ implementation?

steinachim, 2015-03-26:
Hi,
I'm trying to use the `QuasiStaticConstraint` for `RigidBodyManipulator` inverse kinemacs in the C++ implementation. In MATLAB, I simply use `RigidBody::getTerrainContactPoints()`. Since this function is missing in C++, I assumed `RigidBody::contact_points` would do the job, but this matrix is always empty.

I did some digging and found that I have access to the collision elements through `RigidBody::collision_element_ids`, but they are missing the `getTerrainContactPoints` and `getPoints` functions I see in the MATLAB implementation.

Is this a case of "not implemented yet" (if so, I'd start on doing that) or is there another (better) way of handling this now?

Best,
Achim

RussTedrake, 2015-03-26:
Hi Achim,

I believe it's "not implemented yet".  Contributions very welcome!

- Russ

RussTedrake, 2015-03-26:
And I should say that the big reason for this is that our terrain contact logic is actually only implemented in matlab so far.  We do an approximate 2.5D collision check for heightmaps which hasn't been pushed to C yet (but it's definitely on our list!).

steinachim, 2015-03-27:
Could you tell me a bit about the semantics of the `RigidBody::contact_pts` member? Are those the actual contacts of the body with the environment or just the potential contact points as represented by the `DrakeShapes::Geometry` classes?

avalenzu, 2015-03-28:
In C++ manipulators constructed via `constructModelmex`, the contact_points
field corresponds to the points returned by the MATLAB method
`getTertainContactPoints`. That is, the center of all zero radius sphere
geometries and the vertices of all box geometries.

closed 2015-04-20 by RussTedrake