RobotLocomotion/drake #913: testGradientUtil fails with eigen 3.2.4 on drake006

RussTedrake, 2015-03-28:
on line 82.  will a few extra printouts:
```
      if (!dAB.isApprox(dAB_check, 1e-10)) {
        std::cout << "test number " << testnr << " failed" << std::endl;
        std::cout << "error max:"	<< (dAB_check-dAB).maxCoeff() << ", min: " << (dAB_check-dAB).minCoeff() << std::endl;
        throw std::runtime_error("wrong.");
      }
    }
```

i see:
```
russt@drake006:/tmp/drake-master-28238/drake$ ./pod-build/bin/testDrakeGradientUtil
test number 0 failed
error max:0.600422, min: -0.663261
terminate called after throwing an instance of 'std::runtime_error'
  what():  wrong.
Aborted (core dumped)
```
in other words, it's not a tolerance issue and it happens on the very first random test (every time).

tkoolen, 2015-03-30:
The output of matGradMultMat seems fine (compared to our Matlab implementation). However what we're comparing against is wrong I believe. We're using Eigen::kroneckerProduct to compute dAB_check, and it seems that the first element of the matrix is uninitialized. All of the other elements match the output of matGradMultMat. E.g.
dAB:
0.0921673   -1.7191
 0.135026 -0.584029
-0.244339  0.425139
 0.381803  0.954083

vs.
dAB_check:
1.33345e-316      -1.7191
    0.135026    -0.584029
   -0.244339     0.425139
    0.381803     0.954083

I'll look into it some more and possibly file a bug with eigen, but nothing to worry about on our end.

tkoolen, 2015-03-30:
This is actually fixed by adding a .eval to the right hand side. See pull request.

PR #924 (2015-03-31): Fix testDrakeGradientUtil
Add a missing .eval() in testMatGradMultMat.
Fixes #913.
closed 2015-03-31 by RussTedrake