RobotLocomotion/drake #917: testKinCnst segfaults with eigen 3.2.4

RussTedrake, 2015-03-29:
```
Site: drake006
Build Name: master-pull-rdeits-rd-frozen-plan-into-RobotLocomotion-master
Build Server Directory: /tmp/drake-master-12799/drake
Pull Request: #916
CDash: http://kobol.csail.mit.edu/cd/viewTest.php?onlyfailed&buildid=12133

Most recent commits to drake included in this build:

891bb2d Merge branch 'rd-floating-base-weights' into rd-frozen-plan
118fdc4 by default, don't set any weight on floating base pose
0c63346 don't modify the param_set_name in a frozen plan
eaf358f Merge pull request #915 from rdeits/rd-eta
2b71fd9 change eta back to its original value
375 tests run:
359 passed
0 cancelled (with missing dependencies )
6 timed out
10 failed
1 known issue #18: Singularities in the euler-angle floating base joint cause infrequent errors on unit tests with random initial conditions
4 known issue #374: failing imu unit tests
1 known issue #466: implement Jdot*v for drakeFunction.kinematic methods
1 known issue #585: testIndividualCentersOfPressure fails
1 known issue #822: testURDF fails now that it is parsing every urdf in drake
1 known issue #913: testGradientUtil fails with eigen 3.2.4 on drake006
1 unknown

systems/plants/constraint/test/testKinCnst (unknown issue)

error_id = 
error_message = 
is_segfault = 1
                            < M A T L A B (R) >
                  Copyright 1984-2012 The MathWorks, Inc.
                    R2012b (8.0.0.783) 64-bit (glnxa64)
                              August 22, 2012

To get started, type one of these: helpwin, helpdesk, or demo.
For product information, visit www.mathworks.com.

To reproduce this test use rng(788457461,'twister')

[Warning: Disabling autosave for simulink blocks (to avoid generating a lot of
*.mdl.autosave files in your directory.  If you aren't a regular Simulink user
and don't want this disabled, comment out this section in addpath_drake.
To disable this warning in the future, add
   warning('off','Drake:DisablingSimulinkAutosave')
to your matlab startup.m] 
[> In addpath_drake at 74
  In addpath_drake at 4] 
[Warning: Contact is not supported by the dynamics methods of this class.
Consider using TimeSteppingRigidBodyManipulator or
HybridPlanarRigidBodyManipulator] 
[> In RigidBodyManipulator.RigidBodyManipulator>RigidBodyManipulator.compile at 882
  In RigidBodyManipulator.addRobotFromURDF at 59
  In testKinCnst at 23
  In fevalPackageSafe at 21] 
[Warning: Contact is not supported by the dynamics methods of this class.
Consider using TimeSteppingRigidBodyManipulator or
HybridPlanarRigidBodyManipulator] 
[> In RigidBodyManipulator.RigidBodyManipulator>RigidBodyManipulator.compile at 882
  In testKinCnst at 30
  In fevalPackageSafe at 21] 
Check world CoM constraint
Check world CoM constraint with empty robot number
Check world CoM constraint with empty tspan
Check world CoM constraint with nan
Check world CoM constraint with multiple robots
Check world position constraint
Check world position constraint with nan
Check world quaternion constraint
Check world quaternion constraint, a subtle case
Check world euler constraint
Check world gaze orientation constraint
Check world gaze orientation constraint with empty threshold
Check world gaze direction constraint
Check world gaze direction constraint with empty threshold
Check world gaze target constraint
Check world gaze target constraint with empty threshold
Check world fixed position constraint
Check world fixed orientation constraint
Check world fixed body pose constraint
Check all-to-all closest-distance constraint
Check minimum distance constraint

------------------------------------------------------------------------
       Segmentation violation detected at Sat Mar 28 19:10:47 2015
------------------------------------------------------------------------
```
<matlab_test>strcmp(test_name,'systems/plants/constraint/test/testKinCnst') && is_segfault</matlab_test>

tkoolen, 2015-03-30:
Working on this. I got
/home/twan/code/drake-distro/build/include/eigen3/Eigen/src/Core/DenseCoeffsBase.h:394: Eigen::DenseCoeffsBase<Derived, 1>::Scalar& Eigen::DenseCoeffsBase<Derived, 1>::operator()(Eigen::DenseCoeffsBase<Derived, 1>::Index) [with Derived = Eigen::Matrix<double, -1, 1>; Eigen::DenseCoeffsBase<Derived, 1>::Scalar = double; Eigen::DenseCoeffsBase<Derived, 1>::Index = long int]: Assertion `index >= 0 && index < size()' failed.

Stack Trace (from fault):
[  0] 0x00007fa2f3f2fcc9                    /lib/x86_64-linux-gnu/libc.so.6+00224457 gsignal+00000057
[  1] 0x00007fa2f3f330d8                    /lib/x86_64-linux-gnu/libc.so.6+00237784 abort+00000328
[  2] 0x00007fa2f3f28b86                    /lib/x86_64-linux-gnu/libc.so.6+00195462
[  3] 0x00007fa2f3f28c32                    /lib/x86_64-linux-gnu/libc.so.6+00195634
[  4] 0x00007fa18dfc5d22 /home/twan/code/drake-distro/drake/systems/plants/constraint/constructPtrRigidBodyConstraintmex.mexa64+00245026 _ZN5Eigen15DenseCoeffsBaseINS_6MatrixIdLin1ELi1ELi0ELin1ELi1EEELi1EEclEl+00000072
[  5] 0x00007fa18dcae712 /home/twan/code/drake-distro/drake/pod-build/lib/libdrakeRigidBodyConstraint.so+01787666 _ZNK21MinDistanceConstraint4evalEPKdRN5Eigen6MatrixIdLin1ELi1ELi0ELin1ELi1EEERNS3_IdLin1ELin1ELi0ELin1ELin1EEE+00002862
[  6] 0x00007fa18d6e1b70 /home/twan/code/drake-distro/drake/systems/plants/constraint/testSingleTimeKinCnstmex.mexa64+00023408 mexFunction+00000536
[  7] 0x00007fa2ebe5a72a     /usr/local/MATLAB/R2014a/bin/glnxa64/libmex.so+00120618 mexRunMexFile+00000090
[  8] 0x00007fa2ebe56a94     /usr/local/MATLAB/R2014a/bin/glnxa64/libmex.so+00105108
[  9] 0x00007fa2ebe57fb4     /usr/local/MATLAB/R2014a/bin/glnxa64/libmex.so+00110516
[ 10] 0x00007fa2eb251ad9 /usr/local/MATLAB/R2014a/bin/glnxa64/libmwm_dispatcher.so+00670425 _ZN8Mfh_file11dispatch_fhEiPP11mxArray_tagiS2_+00000697
[ 11] 0x00007fa2ea32820e /usr/local/MATLAB/R2014a/bin/glnxa64/libmwm_interpreter.so+02601486
[ 12] 0x00007fa2ea2e31d0 /usr/local/MATLAB/R2014a/bin/glnxa64/libmwm_interpreter.so+02318800
[ 13] 0x00007fa2ea2e51ea /usr/local/MATLAB/R2014a/bin/glnxa64/libmwm_interpreter.so+02327018
[ 14] 0x00007fa2ea2e8167 /usr/local/MATLAB/R2014a/bin/glnxa64/libmwm_interpreter.so+02339175
[ 15] 0x00007fa2ea2e626f /usr/local/MATLAB/R2014a/bin/glnxa64/libmwm_interpreter.so+02331247
[ 16] 0x00007fa2ea2e6ec4 /usr/local/MATLAB/R2014a/bin/glnxa64/libmwm_interpreter.so+02334404
[ 17] 0x00007fa2ea34430b /usr/local/MATLAB/R2014a/bin/glnxa64/libmwm_interpreter.so+02716427
[ 18] 0x00007fa2eb251ad9 /usr/local/MATLAB/R2014a/bin/glnxa64/libmwm_dispatcher.so+00670425 _ZN8Mfh_file11dispatch_fhEiPP11mxArray_tagiS2_+00000697
[ 19] 0x00007fa2ea32820e /usr/local/MATLAB/R2014a/bin/glnxa64/libmwm_interpreter.so+02601486
[ 20] 0x00007fa2ea2e31d0 /usr/local/MATLAB/R2014a/bin/glnxa64/libmwm_interpreter.so+02318800
[ 21] 0x00007fa2ea2e51ea /usr/local/MATLAB/R2014a/bin/glnxa64/libmwm_interpreter.so+02327018
[ 22] 0x00007fa2ea2e8167 /usr/local/MATLAB/R2014a/bin/glnxa64/libmwm_interpreter.so+02339175
[ 23] 0x00007fa2ea2e626f /usr/local/MATLAB/R2014a/bin/glnxa64/libmwm_interpreter.so+02331247
[ 24] 0x00007fa2ea2e6ec4 /usr/local/MATLAB/R2014a/bin/glnxa64/libmwm_interpreter.so+02334404
[ 25] 0x00007fa2ea34430b /usr/local/MATLAB/R2014a/bin/glnxa64/libmwm_interpreter.so+02716427
[ 26] 0x00007fa2eb251ad9 /usr/local/MATLAB/R2014a/bin/glnxa64/libmwm_dispatcher.so+00670425 _ZN8Mfh_file11dispatch_fhEiPP11mxArray_tagiS2_+00000697
[ 27] 0x00007fa2ea32820e /usr/local/MATLAB/R2014a/bin/glnxa64/libmwm_interpreter.so+02601486
[ 28] 0x00007fa2ea2c91b0 /usr/local/MATLAB/R2014a/bin/glnxa64/libmwm_interpreter.so+02212272
[ 29] 0x00007fa2ea2e425f /usr/local/MATLAB/R2014a/bin/glnxa64/libmwm_interpreter.so+02323039
[ 30] 0x00007fa2ea2e8167 /usr/local/MATLAB/R2014a/bin/glnxa64/libmwm_interpreter.so+02339175
[ 31] 0x00007fa2ea2e626f /usr/local/MATLAB/R2014a/bin/glnxa64/libmwm_interpreter.so+02331247
[ 32] 0x00007fa2ea2e6ec4 /usr/local/MATLAB/R2014a/bin/glnxa64/libmwm_interpreter.so+02334404
[ 33] 0x00007fa2ea34430b /usr/local/MATLAB/R2014a/bin/glnxa64/libmwm_interpreter.so+02716427
[ 34] 0x00007fa2eb251c5f /usr/local/MATLAB/R2014a/bin/glnxa64/libmwm_dispatcher.so+00670815 _ZN8Mfh_file11dispatch_fhEiPP11mxArray_tagiS2_+00001087
[ 35] 0x00007fa2ea317135 /usr/local/MATLAB/R2014a/bin/glnxa64/libmwm_interpreter.so+02531637
[ 36] 0x00007fa2ea2de0d9 /usr/local/MATLAB/R2014a/bin/glnxa64/libmwm_interpreter.so+02298073
[ 37] 0x00007fa2ea2dadc7 /usr/local/MATLAB/R2014a/bin/glnxa64/libmwm_interpreter.so+02284999
[ 38] 0x00007fa2ea2db193 /usr/local/MATLAB/R2014a/bin/glnxa64/libmwm_interpreter.so+02285971
[ 39] 0x00007fa2ec084afc /usr/local/MATLAB/R2014a/bin/glnxa64/libmwbridge.so+00142076
[ 40] 0x00007fa2ec085791 /usr/local/MATLAB/R2014a/bin/glnxa64/libmwbridge.so+00145297 _Z8mnParserv+00000721
[ 41] 0x00007fa2f533892f   /usr/local/MATLAB/R2014a/bin/glnxa64/libmwmcr.so+00489775 _ZN11mcrInstance30mnParser_on_interpreter_threadEv+00000031
[ 42] 0x00007fa2f5319b6d   /usr/local/MATLAB/R2014a/bin/glnxa64/libmwmcr.so+00363373
[ 43] 0x00007fa2f5319be9   /usr/local/MATLAB/R2014a/bin/glnxa64/libmwmcr.so+00363497
[ 44] 0x00007fa2e9a10d46   /usr/local/MATLAB/R2014a/bin/glnxa64/libmwuix.so+00343366
[ 45] 0x00007fa2e99f3382   /usr/local/MATLAB/R2014a/bin/glnxa64/libmwuix.so+00222082
[ 46] 0x00007fa2f5a8e50f /usr/local/MATLAB/R2014a/bin/glnxa64/libmwservices.so+02323727
[ 47] 0x00007fa2f5a8e67c /usr/local/MATLAB/R2014a/bin/glnxa64/libmwservices.so+02324092
[... cut ...]
[ 51] 0x00007fa2f5a904c0 /usr/local/MATLAB/R2014a/bin/glnxa64/libmwservices.so+02331840 _Z25svWS_ProcessPendingEventsiib+00000080
[ 52] 0x00007fa2f531a098   /usr/local/MATLAB/R2014a/bin/glnxa64/libmwmcr.so+00364696
[ 53] 0x00007fa2f531a3bf   /usr/local/MATLAB/R2014a/bin/glnxa64/libmwmcr.so+00365503
[ 54] 0x00007fa2f531528f   /usr/local/MATLAB/R2014a/bin/glnxa64/libmwmcr.so+00344719
[ 55] 0x00007fa2f42c6182              /lib/x86_64-linux-gnu/libpthread.so.0+00033154
[ 56] 0x00007fa2f3ff347d                    /lib/x86_64-linux-gnu/libc.so.6+01025149 clone+00000109

tkoolen, 2015-03-30:
Found the cause. Discussed with @avalenzu, he will fix it.

PR #926 (2015-03-31): Correctly handle invalid time in C++ MinDistanceConstraint
Resolves #917.

The diff makes it look like there are a lot more changes than there really are. I moved the body of the function into the if statement that starts on l. 2366 and added the 'else' case to handle invalid times.
commit e9fdc81a (2015-03-31): Correctly handle invalid time in C++ MinDistanceConstraint

Resolves #917.
closed 2015-03-31 by RussTedrake