RobotLocomotion/drake #923: new kinsol c++ methods fail mysteriously if e.g. jdotv is not computed in doKinematicsNew

RussTedrake, 2015-03-30:
computing jdotv is false by default.  but it is required for computing in the inverseDynamics method.  and there are no checks/guards in this (and other methods) to make sure that doKinematicsNew was called with sufficient arguments.  

i'll add the check for jdotv to inverseDynamics, but i think we need to systematically go through and catch these.

tkoolen, 2015-03-30:
Ah, in doKinematicsNewmex I set compute_Jdotv to true by default if v is passed in, but the default value for the method is still false...

I'll go through the methods and check.

What I ultimately want is to have a KinematicSolution class that contains all of the cached computation as well as all of these booleans. Getter methods could check whether the data has actually been computed. It would clean up RigidBodyManipulator tremendously. Also, we could store multiple of these KinematicSolution objects, which @hongkai-dai requested a while back:
> One use case is when solving inverse kinematics problem with a kinematic constraint on multiple time instances. Suppose that we want to constrain the foot does not move between time 0 and time 1. This constraint requires both the kinsol at time 0 and time 1 to evaluate. If we do not cache kinsol, then when we evaluate kinsol at time 1, the kinsol at time 0 will be discarded. But this kinsol can be used in other constraint (for example, the hand should be at some certain position at time 0). When we evaluate that other constraint at time 0, we need to recompute kinsol at time 0, as it was discarded when we evaluate kinsol at time 1. So it is better to compute kinsol for different postures first, cache them, and for each constraint, pick out the corresponding kinsols, rather than computing the kinsol within each constraint.

But for now I'll just do a better job of checking.

tkoolen, 2015-03-30:
Could you assign this to me?

rdeits, 2015-03-31:
The idea of a kinematic solution object that holds all the cached computations sounds very nice indeed

PR #927 (2015-04-12): Add kinematics cache checks to new kinematics methods
Also added a test that ensures that runtime errors are thrown if and only if you don't have the required kinematics quantities cached.
Fixes #923.
closed 2015-04-12 by RussTedrake