RobotLocomotion/drake #935: removeVisualGeometry can fail for welded? links.

RussTedrake, 2015-04-01:
GliderBalanced.urdf was always loaded using floating=true before.  If you load it without the floating base, we get:
```
>> r = RigidBodyManipulator('/Users/russt/robots/leo/drake/examples/Glider/GliderBalanced.urdf')
Constructing a new airfoil
Using a flat plate airfoil.
Index exceeds matrix dimensions.

Error in RigidBodyManipulator/removeVisualGeometryFromBody (line 1393)
      for i = 1:length(obj.body(body_idx).visual_geometry)

Error in RigidBodyWing/updateVisualGeometry (line 612)
        model = model.removeVisualGeometryFromBody(obj.parent_id, [ obj.name
        '_urdf_geometry' ]);

Error in RigidBodyWing/onCompile (line 101)
      model = obj.updateVisualGeometry(model);

Error in RigidBodyManipulator/compile (line 691)
        [new_element, model] = model.force{i}.onCompile(model);

Error in RigidBodyManipulator/addRobotFromURDF (line 59)
model = compile(model);  % ideally this would happen on entry into any
function...

Error in RigidBodyManipulator (line 73)
          obj = addRobotFromURDF(obj,filename,zeros(3,1),zeros(3,1),options);
 ```

avalenzu, 2015-04-07:
This looks like a problem in the `onCompile` method of `RigidBodyWing` the `parent_id` property isn't getting updated when the link gets welded. As a result `removeVisualGeometry` is being called with an invalid body id.

RussTedrake, 2016-05-01:
resolving as "won't fix" due to #2228

closed 2016-05-01 by RussTedrake