RobotLocomotion/drake #939: Error using Vanderpol.

xtechJack, 2015-04-01:
Hi Guys,
 
Thank you for the amazing Drake Tool!
I am on Windows system with Matlab R2013a. 
When I run the VanDerPol.run()​, I am getting the following error.

Error using msspoly
Cannot redefine class 'msspoly' without a call to 'clear classes'.

Error in CoordinateFrame/getPoly (line 95)
          obj.poly=msspoly(obj.prefix(1),obj.dim);

Error in PolynomialSystem/getPolyDynamics (line 86)
        p_x=obj.getStateFrame.getPoly;

Error in PolynomialSystem/timeReverse (line 137)
      [p_dynamics_rhs,p_dynamics_lhs]=getPolyDynamics(obj);

Error in VanDerPol.run (line 32)
      rvdp = timeReverse(vdp);
 
I tried debugging but as isn't much available on msspoly couldn't figure out, I'll thankful if you can shove some light.

Thanks,
-Paras.

tkoolen, 2015-04-03:
Can't reproduce on OSX with Matlab 2014a, will try on Windows in a minute.

tkoolen, 2015-04-03:
Can't reproduce on Windows 7 64 bit with Matlab 2014a. I used the precompiled version 0.9.8 from https://github.com/RobotLocomotion/drake-distro/releases.

Could you tell me exactly what you did from the moment you started Matlab to arrive at this error? Also, could you post your startup.m?

xParas, 2015-04-03:
Hi,
Sorry I should have updated. Apparently I had spot installed previously, so there was conflict with spot in drake. It got resolved when i uninstalled the previous one.

Thank you for your reply though.

closed 2015-04-03 by RussTedrake