RobotLocomotion/drake #960: Error using runMixedIntegerForest.m

sir-avinash, 2015-04-05:
I get the following error when I try to execute runMixedIntegerForest.m. I have istalled your latest binary on Ubuntu 14.04. My gcc and g++ versions are 4.9.  

Error message:

Invalid MEX-file
'/home/avinash/drake-distro/drake/systems/plants/constructModelmex.mexa64':
/usr/local/MATLAB/R2014b/bin/glnxa64/../../sys/os/glnxa64/libstdc++.so.6:
version `GLIBCXX_3.4.18' not found (required by
/home/avinash/drake-distro/drake/pod-build/lib/libdrakeCollision.so)

Error in RigidBodyManipulator/createMexPointer (line 2093)
        obj.mex_model_ptr = constructModelmex(obj);

Error in RigidBodyManipulator/compile (line 831)
      model = createMexPointer(model);

Error in RigidBodyManipulator/addRobotFromURDF (line 59)
model = compile(model);  % ideally this would happen on entry into any
function...

Error in RigidBodyManipulator (line 73)
          obj =
          addRobotFromURDF(obj,filename,zeros(3,1),zeros(3,1),options);

Error in Quadrotor (line 11)
      obj =
      obj@RigidBodyManipulator(getFullPathFromRelativePath('quadrotor.urdf'),options);
      
Error in runMixedIntegerForest (line 5)
r = Quadrotor();

RussTedrake, 2015-04-06:
it looks like we have incompatible glibcxx versions.  i believe our build server is still running gcc 4.6.
it's a busy time for us to look into this, but if you're on ubuntu i'd recommend building from source -- it's quite easy and reliable.

sir-avinash, 2015-04-07:
Thanks Russ. I tried building from source and the error remained. I found a quick fix. I just copied the files 'libstdc++.so.6' and 'libstdc++.so.0.0.21' from /usr/lib/x86_64-linux-gnu (their original location on my system) to  /usr/local/MATLAB/R2014b/bin/glnxa64 (where drake seems to be looking for them). In the hindsight, I think this could have worked with the binary too. Hope this helps others get around this problem.

RussTedrake, 2015-04-08:
excellent.  we've had to do this on some of our project systems before, but i didn't actually realize we needed to for drake as well.  fwiw - these are our internal recommended instructions for addressing this:

After installing MATLAB, two of the symlinks for libraries need to be changed:
```
cd /usr/local/MATLAB/R2014a/sys/os/glnxa64
ls -l
```
The sym links for libstdc++.so.6 and libgfortran.so.3 should point to versions in /usr/lib, not local ones.

Before changing this libraries, first make sure g++ 4.4 is installed:
```
sudo apt-get install g++-4.4
```
Now, modify the symlinks:
```
sudo rm libgfortran.so.3
sudo ln -s /usr/lib/x86_64-linux-gnu/libgfortran.so.3.0.0 libgfortran.so.3
sudo rm libstdc++.so.6
sudo ln -s /usr/lib/gcc/x86_64-linux-gnu/4.4/libstdc++.so libstdc++.so.6
```

RussTedrake, 2015-04-08:
and thanks for your help!

RussTedrake, 2015-10-05:
this reappeared on the newest precompiled distros.

lrozo, 2015-10-09:
Hi guys!

I am running the make test for the latest version in github I just pulled. It seems that I am getting the same error, but I want to double-check that this is related to this post. The error I get when I run the "make test" is (Ubuntu 14.04):

Invalid MEX-file '/home/user/Git/drake-distro/drake/systems/plants/constructModelmex.mexa64': /usr/local/MATLAB/R2014a/bin/glnxa64/../../sys/os/glnxa64/libstdc++.so.6: version `GLIBCXX_3.4.18' not found (required by /home/user/Git/drake-distro/drake/pod-build/lib/libdrakeCollision.so)

Any idea about solving this issue? Maybe just manually moving the .so files? 

Thanks!

Leo

psiorx, 2015-10-09:
> Any idea about solving this issue?
Did you try installing g++4.4 and modifying the symlinks as outlined above?

CarloNicolini, 2015-11-04:
I get the same problem for a project of mine on Matlab R2015b, Ubuntu 14.04. The problem is that if i change the symlinks then other third party Matlab packages are crashing (in particular I'm using SPM Statistical parameter mapping package). Some ideas?

rdeits, 2015-11-10:
One (possibly crazy) idea would be to use the LD_PRELOAD trick before launching matlab. I haven't tried this, but it *might* work. What you would do is *not* create those symlinks, but instead, before launching matlab to use drake, do (from terminal):

```
LD_PRELOAD=/usr/lib/gcc/x86_64-linux-gnu/4.4/libstdc++.so matlab
```

And then when you want to use your other tools, and not drake, just launch matlab with:

```
matlab
```

instead. Of course, if you want to use both in the same matlab session, I'm not sure what to do.

RussTedrake, 2015-11-10:
Pretty sure i tried that (and it had problems).

ankitvora7, 2016-02-15:
I got the same error today. Is there a solution to this bug ?

PR #1862 (2016-03-24): Ensure Clang/GCC are at least the minimum version
This closes #960, closes #1281, and closes #1649.

PTAL @RussTedrake.
closed 2016-03-24 by RussTedrake