RobotLocomotion/drake #965: Calling path from solveLCPmex

RussTedrake, 2015-04-07:
Quick idea:  you could cut out the call to mexcallmatlab if you did a dlopen and called the mexFunction for the path matlab libs directly.  It might be a way to get a little more performance out.  See https://github.com/RobotLocomotion/drake/blob/master/util/drakeDebugMex.cpp 
For a very similar example.

RussTedrake, 2015-04-07:
Also..  Although it wouldn't get rid of the matlab lib dependencies, it would getvrid of the matlab runtime dependecy, so we'd be closer to being able to simulate lcps from a standalone app.

psiorx, 2015-04-07:
Would this work in windows?

RussTedrake, 2015-04-07:
no, but only because the path mex libraries don’t work on windows.

psiorx, 2015-04-07:
Played around with this idea a bit today.  It's close to working but I ran into a few snags.

The ```lcppath.mex*``` files expect a sparse matrix representation of ```M```.  This is normally handled by the matlab wrapper ```pathlcp.m``` which simply calls ```sparse()``` on the matrix before handing it down.  

Doing this directly in mex is a little bit hairier. 

I've written a general purpose ```eigen2mexsparse``` utility function which does this but MATLAB is complaining at runtime with this obscure error:  ```MATLAB:err_sparse_64bit_obsolete```

Digging a little deeper, it seems that our mex files are being compiled against the 32-bit mex API.  Referencing back to the release notes of R2006b, they say that in order to enable compilation against the new 64-bit mex API, we need to compile with the ```-largeArrayDims``` flag.  

I've added this to the ```MEX_COMPILE_FLAGS``` cmake variable in ```./cmake/mex.cmake``` but the error persists.  Manually compiling the mex file from within MATLAB with the same flag seems to work.  

Any idea why the ```-largeArrayDims``` flag doesn't seem to be taking hold?

psiorx, 2015-04-07:
I think I figured this one out.  

Our build logic is forcing 32-bit compatibility by including the ```-DMX_COMPAT_32``` flag somewhere.  Any reason why we need to keep this?

psiorx, 2015-04-08:
I submitted a pull request to the cmake submodule which resolves this.  I tested the performance of the dlopen approach and it seems to be comparable to mexCallMATLAB since the overhead of the function call is small compared with the time the function takes.  

51.896 s  original solveLCP
32.197 s  solveLCPmex w/mexCallMATLAB
29.847 s  solveLCPmex w/dlopen 

It's going to be somewhat messy trying to pick the right platform dependent mex-file if we go this route.

PR #969 (2015-04-12): Direct Mex-to-Mex LCP invocation
This gets rid of the ```mexCallMATLAB``` call in ```solveLCPmex```.

There are also two new utility functions in ```drakeUtil.cpp```.  The first is a ```callMexStandalone()``` function which allows mex files to be called  directly from c++.  The second is a function to build a MATLAB sparse matrix in mex.

In order to build a MATLAB sparse matrix in mex on 64-bit machines, the mex file must be compiled with the ```-largeArrayDims``` flag.  Currently, this is not the case so I suspect this will be broken until https://github.com/RobotLocomotion/cmake/pull/9 is i [... cut]
closed 2015-04-12 by RussTedrake